espressif/esp-idf #11674: Using ESP_LOG and enable system time, `esp_log_system_timestamp` crash

Blueknife-w, 2023-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32UE

### Power Supply used.

Battery

### What is the expected behavior?

Use `ESP_LOG` logs normally without crashing

### What is the actual behavior?

A crash is reported after running the log for a period of time

### Steps to reproduce.

1. Open menuconfig, enable `System Time` (path: Component config -> log output -> log timestamps)
2. copy the log components from esp-idf directory to project components directory.
3. change `log_freertos.c` file, inside `esp_log_system_timestamp`function as follow，the purpose is to add date printing during log time
```c
char *esp_log_system_timestamp(void)
{
    static char buffer[18] = {0};
    static _lock_t bufferLock = 0;

    if (xTaskGetSchedulerState() == taskSCHEDULER_NOT_STARTED) {
        uint32_t timestamp = esp_log_early_timestamp();
        for (uint8_t i = 0; i < sizeof(buffer); i++) {
            if ((timestamp > 0) || (i == 0)) {
                for (uint8_t j = sizeof(buffer) - 1; j > 0; j--) {
                    buffer[j] = buffer[j - 1];
                }
                buffer[0] = (char)(timestamp % 10) + '0';
                timestamp /= 10;
            } else {
                buffer[i] = 0;
                break;
            }
        }
        return buffer;
    } else {
        struct timeval tv;
        struct tm timeinfo;

        gettimeofday(&tv, NULL);
        localtime_r(&tv.tv_sec, &timeinfo);

        _lock_acquire(&bufferLock);
        snprintf(buffer, sizeof(buffer),
                 "%02d-%02d-%02d %02d:%02d:%02d.%03ld",
                 (timeinfo.tm_year - 100),
                 timeinfo.tm_mon + 1,
                 timeinfo.tm_mday,
                 timeinfo.tm_hour,
                 timeinfo.tm_min,
                 timeinfo.tm_sec,
                 tv.tv_usec / 1000);
        _lock_release(&bufferLock);

        return buffer;
    }
}
```

4. run the code. Crash after a period of time, the log is as follows

### Debug Logs.

```plain
I (23-06-14 20:07:31) http: Last esp error code: 0x103

assert failed: xQueueGenericSend queue.c:837 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace: 0x40081efe:0x3ffd78a0 0x4008a051:0x3ffd78c0 0x40091099:0x3ffd78e0 0x4008a984:0x3ffd7a00 0x40082f25:0x3ffd7a40 0x40082fd5:0x3ffd7a70 0x40083046:0x3ffd7a90 0x40184db1:0x3ffd7ac0 0x4017d1f0:0x3ffd7ae0 0x40198440:0x3ffd7b00 0x400dd342:0x3ffd7b70 0x40193c27:0x3ffd7ba0 0x4012110b:0x3ffd7bc0 0x401212fd:0x3ffd7be0 0x400db7ef:0x3ffd7c40 0x400dbfa4:0x3ffd7cb0 0x400db637:0x3ffd7d80 0x4008d226:0x3ffd7db0
0x40081efe: panic_abort at /home/wl/esp/esp-idf/components/esp_system/panic.c:412

0x4008a051: esp_system_abort at /home/wl/esp/esp-idf/components/esp_system/esp_system.c:135

0x40091099: __assert_func at /home/wl/esp/esp-idf/components/newlib/assert.c:78

0x4008a984: xQueueGenericSend at /home/wl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:837 (discriminator 8)

0x40082f25: lock_release_generic at /home/wl/esp/esp-idf/components/newlib/locks.c:192

0x40082fd5: _lock_release at /home/wl/esp/esp-idf/components/newlib/locks.c:198

0x40083046: __retarget_lock_release at /home/wl/esp/esp-idf/components/newlib/locks.c:334

0x40184db1: __tz_unlock at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/time/tzlock.c:50

0x4017d1f0: localtime_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/time/lcltime_r.c:123

0x40198440: esp_log_system_timestamp at /home/wl/workspace/esp32/kingtiger_esp32/kingtiger/build/../components/log/log_freertos.c:84

0x400dd342: _http_event_handler at /home/wl/workspace/esp32/kingtiger_esp32/kingtiger/build/../user/net/net_common/http.c:76 (discriminator 13)

0x40193c27: http_dispatch_event at /home/wl/esp/esp-idf/components/esp_http_client/esp_http_client.c:180

0x4012110b: esp_http_client_close at /home/wl/esp/esp-idf/components/esp_http_client/esp_http_client.c:1472

0x401212fd: esp_http_client_set_url at /home/wl/esp/esp-idf/components/esp_http_client/esp_http_client.c:900

0x400db7ef: _get_connect_secret at /home/wl/workspace/esp32/kingtiger_esp32/kingtiger/build/../user/net/basic/authenticate.c:90

0x400dbfa4: authenticate at /home/wl/workspace/esp32/kingtiger_esp32/kingtiger/build/../user/net/basic/authenticate.c:383

0x400db637: net_connect at /home/wl/workspace/esp32/kingtiger_esp32/kingtiger/build/../user/net/basic/net_app.c:46 (discriminator 1)

0x4008d226: vPortTaskWrapper at /home/wl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```

### More Information.

none

AxelLin, 2023-06-14:
You need this fix 60f678b736dc60257a5af9ea10d73625d58bb6ee , which is available in v5.1+.

Blueknife-w, 2023-06-16:
> You need this fix [60f678b](https://github.com/espressif/esp-idf/commit/60f678b736dc60257a5af9ea10d73625d58bb6ee) , which is available in v5.1+.

I tried to this fix [60f678b](https://github.com/espressif/esp-idf/commit/60f678b736dc60257a5af9ea10d73625d58bb6ee) , but the issue persists

Blueknife-w, 2023-06-20:
In response to this problem,  I added a test program. This program only involves the use of the serial port I use. After running it a few times, you can see the recurrence of this problem. The log is as follows:
```shell

I (23-06-17 06:28:35) main: 2Minimum free heap size: 272788 bytes cnt:301

I (23-06-17 06:28:35) main: 3Minimum free heap size: 272788 bytes cnt:302

I (23-06-17 06:28:35) main: 4Minimum free heap size: 272788 bytes cnt:303

I (23-06-17 06:28:35) main: 5Minimum free heap size: 272788 bytes cnt:304

I (23-06-17 06:28:35) main: 6Minimum free heap size: 272788 bytes cnt:305

I (23-06-17 06:28:35) main: 1Minimum free heap size: 272788 bytes cnt:306

W (23-06-17 06:28:35) main: c_thread1 runing, cnt:6
W (23-06-17 06:28:35) main: c_thread2 runing, cnt:6
W (23-06-17 06:28:35) main: c_thread3 runing, cnt:6
W (23-06-17 06:28:35) main: c_thread4 runing, cnt:6
I (23-06-17 06:28:35) main: a_thread1 runing, cnt:6
W (23-06-17 06:28:35) main: c_thread5 runing, cnt:6

assert failed: xQueueGenericSend queue.c:837 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace: 0x40081c16:0x3ffb99c0 0x40086c79:0x3ffb99e0 0x4008bd05:0x3ffb9a00 0x40087568:0x3ffb9b20 0x40082b65:0x3ffb9b60 0x40082c15:0x3ffb9b90 0x40082c86:0x3ffb9bb0 0x400e42ae:0x3ffb9be0 0x400e0638:0x3ffb9c00 0x400f1773:0x3ffb9c20 0x400d62e8:0x3ffb9c90 0x400896bd:0x3ffb9cb0
0x40081c16: panic_abort at /home/wl/esp/esp-idf/components/esp_system/panic.c:412

0x40086c79: esp_system_abort at /home/wl/esp/esp-idf/components/esp_system/esp_system.c:135

0x4008bd05: __assert_func at /home/wl/esp/esp-idf/components/newlib/assert.c:78

0x40087568: xQueueGenericSend at /home/wl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:837 (discriminator 8)

0x40082b65: lock_release_generic at /home/wl/esp/esp-idf/components/newlib/locks.c:192

0x40082c15: _lock_release at /home/wl/esp/esp-idf/components/newlib/locks.c:198

0x40082c86: __retarget_lock_release at /home/wl/esp/esp-idf/components/newlib/locks.c:334

0x400e42ae: __tz_unlock at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/time/tzlock.c:50

0x400e0638: localtime_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/time/lcltime_r.c:123

0x400f1773: esp_log_system_timestamp at /home/wl/workspace/esp32/hello_world/build/../components/log/log_freertos.c:84

0x400d62e8: a_thread at /home/wl/workspace/esp32/hello_world/build/../main/hello_world_main.c:43

0x400896bd: vPortTaskWrapper at /home/wl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: 2ae096f54168cdb4

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6992
load:0x40078000,len:15452
load:0x40080400,len:3840
0x40080400: _init at ??:?

entry 0x4008064c
I (27) boot: ESP-IDF v5.0 2nd stage bootloader
I (27) boot: compile time 14:25:43
I (27) boot: chip revision: v3.0
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
```
See attachment for full log and example project.

In addition, after running this project, I also found a problem. After program prints at high speed for a periof of time, the speed will suddenly drop. I cheked briefly, and the thread in app_main seems to hang up. I think there is anthor hide bugs.

[log.txt](https://github.com/espressif/esp-idf/files/11795478/log.txt)
[hello_world.zip](https://github.com/espressif/esp-idf/files/11795484/hello_world.zip)

KonstantinKondrashov, 2023-07-04:
Hi @Blueknife-w!
If you change the format you need to update the size of the `buffer`. Keep `buffer` in sync with the format. See below. Seems that is the case of your error.

```diff
 char *esp_log_system_timestamp(void)
 {
-    static char buffer[18] = {0};
+    static char buffer[22] = {0};
     static _lock_t bufferLock = 0;
 
     if (xTaskGetSchedulerState() == taskSCHEDULER_NOT_STARTED) {
@@ -85,11 +85,14 @@ char *esp_log_system_timestamp(void)
 
         _lock_acquire(&bufferLock);
         snprintf(buffer, sizeof(buffer),
-                 "%02d:%02d:%02d.%03ld",
-                 timeinfo.tm_hour,
-                 timeinfo.tm_min,
-                 timeinfo.tm_sec,
-                 tv.tv_usec / 1000);
+                "%02d-%02d-%02d %02d:%02d:%02d.%03ld",
+                (timeinfo.tm_year - 100),
+                timeinfo.tm_mon + 1,
+                timeinfo.tm_mday,
+                timeinfo.tm_hour,
+                timeinfo.tm_min,
+                timeinfo.tm_sec,
+                tv.tv_usec / 1000);
         _lock_release(&bufferLock);
 
         return buffer;
```
I close it, feel free to reopen it if it did not help. Thanks.