espressif/esp-idf #11675: Sunton LCD ESP32-4827S043 unexpected drift occured

Dadigno, 2023-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**IDF version**
v5.0

**Operating System used.**
Windows

**How did you build your project?**
idf.py build 

**If you are using Windows, please specify command line type.**
CMD

**Development Kit.**
Sunton LCD ESP32-4827S043
[4.3” 480X272-TN bare screen specs.pdf](https://github.com/espressif/esp-idf/files/11745755/4.3.480X272-TN.bare.screen.specs.pdf) 

**What is the expected behavior?**
I started from [rgb_lcd_example_main.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/rgb_panel/main/rgb_lcd_example_main.c) and managed to run [lv_demo_widgets](https://github.com/lvgl/lv_demos/tree/master/src/lv_demo_widgets). It works, but drift issue show off if `EXAMPLE_LCD_PIXEL_CLOCK_HZ` is greater than `(6*1000*1000)`. 
I would work with 9Mhz ( typical value in device specs ) 

https://github.com/espressif/esp-idf/assets/32482073/005dfe6e-2988-4e6e-8c00-b6b33dc196da

**What I tried:**
I read all doc included with the dev kit ( schematics, arduino example ecc ) and [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html). Also, as explained [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/rgb_panel), I tried double buffering setting  `CONFIG_EXAMPLE_DOUBLE_FB  = 1` and I'm using `.fb_in_psram = true`

Below my panel_config. I got pin timings and pin number from [here](https://github.com/espressif/esp-idf/files/11745755/4.3.480X272-TN.bare.screen.specs.pdf) and schematics.
[image]
```
.......
....
..
.
 esp_lcd_rgb_panel_config_t panel_config = {
        .clk_src = LCD_CLK_SRC_DEFAULT,
        .timings = {
            .pclk_hz = EXAMPLE_LCD_PIXEL_CLOCK_HZ,
            .h_res = LCD_H_RES,
            .v_res = LCD_V_RES,
            // The following parameters should refer to LCD spec
            .hsync_pulse_width = 4,
            .hsync_back_porch = 43,
            .hsync_front_porch = 8,
            .vsync_pulse_width = 4,
            .vsync_back_porch = 12,
            .vsync_front_porch = 8,
            .flags.pclk_active_neg = true,
        },
        .data_width = 16, // RGB565 in parallel mode, thus 16bit in width
        .num_fbs = 2,
         #if CONFIG_EXAMPLE_USE_BOUNCE_BUFFER
            .bounce_buffer_size_px = 10 * EXAMPLE_LCD_H_RES,
        #endif
        .psram_trans_align = 64,
        .hsync_gpio_num = EXAMPLE_PIN_NUM_HSYNC,
        .vsync_gpio_num = EXAMPLE_PIN_NUM_VSYNC,
        .de_gpio_num = EXAMPLE_PIN_NUM_DE,
        .pclk_gpio_num = EXAMPLE_PIN_NUM_PCLK,
        .disp_gpio_num = EXAMPLE_PIN_NUM_DISP_EN,
        .data_gpio_nums = {
            EXAMPLE_PIN_NUM_DATA0,
            EXAMPLE_PIN_NUM_DATA1,
            EXAMPLE_PIN_NUM_DATA2,
            EXAMPLE_PIN_NUM_DATA3,
            EXAMPLE_PIN_NUM_DATA4,
            EXAMPLE_PIN_NUM_DATA5,
            EXAMPLE_PIN_NUM_DATA6,
            EXAMPLE_PIN_NUM_DATA7,
            EXAMPLE_PIN_NUM_DATA8,
            EXAMPLE_PIN_NUM_DATA9,
            EXAMPLE_PIN_NUM_DATA10,
            EXAMPLE_PIN_NUM_DATA11,
            EXAMPLE_PIN_NUM_DATA12,
            EXAMPLE_PIN_NUM_DATA13,
            EXAMPLE_PIN_NUM_DATA14,
            EXAMPLE_PIN_NUM_DATA15,
        },
        .flags = { 
            .fb_in_psram = true // allocate frame buffer in PSRAM
        },
    };
    ESP_ERROR_CHECK(esp_lcd_new_rgb_panel(&panel_config, &panel_handle));

....
..
.
```

suda-morris, 2023-06-15:
> drift issue show off if EXAMPLE_LCD_PIXEL_CLOCK_HZ is greater than (6*1000*1000).

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/rgb_panel#troubleshooting

Dadigno, 2023-06-15:
Thank you for your response. I checked it out yet. As I said lowering frequency to 6MHz makes it work, but this value is out of LCD spec. I don't know what is depends on.

suda-morris, 2023-06-15:
It's a bandwidth issue of the PSRAM. There're some instructions here: https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/en/latest/software-framework/peripherals/lcd.html#why-do-i-get-drift-overall-drift-of-the-display-when-driving-an-rgb-lcd-screen

Dadigno, 2023-06-15:
Okay. Sorry, I missed this page. 

This: `Reduce data_cache_line_size to 32 bytes` makes it work at 9Mhz.

closed 2023-06-15 by Dadigno