espressif/esp-idf #11679: Example of I2S PDM project does not detect sound from microphone

misterb0407, 2023-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-EYE

### Power Supply used.

USB

### What is the expected behavior?

I expect for the I2S read buffer to display some random number when we speak to the microphone of the ESP-EYE board.

### What is the actual behavior?

The I2S read buffer keep showing dummy value of -30935 like following regardless we speak or not to the microphone.

```
I (341) main_task: Calling app_main()
I2S PDM RX example start
---------------------------
```
```
D (341) i2s_common: rx channel is registered on I2S0 successfully
D (351) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (361) i2s_common: i2s rx channel enabled
I (361) main_task: Returned from app_main()
Read Task: i2s read 2048 bytes
-----------------------------------
[0] -6595 [1] 0 [2] -29199 [3] -32768
[4] -30203 [5] -32156 [6] -30704 [7] -31348

Read Task: i2s read 2048 bytes
-----------------------------------
[0] -30935 [1] -30935 [2] -30935 [3] -30935
[4] -30935 [5] -30935 [6] -30935 [7] -30935
.....
Read Task: i2s read 2048 bytes
-----------------------------------
[0] -30935 [1] -30935 [2] -30935 [3] -30935
[4] -30935 [5] -30935 [6] -30935 [7] -30935
```

### Steps to reproduce.

1. Go to the example project folder
$ cd <esp-idf>/examples/peripherals/i2s/i2s_basic/i2s_pdm
2. Configure the project to use `PDM RX` mode
`$ idf.py menuconfig` -> I2S PDM Example Configuration -> I2S PDM direction -> PDM RX
[... 214 lines cut ...]
Read Task: i2s read 2048 bytes
-----------------------------------
[0] -30935 [1] -30935 [2] -30935 [3] -30935
[4] -30935 [5] -30935 [6] -30935 [7] -30935

Read Task: i2s read 2048 bytes
-----------------------------------
[0] -30935 [1] -30935 [2] -30935 [3] -30935
[4] -30935 [5] -30935 [6] -30935 [7] -30935

Read Task: i2s read 2048 bytes
-----------------------------------
[0] -30935 [1] -30935 [2] -30935 [3] -30935
[4] -30935 [5] -30935 [6] -30935 [7] -30935

Read Task: i2s read 2048 bytes
-----------------------------------
[0] -30935 [1] -30935 [2] -30935 [3] -30935
[4] -30935 [5] -30935 [6] -30935 [7] -30935

Read Task: i2s read 2048 bytes
-----------------------------------
[0] -30935 [1] -30935 [2] -30935 [3] -30935
[4] -30935 [5] -30935 [6] -30935 [7] -30935
```

chucrut, 2023-06-16:
I have the same problem. The buffer is filled with exactly the same "random" numbers each time. 
Is not reading data from the PDM mic

L-KAYA, 2025-01-13:
Hi @chucrut, does the issue still exist?

This 'random' data actually receiving `0` on the signal line, please check if the GPIO assignment is correct. And also please check the minimum clock that supported  by your PDM MIC. Some PDM microphones do not work when the input clock is below its minimum clock.

L-KAYA, 2025-02-24:
Going to close the issue due to lack of further response. Please feel free to re-open it.

closed 2025-02-24 by Alvin1Zhang