espressif/esp-idf #11680: What IO ports are used to connect the LAN8720 network cable to ESP32-S3?

hzhh110, 2023-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

What IO ports are used to connect the LAN8720 network cable to ESP32-S3?
Previously, ESP32:
IO17 ETH for clock_ PHY_ LAN8720 ETH_ CLK_ MODE=ETH_ CLOCK_ GPIO17_ OUT (3)
GPIORMII SignalESP32 EMAC FunctionNotes
GPIO21 TX_ EN EMAC_ TX_ EN
GPIO19 TX0 EMAC_ TXD0
GPIO22 TX1 EMAC_ TXD1
GPIO25 RX0 EMAC_ RXD0
GPIO26 RX1 EMAC_ RXD1
GPIO27 CRS_ DV EMAC_ RX_ DRV
GPIO23 ETH_ MDC_ PIN
GPIO18 ETH_ MDIO_ PIN

### What is the actual behavior?

What IO ports are used to connect the LAN8720 network cable to ESP32-S3?

### Steps to reproduce.

What IO ports are used to connect the LAN8720 network cable to ESP32-S3?

### More Information.

What IO ports are used to connect the LAN8720 network cable to ESP32-S3?

liguilini, 2023-06-15:
s3 no longer supports RMII,it removed integrated MAC.You can use SPI ethernet chip,like w5500

hzhh110, 2023-06-15:
Is there a recommended IO port for ESP32-S3 to connect to W5500 using SPI? I have temporarily referenced the IO port corresponding to the camera in ESP-S3-EYE
#Define CAMERA_ Module_ NAME "ESP-S3-EYE"
#Define CAMERA_ PIN_ PWDN -1
#Define CAMERA_ PIN_ RESET -1
#Define CAMERA_ PIN_ VSYNC 6 OK
#Define CAMERA_ PIN_ HREF 7 OK
#Define CAMERA_ PIN_ PCLK 13 OK
#Define CAMERA_ PIN_ XCLK 15 OK
#Define CAMERA_ PIN_ SIOD 4 OK
#Define CAMERA_ PIN_ SIOC 5 OK
#Define CAMERA_ PIN_ D0 11 OK
#Define CAMERA_ PIN_ D1 9 OK
#Define CAMERA_ PIN_ D2 8 OK
#Define CAMERA_ PIN_ D3 10 OK
#Define CAMERA_ PIN_ D4 12 OK
#Define CAMERA_ PIN_ D5 18 OK
#Define CAMERA_ PIN_ D6 17 OK
#Define CAMERA_ PIN_ D7 16 OK

hzhh110, 2023-06-15:
> s3 no longer supports RMII,it removed integrated MAC.You can use SPI ethernet chip,like w5500

Is there a recommended IO port for ESP32-S3 to connect to W5500 using SPI? I have temporarily referenced the IO port corresponding to the camera in ESP-S3-EYE
#Define CAMERA_ Module_ NAME "ESP-S3-EYE"
#Define CAMERA_ PIN_ PWDN -1
#Define CAMERA_ PIN_ RESET -1
#Define CAMERA_ PIN_ VSYNC 6 OK
#Define CAMERA_ PIN_ HREF 7 OK
#Define CAMERA_ PIN_ PCLK 13 OK
#Define CAMERA_ PIN_ XCLK 15 OK
#Define CAMERA_ PIN_ SIOD 4 OK
#Define CAMERA_ PIN_ SIOC 5 OK
#Define CAMERA_ PIN_ D0 11 OK
#Define CAMERA_ PIN_ D1 9 OK
#Define CAMERA_ PIN_ D2 8 OK
#Define CAMERA_ PIN_ D3 10 OK
#Define CAMERA_ PIN_ D4 12 OK
#Define CAMERA_ PIN_ D5 18 OK
#Define CAMERA_ PIN_ D6 17 OK
#Define CAMERA_ PIN_ D7 16 OK

kostaond, 2023-06-15:
[Here](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/components/ethernet_init/Kconfig.projbuild) you can find default GPIO for SPI Ethernet modules. However, if you use ESP32-S3, you can connect it to whatever IO you want, see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux). Note that it used to be an concern to use dedicated SPI pins via IO_MUX for ESP32 to be able to achieve higher SPI speed but it is no longer applicable to S3.

Please close this issue since it is not real issue.

KaeLL, 2023-06-15:
> but it is no longer applicable to S3

@kostaond Would you mind elaborating further on why is that?

kostaond, 2023-06-16:
[ESP32](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux)
_The GPIO matrix introduces flexibility of routing but also brings the following disadvantages:
Increases the input delay of the MISO signal, which makes MISO setup time violations more likely. If SPI needs to operate at high speeds, use dedicated IO_MUX pins.
Allows signals with clock frequencies only up to 40 MHz, as opposed to 80 MHz if IO_MUX pins are used._

[ESP32-S3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux)
_When an SPI Host is set to 80MHz or lower frequencies, routing SPI pins via the GPIO matrix will behave the same compared to routing them via IOMUX._

KaeLL, 2023-06-16:
I meant what exactly is the cause behind this limitation on the vanilla ESP32 and how did you guys fix it on newer variants?

kostaond, 2023-06-17:
@KaeLL, frankly speaking, I don't know. I'm software guy responsible for Ethernet... This question is beyond ESP-IDF scope. You can try to ask at ESP Forum.

I'm closing the GH issue since the original question has been addressed and there is no real issue related to the ESP-IDF.

closed 2023-06-17 by kostaond