espressif/esp-idf #11681: Bluetooth connection lost after pairing

chucrut, 2023-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 WROOM 32

### Power Supply used.

USB

### What is the expected behavior?

My code communicate an ESP32 with an Android app using BT classic, based in this example code:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor
With the same sdkconfig configuration settings as the example

### What is the actual behavior?

I turn on the bluetooth in the phone. I select the ESP32 device. 

Then pop up a window with the pairing number:
[image]
At the same time this windows pops up, in the terminal i receive the following message:
W (162921) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
W (162940) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000

As soon as I pair the devices in the phone, i receive the following message in the terminal:
W (245220) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x13
W (245296) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x16

The authentication is succesfull, but the connection process stops there.

I think i had this same issue months ago. I think was related with HCI. Now i have a fresh installation, with the latest ESP-IDF version. ESP-IDF works fine. VSCode works fine.
I tried to search the error code rsn:13, but i cannot find any information.

### Steps to reproduce.

Tried to run a code based on the acceptor example.

boblane1, 2023-06-16:
Hi @chucrut,

Thanks for your report! We have introduced L2CAP enhanced retransmission mode recently, however this feature is not compatible with the Bluetooth stack in Android. You can set `esp_spp_enable_l2cap_ertm` to `false` to see if this issue persists.

chucrut, 2023-06-16:
Hello!.
Many thanks for your help. I set esp_spp_enable_l2cap_ertm to false but the
problem persists. After pairing, the bluetooth connection is lost.
I appreciate any ideas!.

Many thanks!

chucrut, 2023-06-16:
Hello @boblane1 
Many thanks for your help. I cannot modify the Android app. Months ago, the bluetooth connection worked fine and the Android app version was the same. I didn't had this problem before. I think is a HCI related problem. 
After pairing, the bluetooth connection is lost.
I appreciate any ideas!.

Many thanks!

chucrut, 2023-06-16:
[image]

chucrut, 2023-06-16:
I tried with this configuration:
[image]

chucrut, 2023-06-16:
And also tried with HCI instead of PCM

chucrut, 2023-06-16:
I tested both ESP32 WROOM 32 and ESP32 PICO MINI 02 and their behaviour is slightly different
ESP32 WROOM 32 not even pair
ESP32 PICO MINI 02 pair and disconnect

xiongweichao, 2023-06-19:
Hi @chucrut ，

1. May I ask if there is this problem when testing with other Android apps?
2. Could you please provide this Android APP?

Thanks

chucrut, 2023-06-19:
@xiongweichao 
This happens when pairing (trying to establish bluetooth connection). Is not an application problem.
Many thanks

xiongweichao, 2023-06-19:
@chucrut  Because the phone does not support SPP by default, it will disconnect after the pairing is completed.

chucrut, 2023-06-19:
@xiongweichao the phone support SPP. I used the same phone for SPP before with ESP32, but now the connection doesn't work.
I tried from different mobile phones and they disconnect after pairing, with the message:
W (245220) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x13
W (245296) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x16

chucrut, 2023-06-19:
@xiongweichao can you run the acceptor example and try to connect from your phone
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor

chucrut, 2023-06-22:
@xiongweichao @boblane1 I cannot run a simple Bluetooth acceptor example. The connection drops after pairing, like the logs shown before.

chucrut, 2023-06-22:
Please, can you run the acceptor example and try to connect from your phone
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor

I cannot run a simple Bluetooth acceptor example using 5.1 version of Espressif ESP-IDF. The connection drops after pairing, like the logs shown before.

chucrut, 2023-06-23:
@zhp0406 @euripedesrocha @Weijian-Espressif @shangke1112 @esp-lis @xiewenxiang @chucrut Please, can you run the acceptor example and try to connect from your phone
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor

I cannot run a simple Bluetooth acceptor example using 5.1 version of Espressif ESP-IDF. The connection drops after pairing, like the logs shown before.

chucrut, 2023-06-23:
@zhp0406 @euripedesrocha @Weijian-Espressif @shangke1112 @esp-lis @xiewenxiang @xiongweichao @boblane1 
I tested ESP-IDF 4.2 , 4.4, 5.0 and  5.1
This happens with all versions.
Before each installation I deleted manually all ESP, ESP-IDF folders.
For installation I used ESP-IDF extension for VS-Code.
The other libraries works fine. Just Bluetooth connection is the problem. 
I tested with 2 android phones: Samsung S7 and Samsung S22
I know S7 is fine because months ago acceptor example worked fine

xiongweichao, 2023-06-25:
@chucrut Have you tried using other apps?

PiotrBzdrega, 2023-07-06:
@chucrut I have the same response as you after the connection (immediate disconnection).
As guys mentioned, with Bluetooth Serial Terminal connection with paired devices is established smoothly.
[Video](https://youtu.be/cDbdaLLhqHs)

xiongweichao, 2024-02-23:
> @chucrut I have the same response as you after the connection (immediate disconnection). As guys mentioned, with Bluetooth Serial Terminal connection with paired devices is established smoothly. [Video](https://youtu.be/qt6hw5CxcPs)

@chucrut Please try this method and let me know the test results. Thanks!

Sherry616, 2024-03-28:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-03-28 by Sherry616

xiongweichao, 2025-08-08:
> Hi @chucrut,
> 
> Thanks for your report! `W (245220) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x13` means that that SPP acceptor break the disconnection locally. You can establish the SPP connection through an APP, e.g. Serial Bluetooth Terminal. Please let me know if the issue persists.

@Bobobel  The root cause of this is that the phone does not support SPP by default.

> I am facing "BT_HCI: hcif disc complete: hdl 0x81, rsn 0x8" witout preample during an open BT classic HID connection.

This seems like a bug. Please open a new issue to report it. Thanks!