espressif/esp-idf #11682: HTTPS_OTA with WSS_SERVER

baldhead69, 2023-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

Is it possible to integrate https_ota with the wss server ?

Is there any example in this regard ?

NOTES:
ESP32-S3 chip
esp-idf v5.0.2
@david

Thank's.

ESP-YJM, 2023-06-21:
@baldhead69 OTA itself does not care what the underlying transmission is, so websocket can also be used for OTA. You can use websocket to tramsmit the firmware data, and use `esp_ota_*` API to write data to partition. There is no full example, but you can follow this example https://github.com/espressif/esp-idf/blob/master/examples/system/ota/native_ota_example/main/native_ota_example.c and use websocket to replace http.

baldhead69, 2023-06-21:
Hi @ESP-YJM ,

There is no possibility to run websocket and http in parallel ?

For example: after upgrade protocol from  http to websocket, the esp is running "only" in websocket, then i establish a new connection with http protocol to do firmware download.

Thank's.

ESP-YJM, 2023-06-25:
Do you mean websocket is not for firmware downlaod? It is possibility to run websocket and http in parallel. But for websocket connections, what does it do? For getting the firmware download url?

baldhead69, 2023-06-25:
> Do you mean websocket is not for firmware downlaod? It is possibility to run websocket and http in parallel. But for websocket connections, what does it do? For getting the firmware download url?

Sorry, i need to read more to understand the ota process.  Actually i need to open an http/s client connection to download the firmware from some server on the web or local server to test, from what i understand.

Can i run http/https client and ws/wss server at the same time on esp32-s3 ?

ESP-YJM, 2023-06-25:
Yes, The http client and websocket client  can run at the same time.

baldhead69, 2023-06-25:
> Yes, The http client and websocket client  can run at the same time.

Not websocket client, websocket server.

I have a websocket secure ( wss ) running on esp32-s3.

ESP-YJM, 2023-06-26:
Yes, also support run websocket server at the same time.

closed 2023-07-17 by ESP-YJM