espressif/esp-idf #11684: ESP_ERROR_CHECK(led_strip_clear(led_strip)); crashes after calling at after deleting FreeRTOS task that controls RGB

krupis, 2023-06-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have the following RGB_setup(){

```
TaskHandle_t rainbow_task;

void RGB_setup(){

    gpio_config_t io_conf;
    //disable interrupt
    io_conf.intr_type = GPIO_INTR_NEGEDGE;
    //set as output mode
    io_conf.mode = GPIO_MODE_INPUT ;
    //bit mask of the pins that you want to set,e.g.GPIO18/19
    io_conf.pin_bit_mask = GPIO_BUTTON_INPUT_SEL;
    //disable pull-down mode
    io_conf.pull_down_en = GPIO_PULLDOWN_ENABLE;
    //disable pull-up mode
    io_conf.pull_up_en = GPIO_PULLUP_ENABLE;
    //configure GPIO with the given settings
    gpio_config(&io_conf);

    gpio_evt_queue = xQueueCreate(2, sizeof(uint32_t));
    //start gpio task
    xTaskCreate(Button_detection_task, "Button_detection_task", 2048, NULL, 10, NULL);

    gpio_install_isr_service(0);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(BUTTON_1, button_handler, (void*) BUTTON_1);

    const esp_timer_create_args_t fading_lights = {
            .callback = &RGB_fade_in_out_callback,
            /* name is optional, but may help identify the timer when debugging */
            .arg = &rgb_params,
            .name = "fade_in_out"
    };
    ESP_ERROR_CHECK(esp_timer_create(&fading_lights, &fading_lights_timer));

    const esp_timer_create_args_t running_lights = {
        .callback = &RGB_running_lights_callback,
        /* name is optional, but may help identify the timer when debugging */
        .arg = &rgb_params,
        .name = "running lights"
    };
    ESP_ERROR_CHECK(esp_timer_create(&running_lights, &running_lights_timer));

    strip_color.red = 0;
    strip_color.blue = 0;
    strip_color.green = 0;

    led_strip_config_t strip_config = {
        .strip_gpio_num = LED_STRIP_BLINK_GPIO,   // The GPIO that connected to the LED strip's data line
        .max_leds = LED_STRIP_LED_NUMBERS,        // The number of LEDs in the strip,
        .led_pixel_format = LED_PIXEL_FORMAT_GRB, // Pixel format of your LED strip
        .led_model = LED_MODEL_WS2812,            // LED strip model
        .flags.invert_out = false,                // whether to invert the output signal
    };

    // LED strip backend configuration: RMT
    led_strip_rmt_config_t rmt_config = {
        .clk_src = RMT_CLK_SRC_DEFAULT,        // different clock source can lead to different power consumption
        .resolution_hz = LED_STRIP_RMT_RES_HZ, // RMT counter clock frequency
        .flags.with_dma = false,               // DMA feature is available on ESP target like ESP32-S3
    };
    ESP_ERROR_CHECK(led_strip_new_rmt_device(&strip_config, &rmt_config, &led_strip));
    ESP_LOGI(TAG, "Created LED strip object with RMT backend");
    bool led_on_off = false;
    ESP_LOGI(TAG, "Start blinking LED strip");
    ESP_ERROR_CHECK(led_strip_clear(led_strip));
}

```

And the following functions:

```
void RGB_clear_strip(){
    ESP_ERROR_CHECK(led_strip_clear(led_strip));
}

void RGB_rainbow(){
    xTaskCreate(RGB_sine_rainbow,"RGB_sine_rainbow",10000,NULL,5,&rainbow_task); // receiving commands from main uart
}

void Delete_RGB_rainbow_task(){
    if(rainbow_task != NULL){
        ESP_LOGW(TAG,"delete rainbow task");
        vTaskDelete(rainbow_task);
        rainbow_task = NULL;

    }
    else{
        printf("Rainbow task is not running \n");
    }
}

```

RGB_rainbow function creates FreeRTOS tasks that is periodically called and changes my RGB strip colors. I want to be able to stop the FreeRTOS task hence I created Delete_RGB_rainbow_task function.

After I delete the rainbow task, the RGB strip saves the last color. I want to be able to clear the whole strip so I use RGB_clear_strip() function. However, after calling RGB_clear_strip() function the program crashes. 

I am testing it via the UART0 commands. I have the following UART0 commands declared:

```
 if (!strncmp("rainbow", line,7))
    {	
        //Stop_current_animation();
        //vTaskDelay(100/portTICK_PERIOD_MS);
        RGB_rainbow();
        printf("Started rgb rainbow lights animation \n");
        return 0;
    }

    if (!strncmp("delete_rainbow_task", line,19))
    {	
        Delete_RGB_rainbow_task();
        return 0;
    }
    if (!strncmp("clear_strip", line,11))
    {	
        RGB_clear_strip();
        return 0;
    }
```

The log output when I call "clear_strip" command

```
rainbow
Started rgb rainbow lights animation 
delete_rainbow_task
[1B][0;33mW (285441) example: delete rainbow task[1B][0m
clear_strip
[1B][0;31mE (288771) rmt: rmt_enable(189): channel not in init state[1B][0m
[1B][0;31mE (288771) led_strip_rmt: led_strip_rmt_refresh(72): enable RMT channel failed[1B][0m
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4200a31a
file: "./components/RGB/RGB.c" line 252
func: RGB_clear_strip
expression: led_strip_clear(led_strip)

abort() was called at PC 0x4037e717 on core 0

Backtrace: 0x40375fd6:0x3fcaafd0 0x4037e721:0x3fcaaff0 0x40384d6a:0x3fcab010 0x4037e717:0x3fcab080 0x4200a31a:0x3fcab0b0 0x4200a8e7:0x3fcab0d0 0x4200a935:0x3fcab100 0x4038197d:0x3fcab520

ELF file SHA256: 2dfaa4ee09a6cada

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375975
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x164c
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2ef8
entry 0x403c9900
[1B][0;32mI (29) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader[1B][0m
[1B][0;32mI (29) boot: compile time 12:04:58[1B][0m
[1B][0;32mI (29) boot: chip revision: v0.1[1B][0m
[1B][0;32mI (32) boot.esp32s3: Boot SPI Speed : 80MHz[1B][0m
[1B][0;32mI (36) boot.esp32s3: SPI Mode       : DIO[1B][0m
[1B][0;32mI (41) boot.esp32s3: SPI Flash Size : 8MB[1B][0m
[1B][0;32mI (46) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (51) boot: Partition Table:[1B][0m
[1B][0;32mI (55) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (62) boot:  0 otadata          OTA data         01 00 0000d000 00002000[1B][0m
[1B][0;32mI (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m

```

suda-morris, 2023-06-16:
hi @krupis this could be that task A is running a "refresh" function, and task B is running a "clear" function concurrently. They're not allowed to work at the same time for the same led_strip handle. If you have to make it work across multiple threads, please consider adding a mutex to serialize the access to the led_strip handle.

krupis, 2023-06-16:
> hi @krupis this could be that task A is running a "refresh" function, and task B is running a "clear" function concurrently. They're not allowed to work at the same time for the same led_strip [...]

I thought the same thing initially but I dont think that is possible.

After I call Delete_RGB_rainbow_task function, it deletes the RGB rainbow task. The only other task running on my ESP32 is UART0 command parser task that is waiting for serial input.

So when I send final command "clear_strip", there are no other tasks that are interacting with RGB strip. The only other task that was interacting with the RGB strip was rainbow task which is now deleted.

suda-morris, 2023-06-16:
Sorry, I miss reading your post. It's not about race conditions. It's simply because when the "rainbow task" exists in the middle, where the RMT channel was enabled, but not disabled. So after you call "clear" next time, it detects that the channel is not in the expected state. To exit the "rainbow task" gracefully, you can use some signal or flag. In that task, you can detect if the "delete" flag is raised, then exit the task.

suda-morris, 2023-06-19:
@krupis yes, stopping an esp timer won't break an undergoing alarm callback. So the "refresh" function can get fully run.

suda-morris, 2023-06-19:
I'm going to close this issue as you've found a way to fix it. But feel free to reopen it if you think there's something that needs improvement in the led_strip component.

closed 2023-06-19 by suda-morris

krupis, 2023-08-11:
@suda-morris 
Hello. I have been working with my project for the RGB lights and I still have some issues regarding crashing when calling ```led_strip_clear()``` function.

I will try to explain what my device does in as much detail as possible, I hope you understand and can help:

I have created a periodic timer that is called and update RGB strip color. It is called periodically at a very fast rate to generate nice Rainbow effect.

```
    const esp_timer_create_args_t rainbow_lights = {
        .callback = &RGB_rainbow_lights_callback,
        /* name is optional, but may help identify the timer when debugging */
        .arg = &rgb_params,
        .name = "rainbow lights"
    };
    ESP_ERROR_CHECK(esp_timer_create(&rainbow_lights, &rainbow_lights_timer));

void RGB_rainbow_lights_callback(void* arg)
{
    static uint16_t offset = 360/LED_STRIP_LED_NUMBERS;
    static uint8_t led_number = 0;
    static uint16_t angle = 0;
    for(int i = 0; i < LED_STRIP_LED_NUMBERS;i++){
        sineLED(i,angle+(offset*i));
    }
    angle++;
    if(angle == 359){
        angle = 0;
    }
    ESP_ERROR_CHECK(led_strip_refresh(led_strip));   
}

```

Everything seems to be okay until I try to call ```led_strip``` from another threads or tasks.

Lets say I have UART0 parser task that is waiting for serial command. When specific command is received, for example ```speed:```, It will stop currently running timer, readjust the timer interval and create the timer again:

```
void Stop_current_animation(){
    if(rainbow_lights_timer != NULL){
        if(esp_timer_is_active(rainbow_lights_timer) == 1){
            ESP_LOGW("RGB","STOPPING rainbow lights timer");
            ESP_ERROR_CHECK(esp_timer_stop(rainbow_lights_timer));
            ESP_ERROR_CHECK(led_strip_clear(led_strip));
        }
    }
    else{
        printf("rainbow lights timer is null \n");
    }
}
```

The following is called when I ```speed:``` command is received:
```
    if (!strncmp("speed:", line,6))
    {	
        char temp_buf[5];
		uint16_t data_to_write=0;
		for(int i = 0;i <=(strlen(line)-6);i++){
			temp_buf[i] = line[6+i];
		}
		data_to_write = atoi(temp_buf);
        Stop_current_animation();
        rgb_params.ramp_up_time = data_to_write; //takes 3 seconds to reach target and another 3 seconds to fade down
    if(esp_timer_is_active(rainbow_lights_timer) == 0){
        ESP_ERROR_CHECK(esp_timer_start_periodic(rainbow_lights_timer, rgb_parameters->ramp_up_time*10)); //
        printf("rainbow lights timer started \n");
    }

        printf("Started rgb rainbow lights animation \n");
        return 0;
    }
```

Ocasionally, when the ```speed``` command is received, the device will crash with the following error message:
```
speed:200
[1B][0;33mW (29706) RGB: STOPPING rainbow lights timer[1B][0m
[1B][0;31mE (29706) rmt: rmt_enable(197): channel not in init state[1B][0m
[1B][0;31mE (29706) led_strip_rmt: led_strip_rmt_refresh(72): enable RMT channel failed[1B][0m
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4200a5f9
file: "./components/RGB/RGB.c" line 564
func: Stop_current_animation
expression: led_strip_clear(led_strip)

abort() was called at PC 0x4037ea37 on core 0

Backtrace: 0x403760ce:0x3fcb8910 0x4037ea41:0x3fcb8930 0x403850f2:0x3fcb8950 0x4037ea37:0x3fcb89c0 0x4200a5f9:0x3fcb89f0 0x4200aa05:0x3fcb8a10 0x4200aa99:0x3fcb8a40 0x40381c9d:0x3fcb8e60

ELF file SHA256: 7b6e7c33849130c4

```

I think this is because ```led_strip_clear``` is being called in ```Stop_current_animation```.

Could you please suggest appropriate way to handle calls to ```led_strip``` functions from different Tasks and places in the code?