espressif/esp-idf #11685: esp32s3 can not accepts ble conn param as gattc

fzboffice, 2023-06-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5-98-ge33a1808d6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

esp32s3 accepts update conn param

### What is the actual behavior?

a slave device update conn param,esp32s3 as master print update connection params status = 19 when min interval = 6 max interval = 6 latency = 5 timeout=500, but it works when slave update min interval = 32 max interval = 32 latency = 50 timeout=500

### Steps to reproduce.

1. Step
2. Step
3. Step
...
.

fzboffice, 2023-06-16:
by the way,esp32 no above issues，esp32 can accepts both conn param

zhp0406, 2023-06-25:
Hello, can you describe how to reproduce this issue? Did you use coded PHY or multiple connections? @fzboffice

fzboffice, 2023-06-26:
@zhp0406  thanks! I'm waitting for update.