espressif/esp-idf #11686: Using pos-edge GPIO interrupt with GPIO light-sleep wake on same pin causes interrupt wdt timeout

filipbudisa, 2023-06-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1115-g3a45d4e949

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

Battery

### What is the expected behavior?

The GPIO pin gets pulled to high, ESP wakes from light-sleep, and executes the ISR.

### What is the actual behavior?

The GPIO pin gets pulled to high, ESP wakes from light-sleep, and then crashes.

### Steps to reproduce.

1. Compile and run the provided example
2. After the ESP goes to sleep, connect IO 34 to 3.3v, and keep the pin connected like this for a second or two.

Connecting the pin high for only a moment won't cause the crash, it needs to be connected for at least a second, until the wdt gets triggered.

### Debug Logs.

```plain
Going to sleep...
GPIO is 0
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x400830dc  PS      : 0x00060035  A0      : 0x80083152  A1      : 0x3ffbe6a0  
0x400830dc: gpio_isr_loop at /home/cm/Repo/ESP32/esp-idf/components/driver/gpio/gpio.c:449

A2      : 0x00000004  A3      : 0x00000020  A4      : 0x00060021  A5      : 0x00000022  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffbe680  
A10     : 0x00000000  A11     : 0xffffff00  A12     : 0x0241e40e  A13     : 0x3ffb10a0  
A14     : 0x00219b1d  A15     : 0x00060323  SAR     : 0x0000001e  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
0x400014fd: strlen in ROM

0x4000150d: strlen in ROM

Core  0 was running in ISR context:
EPC1    : 0x400d1e8b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
0x400d1e8b: uart_hal_write_txfifo at /home/cm/Repo/ESP32/esp-idf/components/hal/uart_hal_iram.c:35

Backtrace: 0x400830d9:0x3ffbe6a0 0x4008314f:0x3ffbe6d0 0x40082e36:0x3ffbe700 0x4000bfed:0x3ffb1100 0x40088693:0x3ffb1110 0x400d8c53:0x3ffb1130 0x400d5262:0x3ffb1170 0x400d52cd:0x3ffb1190 0x400d52d7:0x3ffb1200 0x400e8518:0x3ffb1220 0x40088531:0x3ffb1250
0x400830d9: gpio_ll_clear_intr_status_high at /home/cm/Repo/ESP32/esp-idf/components/hal/esp32/include/hal/gpio_ll.h:273 (discriminator 2)
 (inlined by) gpio_isr_loop at /home/cm/Repo/ESP32/esp-idf/components/driver/gpio/gpio.c:467 (discriminator 2)

0x4008314f: gpio_intr_service at /home/cm/Repo/ESP32/esp-idf/components/driver/gpio/gpio.c:492

0x40082e36: _xt_lowint1 at /home/cm/Repo/ESP32/esp-idf/components/xtensa/xtensa_vectors.S:1226

0x4000bfed: _xtos_set_intlevel in ROM

0x40088693: vPortClearInterruptMaskFromISR at /home/cm/Repo/ESP32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:568
 (inlined by) vPortExitCritical at /home/cm/Repo/ESP32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:532

0x400d8c53: esp_light_sleep_start at /home/cm/Repo/ESP32/esp-idf/components/esp_hw_support/sleep_modes.c:1092

0x400d5262: sleep() at /home/cm/Projects/Clockstar-Firmware/examples/Sleep_Wake_Interrupt.cpp:36

0x400d52cd: init() at /home/cm/Projects/Clockstar-Firmware/examples/Sleep_Wake_Interrupt.cpp:71

0x400d52d7: app_main at /home/cm/Projects/Clockstar-Firmware/examples/Sleep_Wake_Interrupt.cpp:75

0x400e8518: main_task at /home/cm/Repo/ESP32/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)

0x40088531: vPortTaskWrapper at /home/cm/Repo/ESP32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

Use case: We have a custom board with ESP32-WROOM-32 and an IMU (LSM6DS3). One of the interrupt lines from the IMU is connected to the ESP, and the ESP is programmed to trigger an interrupt on the positive edge. The ISR sets a semaphore that signals another thread that data should be processed, and in that thread, the IMU is signaled to pull the interrupt line back down.

The IMU keeps the interrupt line high until it is signaled to pull it back down, therefore, if we use a high-level interrupt instead a pos-edge, the ESP will keep executing the ISR and will crash with an interrupt wdt timeout. So far so good, we just need to use a pos-edge interrupt and everything works as expected. However, problems arise when we also need to wake the ESP from light sleep using that same interrupt signal.

gpio_wakeup_enable(gpio_num_t, gpio_int_type_t) will set the interrupt type to high-level, regardless of whether the gpio wakeup or gpio interrupt is configured first. The result is that the ESP will always crash after waking up from light sleep since the line is kept high and never gets the chance to get pulled low.

A workaround is to re-configure the gpio pin with high-level intr_type after the interrupt and wake are set up.

The reason for this issue is two-fold: 
1) To raise awareness about this behaviour. It is up to you to decide whether this is intended behaviour or a genuine bug.
2) To hear your official thoughts about this, and if you see any problems this workaround could cause in production.

Attached are the minimum example that causes the crash and my sdkconfig. In the example, comment out the initWake() and initInt() function calls (lines 66 and 67) to disable either the gpio wake or gpio interrupt. When only one of those is turned on, everything works a-ok, the program crashes only when both are enabled. Comment out the gpio_config function call on line 68 to enable the workaround. To make the ESP go to sleep after everything is set up, uncomment the last two lines of the init() function.

[example.zip](https://github.com/espressif/esp-idf/files/11770203/example.zip)
The init code in the example sets the IO13 to input mode, which is required on our development board and isn't relevant to this issue.

Thanks,
Filip

songruo, 2023-06-20:
Hi @filipbudisa, 

> if we use a high-level interrupt instead a pos-edge, the ESP will keep executing the ISR and will crash with an interrupt wdt timeout. 

Your reasoning on the trigger of interrupt wdt is correct. And yes, this is the intended behavior. For any level-triggered interrupt, the users are expected to either disable the interrupt by `gpio_intr_disable` or change the interrupt type by `gpio_set_intr_type` in the ISR handler by themselves. You can evaluate which method is more applicable to your application. 

Your workaround could also work. But usually this needs to be handled in the ISR (and re-configure interrupt only is easier than re-configure all IO settings), so that it guarantees the level-triggered interrupt only trigger once.

filipbudisa, 2023-07-06:
Thank you very much. Setting the intr type to pos-edge in the handler resolved the issue.

closed 2023-07-06 by filipbudisa