espressif/esp-idf #11688: A2DP sink apll not synced

mmar22, 2023-06-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I test examples and read other issues over A2DP to I2S or SPDIF.
Every writes about underrun overun buffers on I2S DMA , because data speed isnt synced with apll out. I ask why this elementary not solved?
Some codes solve it with add or drop samples, what is one technique for async data. Second is triming pll clock.

But IDF or no document for howto change fixed_mclk on the fly and how is resolition? **Can here be showed code for** set -+ 1Hz or what is minimum in I2S config fixed_mclk.

L-KAYA, 2023-06-19:
Sorry but I don't really understand what you do mean by **Not Sync**. Do you mean the actually sample rate of I2S is not match what you set in the configuration?

Also not sure which IDF version you're using, but I guess probably v4.4, right? Here are some tips maybe can help you:

1. As you can see in the document, `fixed_mclk` only takes effect when `use_apll = true and fixed_mclk > 0`. I2S in `A2DP sink` example doesn't use APLL by default, you need to config it manually in [i2s_config_t](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/bt_app_av.c#L159)
2. The minimum supported frequency of APLL is `5303031 Hz` on ESP32, and the maximum division of `mclk` is `255` so we can get the minimum `mclk = 5303031 Hz / 255 = 20.797 kHz` (i.e. minimum `fixed_mclk = 20797`).
3. There is no such 'Adjustment' function for APLL to adjust the frequency in runtime, because APLL is shared by multiple peripherals, adjust its frequency in runtime might affect other peripherals.
4. If you're talking about the I2S data dropped, you may want to change the `dma_buf_count` and `dma_buf_len`. Generally, increase these two field can hopefully avoid the data dropping, for the detail of how to calculate a proper value for them, please refer to the [Application Note](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/peripherals/i2s.html#application-notes) section in I2S programming guide.

L-KAYA, 2023-06-19:
Could you take some examples to describe the issue if possible. And if you have found similar issues, you can also paste that issue as well.

mmar22, 2023-06-19:
Your reply is as from GPT.

1. Yes i enable it and is only one periph use it.

2. I ask minimum step for change freq ontime func exist = periph_rtc_apll_freq_set ...

3. at 1

4. buff is irelevant we have source data speed1 from A2DP and i2s out data speed2
if speed1>speed2 overrun buff
if speed1<speed2 underrun buff

Method for handle this is

A. skip/add sample

B. regulate speed

MM

Example for A: git

```
size_t write_ringbuf(const uint8_t *data, size_t size)
{
    // rate control
    UBaseType_t items;
    vRingbufferGetInfo(s_ringbuf_i2s, NULL, NULL, NULL, NULL, &items);
    if (items < RINGBUF_SIZE * 3 / 8) {
        xRingbufferSend(s_ringbuf_i2s, (void *)data, AUDIO_SAMPLE_SIZE, (portTickType)portMAX_DELAY);
    } else if (items > RINGBUF_SIZE * 5 / 8) {
        size -= AUDIO_SAMPLE_SIZE;
    }

```
for B i need change apll freq in ifs.

Thats all folks

mmar22, 2023-06-19:
Or ofcourse most perfect is synced speed1==speed2, is this if dont used APLL?

L-KAYA, 2023-06-19:
I got what you mean, so the issue is that, the speed that A2DP feed into the ringbuffer mismatch with the speed that I2S takes from the ringbuffer, which cause the ringbuffer overflow or underflow, right?

I haven't met such issue before maybe I didn't run this example for a long time, the slight difference of the data rate not appear to be the overflow/underflow issue. But generally it's not an APLL issue.

Actually, we shouldn't think how to match these two rates at they  must have difference as they don't share a same clock signal to synchronize the data rate. Instead, we need to add flow control strategy to slow down or speed up the A2DP rate, to ensure a continuous audio output and a limited data amount in the ringbuffer.

mmar22, 2023-06-23:
Aka you dont have strategy now, i can try trim APLL, but need info howto do this without corrupt i2s runned stream.
Func usble for this is 
`periph_rtc_apll_freq_set()`
or inner better direct
`rtc_clk_apll_coeff_set()`

Can you give info for best testing ?

mmar22, 2023-06-23:
I implement coeff calc func in my arduino platformio project, but need include
rtc_clk_apll_coeff_set() or ll rtc func  and cant handle this. 
Can you help me?

And additional info from testing on my board and phone my buffer overrun , that signal spdif I2S is slower as required. This too agree with mclk calc coef func. For requested 22 579 200 MHz coef returned is for lower freq...

mmar22, 2023-06-24:
> slight difference of the data rate not appear to be the overflow/underflow issue.

@L-KAYA every diference produce over or under issue, but time to arive and repeat changes.

I test A example code , but have trouble detect buff level overflow or underflow moment. If xringbuffer is filled per samples technique is simple, but if is filled by packets is hard to detect. Have you an idea?