espressif/esp-idf #11691: The mesh can still use no oob after static OOB is used

2815918642, 2023-06-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Bluetooth mesh can still use no oob after static OOB is used. Can I disable no oob?

2815918642, 2023-06-17:
Bluetooth mesh can still use ON OOB after using static OOB. Can I cancel ON OOB?

gnameg, 2023-06-30:
Can you give more details on that?

gnameg, 2023-07-03:
If a static OOB value is available, then this value shall be included as part of the confirmation value. If no 
static OOB value is available, then this value shall have a value of zero.
[image]

2815918642, 2023-07-03:
> If a static OOB value is available, then this value shall be included as part of the confirmation value. If no如果静态OOB值可用，则应将该值作为确认值的一部分包括在内。如果没有 static OOB value is available, then this value shall have a value of zero.静态OOB值可用，则该值应为零。 [image]

thank you for response,Can I cancel no oob when static oob is in use ? When configuring the network, do not display the available no oob configuration.

gnameg, 2023-07-03:
> thank you for response,Can I cancel no oob when static oob is in use ? When configuring the network, do not display the available no oob configuration.

Did you set the **static_val** field of the **esp_ble_mesh_prov_t** to a correct address, which pointed to your static OOB value, when initializing? If **static_val** was **NULL**, it would probably mean that device supports No OOB when provisioning.

2815918642, 2023-07-03:
> > thank you for response,Can I cancel no oob when static oob is in use ? When configuring the network, do not display the available no oob configuration.感谢您的回复，当使用静态OOB时，我可以取消任何OOB吗？配置网络时，不要显示可用的无OOB配置。
> 
> Did you set the **static_val** field of the **esp_ble_mesh_prov_t** to a correct address, which pointed to your static OOB value, when initializing? If **static_val** was **NULL**, it would probably mean that device supports No OOB when provisioning.在初始化时，是否将esp_ble_esh_prov_t的Static_val字段设置为指向您的静态OOB值的正确地址？如果STATIC_VAL为NULL，则可能意味着设备在进行资源调配时不支持OOB。

I have successfully configured STATIC OOB ，However, the 【static oob and no oob】 options are displayed when the network is configured。can retain 【static oob】 and cancel the【 no oob】 configuration？

gnameg, 2023-07-05:
> I have successfully configured STATIC OOB ，However, the 【static oob and no oob】 options are displayed when the network is configured。can retain 【static oob】 and cancel the【 no oob】 configuration？

Sorry，i'm not sure where the 【static oob and no oob】options come. Can you show some pictures or logs?

2815918642, 2023-07-06:
> 
[image]

2815918642, 2023-07-06:
> > > > thank you for response,Can I cancel no oob when static oob is in use ? When configuring the network, do not display the available no oob configuration.感谢您的回复，当使用静态OOB时，我可以取消任何OOB吗？配置网络时，不要显示可用的无OOB配置。
> > > 
> > > 
> > > Did you set the **static_val** field of the **esp_ble_mesh_prov_t** to a correct address, which pointed to your static OOB value, when initializing? If **static_val** was **NULL**, it would probably mean that device supports No OOB when provisioning.在初始化时，是否将esp_ble_esh_prov_t的Static_val字段设置为指向您的静态OOB值的正确地址？如果STATIC_VAL为NULL，则可能意味着设备在进行资源调配时不支持OOB。在初始化时，是否将esp_ble_esh_prov_t的Static_val字段设置为指向您的静态OOB值的正确地址？如果STATIC_VAL为NULL，则可能意味着设备在进行资源调配时不支持OOB。在初始化时，是否将esp_ble_esh_prov_t的Static_val字段设置为指向您的静态OOB值的正确地址？如果STATIC_VAL为NULL，则可能意味着设备在进行资源调配时不支持OOB。
> > 
> > 
> > I have successfully configured STATIC OOB ，However, the 【static oob and no oob】 options are displayed when the network is configured。can retain 【static oob】 and cancel the【 no oob】 configuration？我已经成功配置了静态OOB，但是在配置网络时，会显示[Static OOB and no OOB]选项。是否可以保留[静态OOB]并取消[无OOB]配置？
> 
> Sorry，i'm not sure where the 【static oob and no oob】options come. Can you show some pictures or logs?抱歉，我不确定[静态OOB和无OOB]选项从何而来。你能给我看一些照片或日志吗？

use app nrf mesh  I don't want the NO OOB option to show up.

gnameg, 2023-07-06:
> > 
> 
> [image]

For this, you should perhaps look for help on [Android-nRF-Mesh-Library](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library)

2815918642, 2023-07-06:
> > > 
> > 
> > 
> > [image]
> 
> For this, you should perhaps look for help on [Android-nRF-Mesh-Library](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library)为此，您可能应该在Android-NRF-Mesh-Library上寻求帮助

OK, thank you very much. Does esp32C2 support Bluetooth mesh? The official example does not work properly on C2.

2815918642, 2023-07-06:
> > > 
> > 
> > 
> > [image]
> 
> For this, you should perhaps look for help on [Android-nRF-Mesh-Library](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library)为此，您可能应该在Android-NRF-Mesh-Library上寻求帮助

OK, thank you very much. Does esp32C2 support .Bluetooth mesh? The official example does not work properly on C2.

closed 2023-07-06 by 2815918642