espressif/esp-idf #11693: CMake Error: The source directory "xxx" does not appear to contain CMakeLists.txt.

TritiumXs, 2023-06-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-rc1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I want to build a firmware,but it doesn't work

### What is the actual behavior?

Errors popped up about cmake and nanja
The sample code can also be compiled initially
But when I tried to fix it, things just got worse

The environment I am using is VSCode
Some codes cannot be used there but can be used in powershell

### Steps to reproduce.

I don't know what exactly I did, when I modify the code it doesn't compile...

### Build or installation Logs.

```plain
CMake Error: The source directory "E:/Espressif/project/hello_world" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```

### More Information.

Some code like example can be compiled in powershell

dobairoland, 2023-06-19:
> CMake Error: The source directory "E:/Espressif/project/hello_world" does not appear to contain CMakeLists.txt.

Please check the directory content. Probably it is not a proper ESP-IDF project. You might try the hello_world which is in the ESP-IDF's examples/get-started directory.

Also I recommend to run `git status` in the ESP-IDF directory to see if you've made any changes which is causing the above issue.

TritiumXs, 2023-06-21:
> > CMake 错误：源目录“E：/Espressif/project/hello_world”似乎不包含 CMakeLists.txt。
> 
> 请检查目录内容。可能这不是一个合适的 ESP-IDF 项目。您可以尝试 ESP-IDF 的示例/入门目录中的hello_world。
> 
> 此外，我建议在 ESP-IDF 目录中运行，看看您是否进行了导致上述问题的任何更改。`git status`

I found a project on the web. The project is about letting words display on the OLED screen. After I use "u8g2.h", this problem appeared.  I guess u8g2 should not be used here (although it is ported to esp32)
But when I deleted it all, ESP-IDF has such a problem

closed 2023-11-17 by TritiumXs