espressif/esp-idf #11696: the esp32 g++ internal error when build OpenCV

cosmosjiang, 2023-06-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3797-gaf805df3cb

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

When I build OpenCV with the esp32 toolchain, xtensa-g++ reports the internal bug. I create a tiny project contains the OpenCV's preprocessed file to reproduce this bug.

[opencv-bug.tar.gz](https://github.com/espressif/esp-idf/files/11782068/opencv-bug.tar.gz)

### What is the actual behavior?

xtensa-esp32-elf-g++ reports internal IR bug

### Steps to reproduce.

I put the source code which can reproduce this bug to github: https://github.com/cosmosjiang/esp32-opencv-bug

### Build or installation Logs.

```plain
chessboard.cpp:482:1: error: insn does not satisfy its constraints:
482 | }
    | ^
(insn 1117 528 1420 55 (set (reg/v:SF 21 f2 [orig:201 val ] [201])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC1487") [flags 0x2]) [0  S4 A32])) "chessboard.cpp":255:16 49 {movsf_internal}
     (nil))
during RTL pass: postreload
chessboard.cpp:482:1: internal compiler error: in extract_constrain_insn, at recog.cc:2692
0x7f85b0abdd8f __libc_start_call_main
	../sysdeps/nptl/libc_start_call_main.h:58
0x7f85b0abde3f __libc_start_main_impl
	../csu/libc-start.c:392
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
```

jcmvbkbc, 2023-06-18:
This is a known issue in the xtensa gcc port, see https://gcc.gnu.org/bugzilla/show_bug.cgi?id=98470
You may try to build the source that triggers the issue with the compiler option `-fno-if-conversion` that turns off one particular optimization pass.

cosmosjiang, 2023-06-18:
after add -fno-if-conversion, I can build OpenCV now.
Thanks.

closed 2023-06-19 by igrr