espressif/esp-idf #11697: ESP-NOW: allow maximum number of encrypted peers to be set at runtime

anecdata, 2023-06-18:
### Describe the solution you'd like.

> The number of hardware keys for encryption is fixed. And the espnow and SoftAP share the same hardware keys. So this configuration will affect the maximum connection number of SoftAP. Maximum espnow encrypted peers number + maximum number of connections of SoftAP = Max hardware keys number.

[link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/kconfig.html#config-esp-wifi-espnow-max-encrypt-num)

Currently, [CONFIG_ESP_WIFI_ESPNOW_MAX_ENCRYPT_NUM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/kconfig.html#config-esp-wifi-espnow-max-encrypt-num) can be 0-17, but it is a build-time option, whereas the maximum number of connections of SoftAP is a [runtime option](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/network/esp_wifi.html#_CPPv4N16wifi_ap_config_t14max_connectionE), 0-10 presumably under the assumption of ESP-NOW's default 7.

Solution: have a runtime API for maximum espnow encrypted peers, so that with a given firmware, user code (e.g., MicroPython, CircuitPython) can configure SoftAP and ESP-NOW as appropriate for their applications.

zhangyanjiaoesp, 2023-06-21:
@anecdata I think it's enough to set the ESP-NOW encrypt num in  the menuconfig for the customer's applications. Do you have any application scenarios where you need to change the encryption number of ESPNOW at any time？

anecdata, 2023-06-21:
Yes. In the case of CircuitPython and MicroPython (and perhaps other environments), a single firmware build is used by most customers for each particular board. Those customers write their applications in Python and don't have access to menuconfig or the esp-idf build environment. Instead, the esp-idf APIs of interest get exposed to a Python library for those customers to use in their applications. Some of these customers may favor more Wi-Fi Access Point connections. Some of these customers may favor more than 7 ESP-NOW encrypted pairs. Some may not need either. Each customer has unique application requirements and needs the flexibility to implement features at runtime from the same build.

zhangyanjiaoesp, 2024-03-05:
@anecdata Since we won't consider to add a new API,  you can set a large number of encrypted peers in the menuconfig, and use an API at the application layer to limit the number that customers can set.

anecdata, 2024-03-05:
But is it possible to set `CONFIG_ESP_WIFI_ESPNOW_MAX_ENCRYPT_NUM` = 17 at build time, and also have  allow the maximum of 15 stations for an AP at build time?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/wifi.html#ap-basic-configuration

The goal would be to support for the customer any values within the maximum ranges, as long as the sum of the two aren't more than the number of encryption keys:
• encrypted ESP-NOW pairs: 0-17
• stations connecting to an AP: 0-15

For example, using the same build, one customer may want all 17 ESP-NOW encrypted pairs and 0 AP stations. A second customer may want all 15 AP stations and can live with <= 2 ESP-NOW encrypted pairs. A third customer may want 12 ESP-NOW encrypted pairs and a max of 5 AP stations.

zhangyanjiaoesp, 2024-03-06:
ok, we will consider this.

zhangyanjiaoesp, 2024-11-08:
After further consideration, we have decided not to implement this API. Allowing the number of SoftAP and ESP-NOW  encryption key counts to be dynamically configured at any time could impact the STAs already connected to the SoftAP.

anecdata, 2024-11-08:
I ask that this please be reconsidered.

There are many ways user code can cause problems for itself. But in this case, at least two major ecosystems using Espressif esp-idf, namely MicroPython and CircuitPython, would benefit from allowing their users to make the appropriate compromise in their own applications for the number of SoftAP vs ESP-NOW encryption keys. This would typically be done at the start of Python code. But it can't be done at all currently without re-compiling these interpreted environments.

It doesn't seem to be a major risk, since code could also determine the number of connected stations or ESP-NOW connections if for some reason the application wanted to change the encryption key allocations at some time after start-up.

tyeth, 2025-01-29:
> After further consideration, we have decided not to implement this API. Allowing the number of SoftAP and ESP-NOW encryption key counts to be dynamically configured at any time could impact the STAs already connected to the SoftAP，as well as disrupt existing ESP-NOW connections.

What if you just safeguard the new API with "the interfaces must be down not up to reconfigure", then it's less ideal but guaranteed safe, and any downstream users will already be forced to reconnect after reconfiguration and bring up of the interfaces