espressif/esp-idf #11698: Flash write data exception during OTA upgrade

BurnerK, 2023-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C2

### Power Supply used.

External 5V

### What is the expected behavior?

When OTA upgrade, flash write data is the same as read data

### What is the actual behavior?

[image]
When OTA upgrade, calculate the checksum value before writing data to flash, and then read it out after writing to calculate the checksum value, the two are not consistent.
and each time an exception occurs from the offset address 65536.
File transfer protocol is Xmodem-1k.

### Steps to reproduce.

1. Trigger xmodem-1k upgrade process
2. Preparation before transferring files
```
    /* TODO:擦除升级固件所在的所有页 */
    /* 获取当前系统下一个（紧邻当前使用的OTA_X分区）可用于烧录升级固件的Flash分区 */
    partition_data_ota = esp_ota_get_next_update_partition(NULL);
    if (NULL == partition_data_ota)
    {
        HAL_xmodemDebugPrint("esp_ota_get_next_update_partition fail!!!\r\n", strlen("esp_ota_get_next_update_partition fail!!!\r\n"));
    }

    if (esp_ota_begin(partition_data_ota, OTA_SIZE_UNKNOWN, &update_handle) != 0) //启动OTA模块要进行烧写
    {
        HAL_xmodemDebugPrint("esp_ota_begin failed\r\n", strlen("esp_ota_begin failed\r\n"));
    }

    int err = esp_partition_erase_range(partition_data_ota, 0, partition_data_ota->size);
    printf("err = %d\r\n", err);

    printf("size = %ld. addr = %ld", partition_data_ota->size, partition_data_ota->address);
```
3. Flow of writing to flash
```
int32_t HAL_flashWrite(uint8_t *buf, uint32_t len,uint32_t offset)
{
    uint8_t src[1024] = { 0 };
    int32_t err;
    printf("w offset :%ld\r\n", offset);
    // err = flash.ota_write(update_handle, buf, len);
    esp_partition_erase_range(partition_data_ota, (offset / 4096) * 4096 - partition_data_ota->address, 4096);
    err = esp_ota_write_with_offset(update_handle, buf, len, offset);
    printf("w err :%ld\r\n", err);
    err = esp_partition_read(partition_data_ota, offset, src, len);
    printf("R err :%ld\r\n", err);
    write_crc = xmodemCrc16(src, len);
    printf("R crc :%04X\r\n", write_crc);
    esp_rom_md5_update(&md5, src, len);

    return err;
}
```
4. When HAL_flashWrite() passes in an offset real parameter of 65536, the checksum value calculated by xmodemCrc16() is different from the one before writing

hmalpani, 2023-06-23:
Hello @BurnerK 
Can you share the code to reproduce this issue? Please Also share the sdkconfig file you are using when building the application.

BurnerK, 2023-06-23:
> Hello @BurnerK Can you share the code to reproduce this issue? Please also share the sdkconfig file you are using when building the application.

Thanks, the problem has been solved, it's the rest of the code that takes up the flash address.

closed 2023-06-25 by BurnerK