espressif/esp-idf #11700: I ran zigbee esp_zigbee_gateway this routine but no such file #include “zboss api mm.h”

Calm468, 2023-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf v5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

Hopefully, the routine“esp_zigbee_gateway” builds successfully。

### What is the actual behavior?

I ran zigbee esp_zigbee_gateway this routine in idf5.01 and then ordered the official dependency library, but now the error appears that this header file cannot be found.

### Steps to reproduce.

1. No routines are modified, the build fails
2. download espressif/esp-zboss-lib v 0.4.0 
3. found not such files.

### Debug Logs.

```plain
*  正在执行任务: ninja  

[1/10] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/esp_zigbee_gateway.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/esp_zigbee_gateway.c.obj 
F:\5.0.1ESP-IDF\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DRCP_COMPILE_DEFINE="\"zigbee-a4afa44435--0.5.0-2023-06-16 03:46:28 UTC\"" -DUNITY_INCLUDE_CONFIG_H -IC:/Users/Administrator/Desktop/esp_zigbee_gateway/build/config -IC:/Users/Administrator/Desktop/esp_zigbee_gateway/main -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/newlib/platform_include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/freertos/esp_additions/include/freertos -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/freertos/esp_additions/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_hw_support/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_hw_support/include/soc -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_hw_support/include/soc/esp32 -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_hw_support/port/esp32/. -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_hw_support/port/esp32/private_include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/heap/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/log/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/soc/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/soc/esp32/. -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/soc/esp32/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/hal/esp32/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/hal/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/hal/platform_port/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_rom/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_rom/include/esp32 
-IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_rom/esp32 -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_common/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_system/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_system/port/soc -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_system/port/include/private -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/xtensa/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/xtensa/esp32/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/lwip/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/lwip/include/apps -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/lwip/include/apps/sntp -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/lwip/lwip/src/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/lwip/port/esp32/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/lwip/port/esp32/include/arch -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_ringbuf/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/efuse/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/efuse/esp32/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/driver/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/driver/deprecated -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/driver/esp32/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_pm/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/mbedtls/port/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/mbedtls/mbedtls/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/mbedtls/mbedtls/library -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/mbedtls/esp_crt_bundle/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_app_format/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/bootloader_support/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/bootloader_support/bootloader_flash/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_partition/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/app_update/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/spi_flash/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/pthread/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_timer/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/app_trace/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_event/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/nvs_flash/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_phy/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_phy/esp32/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/vfs/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_netif/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/wpa_supplicant/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/wpa_supplicant/port/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/wpa_supplicant/esp_supplicant/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_wifi/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/unity/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/unity/unity/src -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/cmock/CMock/src -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/console -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/http_parser -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp-tls -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp-tls/esp-tls-crypto -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp-zboss-lib/include 
-IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_adc/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_adc/interface -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_adc/esp32/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_adc/deprecated/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_eth/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_gdbstub/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_gdbstub/xtensa -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_gdbstub/esp32 -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_hid/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/tcp_transport/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_http_client/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_http_server/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_https_ota/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_lcd/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_lcd/interface -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/protobuf-c/protobuf-c -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/protocomm/include/common -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/protocomm/include/security -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/protocomm/include/transports -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_local_ctrl/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/esp_psram/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/espcoredump/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/espcoredump/include/port/xtensa -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/wear_levelling/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/sdmmc/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/fatfs/diskio -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/fatfs/vfs -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/fatfs/src -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/idf_test/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/idf_test/include/esp32 -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/ieee802154/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/json/cJSON -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/mqtt/esp-mqtt/include -IF:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/perfmon/include -IF:/5.0.1ESP-IDF/ [... cut ...]
[1/1] cmd.exe /C "cd /D C:\Users\Administrator\Desktop\esp_zigbee_gateway\build\bootloader\esp-idf\esptool_py && f:\5.0.1ESP-IDF\python_env\idf5.0_py3.8_env\Scripts\python.exe F:/5.0.1ESP-IDF/esp-idf-v5.0.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/Administrator/Desktop/esp_zigbee_gateway/build/bootloader/bootloader.bin"
Bootloader binary size 0x6710 bytes. 0x8f0 bytes (8%) free.
ninja: build stopped: subcommand failed.

 *  终端进程“C:\windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja ”已终止，退出代码: 1。
```

### More Information.

no

xieqinan, 2023-06-20:
Hi @Calm468,
By default, the Zigbee gateway depends on espressif/esp-zboss-lib (v0.0.6) in esp-idf v5.0.1.

The espressif/esp-zboss-lib v0.4.0 is too new to be compatible with the Zigbee example in esp-idf v5.0.1.

It may be valid to switch the espressif/esp-zboss-lib from v0.4.0 to v0.0.6.

I recommend using the default esp-zboss-lib version for the Zigbee example in every esp-idf version.

chshu, 2023-08-31:
@Calm468 Please reopen if you still see the compile issue with the match versions.

closed 2023-08-31 by chshu