espressif/esp-idf #11701: Esp_base_mac_addr_set() does not handle overflow when incrementing base mac address for interfaces.

mm66, 2023-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.3-86-g7988aec391

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I have a mac address that looks like this - **CC:CC:CC:CC:41:FF**
If this address is set as the base mac address with **esp_base_mac_addr_set()** then I expect the  BLE mac address to be **CC:CC:CC:CC:42:01** 

### What is the actual behavior?

Instead, reading the BLE mac address gives the following mac - **CC:CC:CC:CC:41:01** . The last byte increments correctly but the byte before does not, it is 41 but i expect it to be 42. 

### Steps to reproduce.

Set base mac address with an address that will overflow by incrementing the last byte and read the derived by incrementation mac addresses from interfaces.

igrr, 2023-06-19:
@mm66 Could you please add, which chip are you using?

mm66, 2023-06-20:
@igrr ESP32-D0WD-V3 (revision 3)

igrr, 2023-06-20:
@mm66 After checking with our team responsible for production, for ESP32 products we shouldn't have chips with MAC address ending in `41:FF`. If your chip has such a MAC address, this indicates either a production mistake or an eFuse corruption.

Could you please run `espefuse.py --port PORT_NAME summary` and share the complete output you get?

mm66, 2023-06-20:
@igrr You misunderstood me. To clarify we are setting our own base mac addresses for production purposes, so it is not about the mac address that is burnt on the chip. So it is an issue for mass production since there would be overlap in mac addresses due to overflow in incrementing the base mac address. So we set the address that ends in 41:FF on purpose expecting it would increment to 42:01 for BLE mac address.

igrr, 2023-06-20:
I'm sorry for misunderstanding what you meant!
How many universal MAC addresses are you reserving for each device? For example, do you assign addresses such as 41:00, 41:04, 41:08, ... or 41:00, 41:02, 41:04, ...?
Also, what is the value of `CONFIG_ESP32_UNIVERSAL_MAC_ADDRESSES` value in sdkconfig? (Should be either CONFIG_ESP32_UNIVERSAL_MAC_ADDRESSES_FOUR or CONFIG_ESP32_UNIVERSAL_MAC_ADDRESSES_TWO).

KonstantinKondrashov, 2023-06-21:
Hi @mm66!
Actually, it is not a bug.  We have a note for [ESP32_UNIVERSAL_MAC_ADDRESSES](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/port/esp32/Kconfig.mac#L10) which says the following:
> ... If the number of universal MAC addresses is four, all four interfaces (WiFi station, WiFi softap,
        Bluetooth and Ethernet) receive a universally administered MAC address. These are generated
        sequentially by adding 0, 1, 2 and 3 (respectively) to **the final octet** of the base MAC address....

It is expected behavior that we add 0, 1, 2, 3 to only **the final octet**. Probably we need to make it clear in the doc.

Starting from v5.1, we have [esp_iface_mac_addr_set()](https://github.com/espressif/esp-idf/blob/release/v5.1/components/esp_hw_support/include/esp_mac.h#L174) that allows changing MAC address for every interface. So it will help for your case but as I can see you use the v4.3. We can not backport features to the old versions and note that the service period is ended.

4 MACs per device:
```
CC:CC:CC:CC:00:00
CC:CC:CC:CC:00:04
CC:CC:CC:CC:00:08
CC:CC:CC:CC:00:0c
CC:CC:CC:CC:00:10
...
CC:CC:CC:CC:00:f0
CC:CC:CC:CC:00:f4
CC:CC:CC:CC:00:f8
CC:CC:CC:CC:00:fc
CC:CC:CC:CC:01:00
CC:CC:CC:CC:01:04
CC:CC:CC:CC:01:08
```

You should not start to give MAC addresses from random offsets. You must start at an address that is a multiple of 4 then everything gets correct. It is how we are doing on our side.

mm66, 2023-06-21:
I understand, thank you for a quick response :)

closed 2023-06-26 by KonstantinKondrashov