espressif/esp-idf #11702: Example for ESP32-S3 as HID Host communicate with HID device

shlomo-edu, 2023-06-19:
I have a logger that uses HID USB to communicate with a PC via a serial cable using a proprietary protocol (raw data).
I want to us ESP32-S3 as a bridge to connect the logger to a PC via BLE.
Currently, the example for HID that is in the examples is for a keyboard and a mouse and doesn't work for raw protocol (see the description of the bug after the image).

[image]

I connected my HID device to the ESP232-S3 and run the example.
The device sends data in a loop.

In: app_main
In the “do” loop after line 468:  EventBits_t event = xEventGroupWaitBit
I added a log line: ESP_LOGI(TAG, "xEventGroupWaitBits %d",(int) event);
And what I see in the terminal: every second 1 line of:
I (525846) example: xEventGroupWaitBits 0
So I assume it didn’t see the any incoming data.

shlomo-edu, 2023-06-19:
I have more info on this problem.
Now I see that when I connect the device to the ESP32 the terminal (the VS Code terminal) shows an error:
ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
I search the code but I didn't find this error in the code.

roma-jam, 2023-06-19:
Hi @shlomo-edu, 

Thanks for the opening the ticket. 

May I ask you to provide some more additional information, as: 
1. Full debug log during the connection of your HID device to the ESP. 
2. About the second error `ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))`. It seems that the problem is with connection to the serial output device (either /dev/ttyX on Unix or COMx on Win). To solve this it is better to check the correction of the port name or verify the access settings to the port. 

Thanks

shlomo-edu, 2023-06-19:
At the end of this post I added the log of the setup.
Please note:
1. When I add log lines to the hid_host_install(...) in hid_host.c (line 880)
The log lines are displayed till the: HID_ENTER_CRITICAL()
But then it shows this log lines
abort() was called at PC 0x403773a3 on core 0
0x403773a3: lock_acquire_generic at C:/Users/shlomoa/esp/esp-idf/components/newlib/locks.c:130
and many more log lines, and then it restarts and again (in endless loop of restarts).
I don't get this problem (or restarts) if I don't add the log lines.

The following lines are the setup log lines when I don't add the log lines to the hid_host_install(...) function.

PS C:\Users\shlomoa\hid> set IDF_PATH=C:\Users\shlomoa\esp\esp-idf
PS C:\Users\shlomoa\hid> C:\Users\shlomoa\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\shlomoa\esp\esp-idf\tools\idf_monitor.py -p COM9 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 c:\Users\shlomoa\hid\build\hid.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM9 instead...
--- idf_monitor on \\.\COM9 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1658
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2d9c
entry 0x403c9900
I (24) boot: ESP-IDF v5.0.2 2nd stage bootloader
I (25) boot: compile time 13:27:13
I (25) boot: chip revision: v0.1
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0ce2ch ( 52780) map
I (102) esp_image: segment 1: paddr=0001ce54 vaddr=3fc92200 size=023b8h (  9144) load
I (104) esp_image: segment 2: paddr=0001f214 vaddr=40374000 size=00e04h (  3588) load
I (110) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=24200h (147968) map
I (144) esp_image: segment 4: paddr=00044228 vaddr=40374e04 size=0d368h ( 54120) load
I (162) boot: Loaded app from partition at offset 0x10000
I (162) boot: Disabling RNG early entropy source...
I (174) cpu_start: Pro cpu up.
I (174) cpu_start: Starting app cpu, entry point is 0x40375228
0x40375228: call_start_cpu1 at C:/Users/shlomoa/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (188) cpu_start: Pro cpu start user code
I (188) cpu_start: cpu freq: 160000000 Hz
I (188) cpu_start: Application information:
I (191) cpu_start: Project name:     hid
I (196) cpu_start: App version:      1
I (200) cpu_start: Compile time:     Jun 19 2023 13:25:06
I (206) cpu_start: ELF file SHA256:  4cb8b40f805434a4...
I (212) cpu_start: ESP-IDF:          v5.0.2
I (217) cpu_start: Min chip rev:     v0.0
I (222) cpu_start: Max chip rev:     v0.99 
I (227) cpu_start: Chip rev:         v0.1
I (231) heap_init: Initializing. RAM available for dynamic allocation:
I (239) heap_init: At 3FC95020 len 000546F0 (337 KiB): DRAM
I (245) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (252) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (258) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (265) spi_flash: detected chip: gd
I (268) spi_flash: flash io: dio
W (272) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size 
in the binary image header.
I (286) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (306) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (306) example: HID HOST example

roma-jam, 2023-06-19:
Hi @shlomo-edu, 

There are couple of thing that you need to consider.
1. When you want to print anything from the critical section, you need to use not blocking version of `ESP_LOGx` which is `ESP_EARLY_LOGx;`. `ESP_LOGx` is a blocking API, so we'll usually get an assert you attempt to block in a critical section.
2. For the HID driver the device should have a HID class device value in USB device descriptor. Otherwise, there will be an error such as `Device at addr x has no HID Interface`, where x - is a device address on USB line. 

To proceed, may I ask you to provide a description of USB descriptor from the device you are trying to attach?

Thanks

shlomo-edu, 2023-06-20:
Hi Roma,
Thanks for helping me cross this bridge.
Yes, the device class has an HID descriptor.
Here is how it is presented in YAT (a terminl) application.
Please note that I see that "client_event_cb()" (in hist_hid.c line 550) is never called.
[image]

shlomo-edu, 2023-06-20:
More on this.
I see in the code that the only functions that change the xEventGroupWaitBits, are:
test_hid_wait_connection_and_install()
and
test_hid_wait_for_removal()
So I added, log lines in the beginning of the functions.
I see that it never gets there.
I don't know what else to check!!!

roma-jam, 2023-06-20:
Hi @shlomo-edu, 

Thanks for the additional information. 

Meanwhile, to proceed I need to ask you the following: 

1. Which board do you use? Probably, the problem can be not related to the HID or USB
2. To verify USB connection in general, you can use the `usb_host_lib` [example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib)
3. To be more specific, it is better to have a device descriptor, which can be obtained either from system information or with any external application (for example, you can use this one: [Thesycon USB Descriptor Dumper](https://www.thesycon.de/eng/usb_descriptordumper.shtml) )

Thanks

shlomo-edu, 2023-06-21:
1. ESP32-S3-DevKitM 1 v1.0
2. My programmer used the usb_host_lib and it didn't show anything.
3. Following is the description of this HID usb from the Thesycon" (tdd.exe).
This means that this program can also interface with our device (as well as YAT).
So now the question is why the ESP32-S3 can't?
__________________________________________________________________________

Information for device EC8xx III (VID=0x10C4 PID=0x1F0C):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0015
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x10C4	idVendor
0x1F0C	idProduct
0x0200	bcdDevice
0x01	iManufacturer   "Fourtec"
0x02	iProduct        "EC8xx III"
0x03	iSerialNumber   "0E5139433135"
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xC0	bmAttributes   (Self-powered Device)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0021	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x20	bInterval         (32 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x01	bEndpointAddress  (OUT endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x20	bInterval         (32 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Fourtec"
0x02   0x0409  "EC8xx III"
0x03   0x0409  "0E5139433135"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
EC8xx III (VID=0x10C4 PID=0x1F0C) Port: 3

Running on: Windows 10 or greater (Build Version 19044)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02

roma-jam, 2023-06-21:
@shlomo-edu , 

Thanks for sharing additional information. 

1. With that board there is no power supply (+5 V) for the usb otg port. To make it work you need to deliver a voltage supply of +5 V to the USB Device. It could be done with external wire. 

2. The descriptor looks fine, it is a HID comatible device without any protocol. 
The HID driver will treated such device with HID_PROTOCOL_NONE. 

Both example should work (HID example then need a small change) after power will be delivered. 

Thanks

shlomo-edu, 2023-06-21:
Hi Roma,
Redarding the usb_host_lib
I tried to build it on my my computer and it failed because it didn't find the "usb/usb_host.h".

shlomo-edu, 2023-06-21:
The idea that the problem is the power makes sence, but -
The serial port is connected to my laptop that gives a power.
I see that the software works (I see the log lines in the VSC terminal).
But, I see that the led is RED.
Maybe it is an indication of a problem!!!???

roma-jam, 2023-06-21:
Hey @shlomo-edu, 

1. To be able to use usb component you need to set terget to esp-idf. This can be done with `idf.py set-target esp32s2` (or esp32s3, depending on a board)
2. I mean that there are two usb ports on a board. The one, which is used for programming is power the esp32s2 and the board itself. 
But for the second port (to which the another USB Device is attaching) needs to be powered additionally. 

Thanks

shlomo-edu, 2023-06-21:
OK, I think that you are on the right path to solve the problem.
How do I power it?

shlomo-edu, 2023-06-21:
You were right. I added 5V to the USB and now it calls is_hid_device() with dev_addr=1
But,
It fails on this line:
if (get_interface_desc_by_proto(config_desc, &offset, HID_PROTOCOL_NONE))

Maybe the reason is that this is the wrong dev_addr, because I see that in the tdd.exe application, the 
Device address: 0x0015 (see in previous post).

roma-jam, 2023-06-21:
Hey @shlomo-edu, 

Glad to hear it works. 
Unfortunately, the ESP32-S3-DevKitM 1 v1.0 board was designed for the USB Device examples, that is why to launch USB Host examples there is a necessary to fix it yourself. 
Just in case, there are several OTG boards, that can easily be used for the USB Host examples without any additional modifications.

Back to the example. 
That is exactly, what we are talked about when we discussed the first version of HID driver. The idea is that it supports only Keyboard and Mice, Boot protocols ([here](https://github.com/espressif/esp-idf/issues/9330#issuecomment-1596701051))

So, to make it work, please feel free to: 
1. Remove `HID_SUBCLASS_BOOT_INTERFACE` comparison statement in the file [hid_host.c, line 282](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/components/hid/hid_host.c#L282)
2. Change the protocol from `HID_SUBCLASS_BOOT_INTERFACE` to `HID_SUBCLASS_NO_SUBCLASS` in the file [hid_host.c, line 1067](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/components/hid/hid_host.c#L1067)
3. Change the comparison `iface_config->proto > HID_PROTOCOL_NONE` to `iface_config->proto >= HID_PROTOCOL_NONE` in the file [hid_host.c, line 966](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/components/hid/hid_host.c#L966)
4. Comment out or remove lines from 986 to 1005 in the file [hid_host.c, line 986](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/components/hid/hid_host.c#L986)
5. Install the device and register an event callback `hid_host_interface_event_callback` during the `DEVICE_ADDRESS_MASK` event in the app task
6. Claim the interface with `hid_host_claim_interface` during the `hid_host_interface_event_callback`. But be careful, in example there is a ternary operator to distinguish the protocol, so change the code here to claim interface with protocol you need `HID_PROTOCOL_NONE`

After that you should be able to get the raw data report callback with params `(const uint8_t *const data, const int length)`, where is data - is a pointer to raw data of the input report and the length - is the number of bytes in this input report. 

There still can be problems during device detaching (such as memory leakage e.t.c), because to fully support HID_PROTOCOL_NONE in the first version there little bot more things need to be done. 

But they are already done in the HID driver, which will be released this week as an idf-extra-component. 

Thanks

roma-jam, 2023-06-21:
Hey @shlomo-edu , 

> I don't understand what is "other OTG boards", what are they, what are the differences (between them and the S3) and why should I avoid using them.
Please elaborate on that.

There are at least two OTG boards from Espressif which easily coulb be used for OTG or USB Host example without any changes: 
- ESP32-S3-USB-OTG
- ESP32-S3 WROOM FREENOVE

> Regarding our example, I started to do the changes, but I don't understand what to do in 5 and 6.

These logic is implemented in the example: 
- For point 5 file [hid_host_example.c, line 484](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c#L484)
- For point 6 file [hid_host_example.c, line 364](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c#L364)
Feel free to refer to it.

> I don't understand the following:
"But they are already done in the HID driver, which will be released this week as an idf-extra-component."
What is the driver, how can I get it, how do I work with it (is there a sample code for it)?

Now, the idf-extra-component is in merging state and is ready to be merged. So it will be available as an idf-extra-component very soon. 

> Is there a complete project that I can install in VCS and use (as is)?

Not yet. After the HID driver idf-extra-component will be released, the HID Host example should be updated to use this component. So, that the next step after component release. 

If there is still question, feel free to ask them.
Otherwise, probably it is better to wait the release of HID Driver idf-extra-component and example with usage of it. 

Thanks

shlomo-edu, 2023-06-22:
Hi Roma,

Is it better to just purchase ESP32-S3-USB-OTG or ESP32-S3 WROOM FREENOVE for my purpose.
What are the adavantages of each of them?
Do you have a sample code to work with them?
Can I use them just as I use the ESP32-S3-DevKit? 

Currently I added the 5V from the PC, If I add a battery to the ESP32, how do I connect it so it will supply the 5V to the USB?
Do you have a complete board with a battery?

shlomo-edu, 2023-06-25:
Hi Roma.
I am still waiting for yoour reply on the ESP32-S3_OTG and ESP32-S3-WROOM?
And also on the question regarding the board with a battery?

I rub the code with some extra log lines:
Please note that the device keep sending raw data every second.

This is the log lines that I receive:
Can you see from the log why I don't receive the chracters?

I (3491) example: app_main: event=16
I (3491) example: Interface number 0, protocol Mouse
I (3491) example: after hid_host_claim_interface
E (3551) USBH: Device 1 gone
I (3551) example: hid_host_event_callback
I (3551) example: app_main: event=a
E (3551) hid-host: hid_host_release_interface(1027): 
E (3551) hid-host: hid_host_release_interface(1027): 
I (3951) example: hid_host_event_callback
I (3951) example: app_main: event=16
I (3951) example: Interface number 0, protocol Mouse
I (3951) example: after hid_host_claim_interface
E (4011) USBH: Device 1 gone
I (4011) example: hid_host_event_callback
I (4011) example: app_main: event=a
E (4011) hid-host: hid_host_release_interface(1027): 
E (4011) hid-host: hid_host_release_interface(1027): 
I (4411) example: hid_host_event_callback
I (4411) example: app_main: event=16
I (4411) example: Interface number 0, protocol Mouse
I (4411) example: after hid_host_claim_interface
E (4471) USBH: Device 1 gone

roma-jam, 2023-06-26:
Hi @shlomo-edu, 
Sorry for the long response. 

> Is it better to just purchase ESP32-S3-USB-OTG or ESP32-S3 WROOM FREENOVE for my purpose.
What are the adavantages of each of them?

That is fully up on you. That is different boards for different purposes, so they have different peripheral on the board. Please, refer to their description to pick one of them that suits your requirements. 
The main idea - there is nothing to modify in them, so you can use them with usb_host examples without any changes. (For the ESP32-S3_OTG board some lines to the example should be added to power up the OTG port, ESP32-S3 WROOM FREENOVE you can use without any changes in the examples code).

> Do you have a sample code to work with them?

Any example from esp-idf works with these boards. 

> Can I use them just as I use the ESP32-S3-DevKit?

Yes, but the power supply should be added to the USB OTG port with an external wire. 

> Currently I added the 5V from the PC, If I add a battery to the ESP32, how do I connect it so it will supply the 5V to the USB?

The same way. The difference will be that the +5 V will be delivered not from PC USB port but from external battery. 

> Do you have a complete board with a battery?

I am not sure. Probably, there is some boards from third party companies, but it is better to search them by yourself, because there is a huge variety of them.

roma-jam, 2023-06-26:
@shlomo-edu, 

> Can you see from the log why I don't receive the chracters?

Unfortunately, there is not enough information. 
The strange thing is that after interface claiming there is a USB dev gone event, which means that the device was removed. 
There should be a cause of that, but that is not how it should be. 

The idf-extra-component has been released last week: https://components.espressif.com/components/espressif/usb_host_hid
The next step will be the example update in the esp-idf master repository, so this version will work with any protocol of HID compatible device and provide an option to receive the raw data input report. 

By the way, do you need the possibility to send some information back to the HID device from the host? 

Thanks

shlomo-edu, 2023-06-26:
Thanks Roma,
I will putchase one of them and I will then add a comment to let you know how it works.
Thank you very much for your help.

shlomo-edu, 2023-06-27:
Hi Roma,

1. I tried to find where I can purchase the  ESP32-S3 WROOM FREENOVE, but could find a relibale source.
Maybe you can help me with that.
2. I found the code for it in GitHub and the email of the company: sale@freenove.com, but they haven't replied. Maybe you can ask them.
3. It looks like this path will take a lot of time till I recieive it, and I am afraid that again it will not work as expected (this will be my fouth purchase of PCB's to solve the problem).
So maybe you can help me make the ESP32-S3-DevKit work with the hid_host_example.c?
I can provide add any log lines that you ask.

roma-jam, 2023-06-27:
Hi @shlomo-edu, 

1 and 2. For this topic is better to use official contacts: https://www.espressif.com/en/contact-us/sales-questions, or you can user esp32 forum https://www.esp32.com/ which is intended for all kind of questions, not only bugs or feature requests, as the github.

3. From the point that this is lead to the code anyway, it is better to use examples 'as is' to get better experience. Examples in esp-idf show a way how the esp-idf can be used and they do not cover all the scenarios. If the option to change the current example in esp-idf for HID_PROTOCOL_NONE seems not reasonable, probably I would suggest then to wait for the example update to get full support of HID devices with any protocol. 

Thanks

roma-jam, 2023-07-06:
Hi @shlomo-edu, 

FYI, there was an example update: https://github.com/espressif/esp-idf/commit/5ceac3cd19f2516b73746239d13c45df30675a99
Now the example uses the idf-extra-component [USB Host HID (Human Interface Device) Driver](https://components.espressif.com/components/espressif/usb_host_hid) and shows how input report can be obtained even with a generic HID device. 

Feel free to try it to get the data from your device. Hope it will help with your case.
In addition, feel free to ask more questions if it helps. Otherwise, if there will be no activity the issue will be closed. 
Thanks

shlomo-edu, 2023-07-10:
Hi Roma,
Thanks, but we can't compile it because it doesn't find the macro: pdMS_TO_TICKS
[image]

roma-jam, 2023-07-10:
Hi @shlomo-edu, 

It seems that it is still the code of the old version of HID example. 
The HID example was updated in master branch and was backported to release v5.1. 
To experience new version of HID example, please use one of these branches. 

Thanks

shlomo-edu, 2023-07-10:
Hi Roma,
Please note that we downloaded the latest brunch.
Some of the files where updated 2 days ago.
Please attach the specific file that implements the: pdMS_TO_TICKS
Maybe it was not uploaded to the GIT.

roma-jam, 2023-07-19:
Hi @shlomo-edu, 

Please, try to get the final version of the code. 
Pay attention to the` hid_host_example.c`, the new version has a new way of polling the `APP_QUIT_PIN` state verification to exit the example (refer to updated file [here](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c#L456))

If there is still problem with building, please attach the full build log output for me to be able to understand the problem.
Otherwise, the Visual Studio Code can miss some headers and marked the lines with the red underlines. 
That is not affected the build process, that is only a indexing problem.

shlomo-edu, 2023-07-19:
It didn't work.
Here is the log -
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated C:/Users/Victoria/Desktop/ESP-PRJ/hid/build/bootloader/bootloader.bin
[110/110] cmd.exe /C "cd /D C:\Users\Victoria\Desktop\ESP-PRJ\hid\build\bootloader\esp-idf\esptool_py && C:\Users\Victoria\.espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:/Users/Victoria/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/Victoria/Desktop/ESP-PRJ/hid/build/bootloader/bootloader.bin"
Bootloader binary size 0x52b0 bytes. 0x2d50 bytes (35%) free.
[941/943] Linking CXX executable hid.elf
FAILED: hid.elf 
cmd.exe /C "cd . && C:\Users\Victoria\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Users/Victoria/Desktop/ESP-PRJ/hid/build/hid.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T 
esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\hid.elf.rsp -o hid.elf  && cd ."
c:/users/victoria/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/espressif__usb_host_hid/libespressif__usb_host_hid.a(hid_host.c.obj): in function `hid_host_install':
C:/Users/Victoria/Desktop/ESP-PRJ/hid/managed_components/espressif__usb_host_hid/hid_host.c:1082: multiple definition of `hid_host_install'; esp-idf/hid/libhid.a(hid_host.c.obj):C:/Users/Victoria/Desktop/ESP-PRJ/hid/components/hid/hid_host.c:875: first defined here
c:/users/victoria/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/espressif__usb_host_hid/libespressif__usb_host_hid.a(hid_host.c.obj): in function `hid_host_uninstall':
C:/Users/Victoria/Desktop/ESP-PRJ/hid/managed_components/espressif__usb_host_hid/hid_host.c:1163: multiple definition of `hid_host_uninstall'; esp-idf/hid/libhid.a(hid_host.c.obj):C:/Users/Victoria/Desktop/ESP-PRJ/hid/components/hid/hid_host.c:936: first defined here
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

*  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.

roma-jam, 2023-07-26:
Hi @shlomo-edu, 

Does the example build after cleaning the folder? 
Is there any other problems?

Thanks

shlomo-edu, 2023-07-26:
Hi Roma,
Thanks for your inquery.
My programmer had tested it, and now it compiles OK.
But when she run it, it "crashes" (this is what she told me).
I didn't have the time (yet) to examine it deeper.
Please note that what she does is just receive the data from the device.
The device is continuously sending data to the ESP.

roma-jam, 2023-07-26:
@shlomo-edu ,
Thanks for the response. 

Please, feel free to attach the crash report or any log during your (or her) testing. It helps me to solve the problem faster (if there is one). 

Thanks

shlomo-edu, 2023-07-26:
Hi Roma,
Thank you very much for your assistance.
Now I realize that it is not crash, but rather it just doesn't load the binary to the ESP32.
This is what my progammer tells me.
[image]

roma-jam, 2023-08-14:
@shlomo-edu Closing this issue as it's more likely problem with building/programming firmware and there is no issue with getting input report with HID Host example. 
Feel free to reopen if it's not.

closed 2023-08-14 by roma-jam

shlomo-edu, 2023-08-30:
Hi Roma,

I have received the ESP32-S3-WROOM Board from Freenove (the OTG).
But I don’t see any example how to check the connection to the device.
It does have two similar USB ports, and I can download the sketch and I can read and write from/to the Arduino Serial Monitor (using the “Serail” object).
It looks like the two ports are similar, so I assume that it can Read/Write from the device also with the “Serial” object, but how do I use two Serial objects (one for the PC and one for the device).

***@***.***

shlomo-edu, 2023-08-30:
Hi Roma,

Now I see that the two ports are not similar.
I can upload the sketch only when the PC is connected to the right port :
***@***.***

Closed #11702<https://github.com/espressif/esp-idf/issues/11702> as completed.

Postlast, 2023-09-07:
Hi Roma, 
I used https://components.espressif.com/components/espressif/usb_host_hid and it works. I still have little experience with the host and I can not find the functionality of how to get from the device in the program string_descriptor[] and report_descriptor[] Is there such a possibility?