espressif/esp-idf #11703: esp32c2/esp8464 wifi example connecting AP fails

Liming-Wang, 2023-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1, v5.0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP8684-DevKit M-1 v1.1

### Power Supply used.

USB

### What is the expected behavior?

when I supply the correct SSID and password, the example code should connect to the AP successfully. (I've tested the same AP with a ESP32 DevKit, and it works)

### What is the actual behavior?

The device fails to connect to the AP.

### Steps to reproduce.

1. Go to idf5.0.1/idf5.0.2 folder
2. . ./export.sh
3. cd examples/wifi/getting_started/station/
4. idf.py set-target esp32c2
5. idf.py menuconfig # configure ssid and password
6. idf.py flash
7. idf.py -b 74800 monitor

### Debug Logs.

```plain
--- idf_monitor on /dev/ttyUSB1 74800 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x1658
load:0x403acb70,len:0x968
load:0x403aeb70,len:0x27d8
entry 0x403acb70
I (21) boot: ESP-IDF v5.0.2 2nd stage bootloader
I (21) boot: compile time 21:04:37
I (21) boot: chip revision: v1.0
I (23) boot.esp32c2: MMU Page Size  : 32K
I (27) boot.esp32c2: SPI Speed      : 60MHz
I (32) boot.esp32c2: SPI Mode       : DIO
I (37) boot.esp32c2: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c068020 size=1c938h (117048) map
I (119) esp_image: segment 1: paddr=0002c960 vaddr=3fcab3f0 size=01f94h (  8084) load
I (122) esp_image: segment 2: paddr=0002e8fc vaddr=40380000 size=0171ch (  5916) load
I (127) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=67824h (423972) map
I (228) esp_image: segment 4: paddr=0009784c vaddr=4038171c size=09cd0h ( 40144) load
I (242) boot: Loaded app from partition at offset 0x10000
I (243) boot: Disabling RNG early entropy source...
I (254) cpu_start: Pro cpu up.
I (263) cpu_start: Pro cpu start user code
I (263) cpu_start: cpu freq: 120000000 Hz
I (263) cpu_start: Application information:
I (266) cpu_start: Project name:     wifi_station
I (271) cpu_start: App version:      1
I (276) cpu_start: Compile time:     Jun 19 2023 21:04:35
I (282) cpu_start: ELF file SHA256:  226ac53b61a5d9ad...
I (288) cpu_start: ESP-IDF:          v5.0.2
I (293) cpu_start: Min chip rev:     v1.0
I (297) cpu_start: Max chip rev:     v1.99 
I (302) cpu_start: Chip rev:         v1.0
I (307) heap_init: Initializing. RAM available for dynamic allocation:
I (314) heap_init: At 3FCB0FD0 len 0002BBA0 (174 KiB): D/IRAM
I (321) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (328) spi_flash: detected chip: gd
I (332) spi_flash: flash io: dio
W (336) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (349) cpu_start: Starting scheduler.
I (363) wifi station: ESP_WIFI_MODE_STA
I (373) pp: pp rom version: de8c800
I (373) net80211: net80211 rom version: de8c800
I (383) wifi:wifi driver task: 3fcb72cc, prio:23, stack:6144, core=0
I (383) system_api: Base MAC address is not set
I (383) system_api: read default base MAC address from EFUSE
I (393) wifi:wifi firmware version: 57982fe
I (393) wifi:wifi certification version: v7.0
I (393) wifi:config NVS flash: enabled
I (403) wifi:config nano formating: enabled
I (403) wifi:Init data frame dynamic rx buffer num: 32
I (413) wifi:Init management frame dynamic rx buffer num: 32
I (413) wifi:Init management short buffer num: 32
I (423) wifi:Init dynamic tx buffer num: 32
I (423) wifi:Init static tx FG buffer num: 2
I (423) wifi:Init static rx buffer size: 1600
I (433) wifi:Init static rx buffer num: 10
I (433) wifi:Init dynamic rx buffer num: 32
I (443) wifi_init: rx ba win: 6
I (443) wifi_init: tcpip mbox: 32
I (443) wifi_init: udp mbox: 6
I (453) wifi_init: tcp mbox: 6
I (453) wifi_init: tcp tx win: 5744
I (463) wifi_init: tcp rx win: 5744
I (463) wifi_init: tcp mss: 1440
I (463) wifi_init: WiFi IRAM OP enabled
I (473) wifi_init: WiFi RX IRAM OP enabled
I (493) phy_init: phy_version 210,e134770,Oct 20 2022,15:26:13
I (523) wifi:mode : sta (08:3a:8d:41:12:ac)
I (523) wifi:enable tsf
I (523) wifi station: wifi_init_sta finished.
I (2933) wifi station: retry to connect to the AP
I (2933) wifi station: connect to the AP fail
I (5343) wifi station: retry to connect to the AP
I (5343) wifi station: connect to the AP fail
I (7753) wifi station: retry to connect to the AP
I (7753) wifi station: connect to the AP fail
I (10163) wifi station: retry to connect to the AP
I (10163) wifi station: connect to the AP fail
I (12563) wifi station: retry to connect to the AP
I (12573) wifi station: connect to the AP fail
I (14973) wifi station: connect to the AP fail
I (14973) wifi station: Failed to connect to SSID:MyAP, password:myAPpassword
```

igrr, 2023-06-19:
Please try the suggestion here: https://github.com/espressif/esp-idf/issues/10782#issuecomment-1547495560

Liming-Wang, 2023-06-20:
Thanks @igrr , it works following you advice! It would be nice to have it noted in some README file. I guess it is a common issue people may encountered with esp32c2.

closed 2023-06-20 by Liming-Wang

Liming-Wang, 2023-06-22:
@igrr Thanks for the reference link. I had once read the getting-started guide for esp32, and I recently switched to esp32c2 developement, so I did not check that document carefully.  It does resolve the monitor garbage characters problem, but it does not  mention the wifi problem, actually, I just found out without changing the xtal frequency, bluetooth won't work either. So far as I know, this parameter update is needed for wifi or bluetooth to work properly. 

Thanks again for the reply!

catalineacsu, 2024-01-06:
I had the same issue. Changing the speed to 26MHz solved my problem. UART stayed at 74880 for some apps though.