espressif/esp-idf #11704: esp_pm_dump_locks() Wrong report data format with CONFIG_PM_PROFILING enabled

MacWyznawca, 2023-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-rc1-dirty

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Meaningless

### Power Supply used.

External 5V

### What is the expected behavior?

Well-formatted and readable text report from `esp_pm_dump_locks()` function with `CONFIG_PM_PROFILING` enabled.

### What is the actual behavior?

No end-of-line characters in `esp_pm_dump_locks() `function in lines of `pm_locks.c` file 177 and 179.

The effect in the attached image.
[image]

### Steps to reproduce.

Enable `CONFIG_PM_PROFILING` and use  `esp_pm_dump_locks()`

esp-wzh, 2023-06-20:
This feature works well on my side,
[image]

Are you using idf.py monitor? Or can you try a different terminal or monitor tool?

MacWyznawca, 2023-06-20:
@esp-wzh  As you can see I'm using a normal IDF (you can see in the window bar at the top). 
As I manually typed the end-of-line characters in the mentioned function, it helps, although not always.

[image]

esp-wzh, 2023-06-20:
OK, I reproduced it, seems it caused by `Time` field string too long, will give fix soon.

MacWyznawca, 2023-06-20:
@esp-wzh It is possible that the problem does not occur when you use a small terminal window (80 characters). However, I can't work that way.

MacWyznawca, 2023-06-20:
Great! Thanks @esp-wzh

esp-wzh, 2023-06-20:
[0001-bugfix-increase-pm_dump-time-field-width.patch.txt](https://github.com/espressif/esp-idf/files/11799298/0001-bugfix-increase-pm_dump-time-field-width.patch.txt)
This patch can fix the problem, thanks for report! @MacWyznawca

MacWyznawca, 2023-06-20:
Thanks @esp-wzh Works! ;-)

MacWyznawca, 2023-06-20:
By the way @esp-wzh, could you suggest where to look for help in lowering the power consumption of the ESP32-C3? I'm making a light bulb, but the power consumption is 0.66, and the EU requires less than 0.5 W. Note: we are only using ESP-NOW.
I use combinations with `esp_now_set_wake_window` and `esp_wifi_connectionless_module_set_wake_interval`, which lowers power consumption, but unfortunately causes sometimes no response (transmission errors).

Please ;-)

esp-wzh, 2023-06-20:
about ESP-NOW low power issues, please create a new issue to discuss and ask @Espressif-liuuuu.

closed 2023-06-25 by espressif-bot