espressif/esp-idf #11706: The eFuse bit mean?

WangShuoran, 2023-06-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Because of the espressif.com callback website can not running correctly. So I put the issue to this.

There is no detailed distinction between the two functions EFUSE_DIS_FORCE_DOWNLOAD and EFUSE_DIS_DOWNLOAD_MODE in eFuse in the documentation, and only the EFUSE_DIS_DOWNLOAD_MODE function is used in Secure Boot and Flash Encryption. So what is the role of EFUSE_DIS_FORCE_DOWNLOAD?
And in the documentation there is no introduction to EFUSE_WDT_DELAY_SEL, EFUSE_FLASH_TPUW (at the same time in ESP32C2 this EFUSE_FLASH_TPUW and write the wrong case).

在eFuse中的EFUSE_DIS_FORCE_DOWNLOAD和EFUSE_DIS_DOWNLOAD_MODE这两个功能中在文档中并没有详细的区分，而且在安全启动和Flash加密中仅仅使用了EFUSE_DIS_DOWNLOAD_MODE功能。那么EFUSE_DIS_FORCE_DOWNLOAD的作用是什么？
而且在文档中也没有对EFUSE_WDT_DELAY_SEL，EFUSE_FLASH_TPUW进行介绍（同时在ESP32C2中这个EFUSE_FLASH_TPUW还有写错的情况）

KonstantinKondrashov, 2023-07-04:
Hi @WangShuoran! You did not say which chip this issue relates to. ESP32 and ESP32-C2 do not have DIS_FORCE_DOWNLOAD efuse. The rest chips (C3, C6, H2, S2, S3) have DIS_FORCE_DOWNLOAD efuse.
You can take a look at the TRM in the "7.2 Boot Mode" chapter, there you can find notes how DIS_FORCE_DOWNLOAD works. 
[image]
The DIS_DOWNLOAD_MODE efuse disables the download mode completely.

For Security features (FE and SB), we have a Kconfig option to disable the download mode completely (DIS_DOWNLOAD_MODE) or enable Secure Download mode (EFUSE_ENABLE_SECURITY_DOWNLOAD). https://github.com/espressif/esp-idf/blob/c448597f24381d6c21294d1ecef10cc4533bcaa7/components/bootloader/Kconfig.projbuild#L1049
We do not touch the EFUSE_DIS_FORCE_DOWNLOAD because it does not affect the security (see "7.2 Boot Mode" chapter).

I have checked the EFUSE_WDT_DELAY_SEL in C2 TRM, and it is correct, it is placed in the register EFUSE_RD_REPEAT_DATA0_REG (0x0030). Both EFUSE_WDT_DELAY_SEL, and EFUSE_FLASH_TPUW are listed in TRM, maybe not directly in the efuse table, but in the register description they are there.

KonstantinKondrashov, 2023-07-05:
I opened a ticket for the TRM team to improve it. I close this issue for now, feel free to re-open it if any questions. Thanks.

closed 2023-07-05 by KonstantinKondrashov