espressif/esp-idf #11707: OTA 时执行get_sha256_of_partitions触发TG1WDT_SYS_RST

zbqxyz, 2023-06-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM

### Power Supply used.

External 3.3V

### What is the expected behavior?

OTA成功

### What is the actual behavior?

OTA过程执行到
void ota_app_main(void)
{
    // Initialize NVS.
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // 1.OTA app partition table has a smaller NVS partition size than the non-OTA
        // partition table. This size mismatch may cause NVS initialization to fail.
        // 2.NVS partition contains data in new format and cannot be recognized by this version of code.
        // If this happens, we erase NVS partition and initialize NVS again.
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);

    get_sha256_of_partitions();  这里触发看门狗复位

### Steps to reproduce.

1，链接wifi;
2,  链接MQTT；
3，MQTT推送OTA链接；
4，收到链接后执行OTA；

### Debug Logs.

```plain
esp32s3> wifi_event:scan done
get ap num 11
I (299378) BLUFI_EXAMPLE: BLUFI Set WIFI opmode 1

I (299418) BLUFI_EXAMPLE: Recv STA SSID shiyanshi

I (299518) PARAMTBL: param save ok
[... 199 lines cut ...]

        revision number:1
I (899) ADC: ADC1_0 voltage: 1 2344 1023mV
I (899) ADC: ADC1_0 voltage: 1 2348 1024mV
I (909) ADC: ADC1_0 voltage: 1 2347 1023mV
I (919) ADC: ADC1_0 voltage: 1 2345 1023mV
I (919) ADC: ADC1_0 voltage: 1 2347 1023mV
I (919) ADC: ADC1_0 voltage: 1 2345 1023mV
I (929) ADC: ADC1_0 voltage: 1 2345 1023mV
I (929) ADC: ADC1_0 voltage: 1 2345 1023mV
I (939) ADC: ADC1_0 voltage: 1 2347 1023mV
I (939) ADC: ADC1_0 voltage: 1 2346 1023mV
I (939) ADC: ADC1_0 voltage: 1 2345 1023mV
I (949) ADC: ADC1_0 voltage: 1 2348 1024mV
I (949) ADC: ADC1_0 voltage: 1 2345 1023mV
I (969) ADC: ADC1_0 voltage: 1 2347 1023mV
I (969) ADC: ADC1_0 voltage: 1 2345 1023mV
I (969) ADC: ADC1_0 voltage: 1 2344 1023mV
I (979) CD4051F:  adc id 246 u32dataVal=:  1023
```

### More Information.

[image]
[image]

ESP-YJM, 2023-06-20:
@zbqxyz ESP32S3 模组内部 PSRAM使用 IO35，IO36 和 IO37。你看下你有没有使用这几个 IO 口？

zbqxyz, 2023-06-20:
> @zbqxyz ESP32S3 模组内部 PSRAM使用 IO35，IO36 和 IO37。你看下你有没有使用这几个 IO 口？

我们用的模组里面没有PSRAM，至于ESP32S3R8芯片的才会有PSRAM

ESP-YJM, 2023-06-21:
请问下你这个 `ota_app_main` 是跑在哪个 task 里的？能简单的罗列下调用流程吗？或者，你能否重新创建个 task 去执行 ota_app_main 里的函数。

zbqxyz, 2023-06-21:
[image]

zbqxyz, 2023-06-21:
> 请问下你这个 `ota_app_main` 是跑在哪个 task 里的？能简单的罗列下调用流程吗？或者，你能否重新创建个 task 去执行 ota_app_main 里的函数。

 如上图，ota_app_main 是在mqtt解析任务重，接收到mqtt下发得升级链接后，执行ota_app_main

ESP-YJM, 2023-06-25:
@zbqxyz 这个截图还是没看出 ota_app_main 是执行在哪个 task 里。保险起见，你可以重新起一个 Task 去进行 flash 相关的操作。

zbqxyz, 2023-06-28:
> ota_app_main是在mqtt接收任务中执行的，但是 ota_app_main里面会创建一个OTA任务执行OTA功能，我现在把get_sha256_of_partitions屏蔽了，OTA数据下载可以正常执行，但是下载完一样会触发TG1WDT_SYS_RST复位，OTA用的是例程sample_ota ,通过HTTP更新的

zbqxyz, 2023-06-28:
[image]

ESP-YJM, 2023-06-29:
@zbqxyz `get_sha256_of_partitions` 也放到 task 里，另外，ota 完复位的日志是啥？

Alvin1Zhang, 2023-08-04:
Thanks for reporting and will close due to short of feedback, feel free to reopen with more updates.

closed 2023-08-04 by Alvin1Zhang