espressif/esp-idf #11708: Certain errors in mqtt client do not trigger MQTT_EVENT_ERROR

rkit1, 2023-06-20:
Couple of examples i came across:
```
E (39231) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (39231) mqtt_client: Poll read error: 119, aborting connection
```
`No PING_RESP, disconnected`

The easy way to reproduce is to establish a connection and then reboot the router.

euripedesrocha, 2023-07-11:
Hi @rkit1, for the particular situation you pointed the event that is dispatched is `MQTT_EVENT_DISCONNECTED` could you clarify why is helpful to also dispatch `MQTT_EVENT_ERROR`?

rkit1, 2023-07-11:
It's an error. Event should be dispatched on every error, not on some arbitrary subset. I spend hours trying to find out why my program is inconsistent, because I relied on this event to schedule a restart. And MQTT_EVENT_DISCONNECTED would also fire on disconnects requested by the program, where no restart is required. 
Yes, i could and did work around that, but either make it consistent or at least document it.

euripedesrocha, 2023-07-11:
@rkit1, while I agree with you that it's an error and consistency is helpful for the users, this is the behavior we have at the moment. 
Thanks for clarifying the issue you had, It's good to know the problems you had. 

I will look into our documentation and find ways to improve it.