espressif/esp-idf #11710: Expose more of Unity - Allow for easier implementation of user unit tests

Josh-TapNoa, 2023-06-20:
### Is your feature request related to a problem?

Currently there is no way to access the static variables s_unity_tests_first, and s_unity_tests_last, used to navigate the linked list of structs that makes up the internal unity test list. 

Navigating this list is straight forward, and would allow such things as component wise unit testing, custom unit test menus etc to be built trivially.

### Describe the solution you'd like.

Just expose them! 

Or give us a get function!

### Describe alternatives you've considered.

Running my own unity component in order to make the change.

Josh-TapNoa, 2023-06-20:
Functioning Workaround - 

unity_testcase_register allows the registering of an additional test_desc_t within the linked list. Create a local reference of this type, register it as a test, and then your original reference should contain the required linking. 

e.g.
void component_menu(const char *group_name)
{
    test_desc_t linked_test_list_start;
    unity_testcase_register(&linked_test_list_start);
    for (const test_desc_t *test = &linked_test_list_start; test != NULL; test = test->next) {
        printf("%s \n", test->desc);
        if (strstr(test->desc, group_name) != NULL){
            printf("TEST MATCH! \n");
        }
    }
}

igrr, 2023-06-20:
@Josh-TapNoa Which IDF version are you using? Starting from IDF v5.0, we have added two new functions:
* int unity_get_test_count(void);
* bool unity_get_test_info(int test_index, test_desc_t* out_info);

which allow iterating over the test cases list. Here is a piece of code which uses these functions: https://github.com/espressif/esp-idf/blob/03d4fa28694ee15ccfd5a97447575de2d1655026/components/sdmmc/test_apps/sdmmc_console/components/cmd_unity/cmd_unity.c#L37-L44

Josh-TapNoa, 2023-06-20:
Hi @igrr , we are on release 5.1, trying our best to stay up to date!

Thanks for the above, that seems like a simple way of extracting unity test information.

Many Thanks for the speed of your reply, you are considered a wizard in my office! :)

closed 2023-06-20 by Josh-TapNoa