espressif/esp-idf #11711: Post and Verify Data using esp_http_client example code.

radhikaalpha, 2023-06-20:
I have installed eclipse ide. The versions of all related installed software are as follow,
Operating System: windows 10
            Java Runtime Version: 17.0.7+8-LTS-224
            Eclipse Version: 4.27.0.v20230302-0300
            Eclipse CDT Version: 11.1.1.202303201356
            IDF Eclipse Plugin Version: 2.10.0.202306060331

Created a project using existing example code esp_http_client for ESP32-S3-DevKitM-1.
I want to Post data to https:/url OR to Posthere.io url. For that which function in this example code I need to use?

I have tried using http_rest_with_hostname_path() function. replaced the host in config with posthere.io url.

hmalpani, 2023-06-21:
Hello @radhikaalpha 
Can you update the client config as follows:
```
esp_http_client_config_t config = {
    .host = "posthere.io",
    .path = "/bxx7-4xx4-xxxx/post",
    .transport_type = HTTP_TRANSPORT_OVER_TCP,
    .event_handler = _http_event_handler,
};
```

Can you try with this configuration and let me know if this solves your problem.

radhikaalpha, 2023-06-22:
[image]

radhikaalpha, 2023-06-22:
attached screenshot of debug logs of updated code.
As per my understanding, my client is getting connected to server(posthere.io) but the data is not being posted there.
Can you please go through logs and let me know what is wrong in the code?

hmalpani, 2023-06-22:
@radhikaalpha

I tried with the client configuration I shared earlier. Data was posted on `posthere.io/bxx7-4xx4-xxxx/post`
What URL are you checking to see the posted data? 
The logs you shared looks good to me.

radhikaalpha, 2023-06-22:
esp_http_client_config_t config = {
        .host = "posthere.io",
        .path = "/0cb5-4524-9236/post",
        .transport_type = HTTP_TRANSPORT_OVER_TCP,
        .event_handler = _http_event_handler,
    };

The url is  http://posthere.io/0cb5-4524-9236

hmalpani, 2023-06-22:
The data will be posted on  `http://posthere.io/0cb5-4524-9236/post` for this client configuration

radhikaalpha, 2023-06-22:
[image]
It is showing no data

radhikaalpha, 2023-06-22:
[image]

radhikaalpha, 2023-06-22:
There haven't been any POST requests to http://posthere.io/0cb5-4524-9236/post in the last 24 hours.

hmalpani, 2023-06-22:
Hello @radhikaalpha 
I tried to do post request on the URL you shared above:

Code:
```
    esp_http_client_config_t config = {
        .host = "posthere.io",
        .path = "/0cb5-4524-9236/post",
        .event_handler = _http_event_handler,
        .user_data = local_response_buffer,        // Pass address of local buffer to get response
        .transport_type = HTTP_TRANSPORT_OVER_TCP,

    };
    esp_http_client_handle_t client = esp_http_client_init(&config);

    // POST
    const char *post_data = "{\"Espressif\":\"test2\"}";
    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_header(client, "Content-Type", "application/json");
    esp_http_client_set_post_field(client, post_data, strlen(post_data));
    esp_err_t err = esp_http_client_perform(client);
    if (err == ESP_OK) {
        ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %"PRIu64,
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
    } else {
        ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
    }

```

I can see the data posted on http://posthere.io/0cb5-4524-9236/post

radhikaalpha, 2023-06-22:
It is posting successfully now.
Thank you for your support.
Now, I am trying to send whatsapp message using post http. I will now change the url and try again. Will let you know if I need any help on that.

hmalpani, 2023-06-22:
Closing this issue. You can post a how-to question on the esp32.com forum for the implementation of the feature mentioned above.