espressif/esp-idf #11712: installer breaks something. Causes a DeprecationWarning.

nopnop2002, 2023-06-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.2-dev-1128

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

```
$ idf.py --version
ESP-IDF v5.2-dev-1128-g03d4fa2869
```

### What is the actual behavior?

I updated from esp-idf-v5.2-dev-544 to esp-idf-v5.2-dev-1128.

The following warning will occur:

```
$ idf.py --version
/home/nop/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
ESP-IDF v5.2-dev-1128-g03d4fa2869
```

So I changed it back to esp-idf-v5.2-dev-544.
```
$ idf.py --version
/home/nop/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
ESP-IDF v5.2-dev-544-g54576b7528
```

esp-idf-v5.2-dev-1128's installer breaks something.

### Steps to reproduce.

1. install esp-idf-v5.2-dev-1128.
2. Execute idf.py --version.

dobairoland, 2023-06-20:
We know about this and will be fixed soon.

I don't think anything is broken. You can ignore that warning. Everything should work.

nopnop2002, 2023-07-02:
This problem also occurs in ESP-IDF V5.0 and V5.1.

peterdragun, 2023-07-03:
Yes, the pkg_resources was used in all versions and was recently deprecated by the setuptools in version [67.5.0](https://setuptools.pypa.io/en/latest/history.html#v67-5-0)

nopnop2002, 2023-07-03:
So when we using setuptools older than 67.5.0, does the problem not occur?

nopnop2002, 2023-07-03:
>So when we using setuptools older than 67.5.0, does the problem not occur?

export.sh seems to force the setuptools to rewrite to the latest.

```
$ pip install setuptools==45.2.0

$ pip list | grep setuptools
setuptools             45.2.0

$ cd esp-idf

$ . ./export.sh

$ pip list | grep setuptools
setuptools            68.0.0
```

peterdragun, 2023-07-03:
> export.sh seems to force the setuptools to rewrite to the latest.

export.sh is switching to a different Python environment. So changing any package locally does not affect the IDF environment. If you care about the warning you can downgrade setuptools **after** running the export.sh as a temporary workaround until we implement a proper fix. We are already working on it and this issue will be fixed soon.

nopnop2002, 2023-07-03:
i am waiting for a fix.
Thanks for your comment.

bbannert, 2023-07-06:
Hi all. I'm getting the same message from ESP-IDF environment when compiling my code. The problem is that I'm getting this like an ERROR, not like a Warning. I'll be waiting for the fix too. Hope to get it soon. 
[image]

nopnop2002, 2023-07-08:
@bbannert 
Temporary measures are as follows.

```
$ cd esp-idf

$ . ./export.sh

$ python3 -m pip uninstall setuptools

$ python3 -m pip install setuptools==45.2.0

$ idf.py --version
ESP-IDF v5.2-dev-1449-gd2471b11e7
```

dobairoland, 2023-07-10:
@bbannert Which IDE are you using? Vscode or Eclipse?

closed 2023-07-15 by espressif-bot