espressif/esp-idf #11714: Power Management with ESP-NOW

MacWyznawca, 2023-06-20:
### Is your feature request related to a problem?

I have a problem with the power consumption of the ESP32-C3 used in the light bulb. I am using only ESP-NOW communication. The power consumption is currently 0.66 watts, and we need to go below 0.5 watts (EU and CE requirements). Unfortunately, ESP-NOW does not release the WiFi module and 
APB_FREQ_MAX for WiFi is 100%. Experiments with 
`esp_now_set_wake_window()` and 
`esp_wifi_connectionless_module_set_wake_interva()` reduce consumption, but cause frequent non-responses.

Translated with www.DeepL.com/Translator (free version)

### Describe the solution you'd like.

Can you @Espressif-liuuuu suggest any solution? How to optimally choose the parameters of Power Manager? Now I use max CPU 80 MHz and min 10 MHz and allow Light Sleep

Espressif-liuuuu, 2023-06-21:
hi @MacWyznawca , thanks for your report. 

There is no time syncronizing for ESPNOW yet, so there is do some packet loss if you use power-save for ESPNOW.   

And we have to do ESPNOW powersave before lightsleep. Actually we shall release RF at frist, which by calling `esp_now_set_wake_window` and `esp_wifi_connectionless_module_set_wake_interval`, then we are able to enter lightsleep.   

For your request, the quickest way is to adjust the `window` paramater to about 70% of `interval` parameter, e.g 70ms for window and 100ms for interval. In that case, the packet loose mayby OK and the power consumption meets the requirement also.

MacWyznawca, 2023-06-21:
OK, that's how I wanted to experiment. I will probably have to adjust the retransmission times.
Please, write me more @Espressif-liuuuu, will ESP-NOW be developed/maintained? Because we are doing a big smarthome system entirely based on ESP-NOW and TWAI. In addition, we plan to base the design of the next system implemented for 4-6 years on this.

MacWyznawca, 2023-06-21:
Thanks @Espressif-liuuuu 
Now we need to run a test. We have an elaborate communications system based on ESP-NOW with pairing, proprietary encryption, ACK and forwarding of packets. So far tested with ~100 simultaneously operating devices transmitting a dozen or so packets per second.

I'll report how progress with power saving or ask for support 😉

dpr-0, 2024-06-12:
> hi @MacWyznawca , thanks for your report.
>
> There is no time syncronizing for ESPNOW yet, so there is do some packet loss if you use power-save for ESPNOW.
>
> And we have to do ESPNOW powersave before [...]

After I set these two, I still get `ESP_ERR_NOT_SUPPORTED` during `esp_sleep_enable_wifi_beacon_wakeup()`  or `esp_sleep_enable_wifi_wakeup()` at my ESP32 DevKit board. Is `esp_sleep_enable_wifi_wakeup()` only for Wifi connected state?

MacWyznawca, 2024-06-12:
Hi @dpr-0 I use for ESP-NOW only: 
`esp_wifi_connectionless_module_set_wake_interval(interval) // interval = ex. 100`
and
`esp_now_set_wake_window(window) // window = ex 56 for ESP32-C or 90 for old ESP32`
and proj. setting:` CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE=y`
Power used on my device (Bulb) with ESP32-C3 droped from 0,6 to 0,48 W (measurewd in main power 230 V)

dpr-0, 2024-06-12:
I use ESP-NOW only too, so you still can receive ESP-NOW data during light sleep just based-on settings above? Is `esp_sleep_enable_wifi_wakeup()` actually not needed?

MacWyznawca, 2024-06-12:
I for init use only this code:
```c
static void LuAirRadio_init(void){
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());

    esp_wifi_set_ps(WIFI_PS_NONE); // WIFI_PS_MAX_MODEM

    ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCOL_LR | WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G)); // WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_LR
    ESP_ERROR_CHECK(esp_wifi_set_max_tx_power(_curentPower));
    ESP_ERROR_CHECK(esp_wifi_set_channel(_curentChannel, WIFI_SECOND_CHAN_NONE)); 
    ESP_ERROR_CHECK(esp_now_init());

    esp_now_peer_info_t *peer = malloc(sizeof(esp_now_peer_info_t));
    if (peer == NULL) {
        ESP_LOGE(LUAIR_TAG, "Malloc peer information fail");
        LuAirDeinit();
        return;
    }
    memset(peer, 0, sizeof(esp_now_peer_info_t));
    peer->channel = 0;
    peer->ifidx = ESP_IF_WIFI_STA;
    peer->encrypt = false;
    memcpy(peer->peer_addr, broadcastMac, ESP_NOW_ETH_ALEN);
    ESP_ERROR_CHECK(esp_now_add_peer(peer));
    LuPlatformMemoryFree(peer);

    ESP_ERROR_CHECK(esp_now_register_send_cb(LuAirSend_cb));
    ESP_ERROR_CHECK(esp_now_register_recv_cb(LuAirRecieve_cb));
    ESP_ERROR_CHECK(esp_wifi_connectionless_module_set_wake_interval(100)); 
    ESP_ERROR_CHECK(esp_now_set_wake_window(60));
}
```

Espressif-liuuuu, 2024-06-12:
> I use ESP-NOW only too, so you still can receive ESP-NOW data during light sleep just based-on settings above? Is `esp_sleep_enable_wifi_wakeup()` actually not needed?

Yes it didnt. its for wifi station mode & connected & internal use for lightsleep. For ESP-NOW only, its not needed.

Espressif-liuuuu, 2024-11-13:
> Pnow? I really need to achieve

Hi, @0x0fe. thanks for your report. 
There is indeed no direct way for promiscuous RX mode to sleep, since RF is essential if you want to receive some packet.  

If the sleep time is long, like you mentioned
`I try to monitor all channels during about 1200ms (100ms per channel) and then deep sleep 20 second`

How about disable promiscuous RX mode or call `esp_wifi_stop` before you sleep? Thats the quickest way to achieve the logic.

Please let me know if any issue in that solution. Thanks!

0x0fe, 2024-11-13:
@Espressif-liuuuu 
Hi,
I was hoping for a mode to reach a better current in promiscuous RX only mode, but after checking the datasheet i realize 84mA is the minimum expected current for RX operation.

[image]

> How about disable promiscuous RX mode or call esp_wifi_stop before you sleep? That's the quickest way to achieve the logic.

Well, during the sleep phase (20 second) i enter in deep sleep mode and i reach the expected deep sleep current : 7 uA at 5V (before the LDO). By the way, esp_wifi_stop is unnecessary, it is probably called by the deep sleep API.

The 24mA i have seen in light sleep is in fact when the radio is disabled, which explains the jumps around 80 when i switch channel.
Is there another ESP32 variant that could reach a more reasonable RX current in promiscuous mode? Other generic wifi SoCs reach 34mA and down to 14mA for ultra low power wifi socs.

Espressif-liuuuu, 2024-11-14:
> @Espressif-liuuuu Hi, I was hoping for a mode to reach a better current in promiscuous RX only mode, but after checking the datasheet i realize 84mA is the minimum expected current for RX [...]

Rx current of ESP32C2 can be 65mA. Sorry that it cant achieve so low Rx current for ESP32 now, but we will have that type of chips in the future.

0x0fe, 2024-11-14:
@Espressif-liuuuu OK, it is clear, thank you for the information.