espressif/esp-idf #11717: My definitions in CMakeLists of my component aren´t added to the build process

Kampi, 2023-06-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I´m writing on a LoRaWAN component for an ESP32 and I want to define a library version number in the CMakeLists of my component.

For this, I have added 

```
add_compile_definitions(CONFIG_RAK3172_LIB_MAJOR=4)
add_compile_definitions(CONFIG_RAK3172_LIB_MINOR=2)
add_compile_definitions(CONFIG_RAK3172_LIB_BUILD=0)
```

on the top of the `CMakeLists.txt` of my library. I use the following function to print the version number

```
inline __attribute__((always_inline)) const std::string RAK3172_LibVersion(void)
{
    #if((defined CONFIG_RAK3172_LIB_MAJOR) && (defined CONFIG_RAK3172_LIB_MINOR) && (defined CONFIG_RAK3172_LIB_BUILD))
        return std::string(STRINGIFY(RAK3172_LIB_MAJOR)) + "." + std::string(STRINGIFY(RAK3172_LIB_MINOR)) + "." + std::string(STRINGIFY(RAK3172_LIB_BUILD));
    #else
        return "<Not defined>";
    #endif
}
```

But the function always prints `<Not defined>`, so it looks like the definitions aren´t defined during the build. How can I fix it? Where is the issue?

Kampi, 2023-06-20:
Never mind. Was an issue on my side :)

igrr, 2023-06-20:
Could you please try calling target_compile_definitions after idf_component_register?

https://github.com/espressif/esp-idf/blob/03d4fa28694ee15ccfd5a97447575de2d1655026/components/esp-tls/CMakeLists.txt#L39

One word of caution: I'd recommend not using macro names prefixed with `CONFIG_` for custom definitions, since normally such macros are generated automatically from Kconfig options. Picking a different name prefix can help the reader of the code understand that what they are looking at isn't in fact a Kconfig option.

Kampi, 2023-06-20:
Hi @igrr 

thanks for the warning. I have changed `CMakeList.txt` in the following way:

```
register_component()
target_compile_definitions(${COMPONENT_LIB} PUBLIC RAK3172_LIB_MAJOR=4)
target_compile_definitions(${COMPONENT_LIB} PUBLIC RAK3172_LIB_MINOR=2)
target_compile_definitions(${COMPONENT_LIB} PUBLIC RAK3172_LIB_BUILD=0)
```

and I have still the issue

```
components/RAK3172/include/rak3172.h: In function 'const string RAK3172_LibVersion()':
components/RAK3172/include/rak3172.h:50:27: error: 'RAK3172_LIB_MAJOR' was not declared in this scope
   50 |     return std::to_string(RAK3172_LIB_MAJOR) + "." + std::to_string(RAK3172_LIB_MINOR) + "." + std::to_string(RAK3172_LIB_BUILD);
      |                           ^~~~~~~~~~~~~~~~~
Compiling .pio\build\debug\console\argtable3\arg_end.o
components/RAK3172/include/rak3172.h:50:69: error: 'RAK3172_LIB_MINOR' was not declared in this scope
   50 |     return std::to_string(RAK3172_LIB_MAJOR) + "." + std::to_string(RAK3172_LIB_MINOR) + "." + std::to_string(RAK3172_LIB_BUILD);
      |                                                                     ^~~~~~~~~~~~~~~~~
Compiling .pio\build\debug\console\argtable3\arg_file.o
components/RAK3172/include/rak3172.h:50:111: error: 'RAK3172_LIB_BUILD' was not declared in this scope
   50 |     return std::to_string(RAK3172_LIB_MAJOR) + "." + std::to_string(RAK3172_LIB_MINOR) + "." + std::to_string(RAK3172_LIB_BUILD);
```

igrr, 2023-06-20:
I'm not sure about the last error, which compilation unit (.c or .cpp file) is it produced from? After you find which .cpp file fails to compile, please search for that file in build/compile_commands.json and find the full command line used to compile that file. This will help determine: 1) which component is the source file located in, and 2) whether the compilation flags are propagated to that component correctly.

Kampi, 2023-06-21:
Hi @igrr,
the error is generated in `rak3172.cpp` (see https://github.com/Kampi/ESP32-RAK3172/tree/4.2_Dev for reference). The output from  `build/compile_commands.json` is the following one

```
{
  "directory": "F:/NextCloud/Git/ESP32/Projects/RAK3172_Dev/.pio/build/debug",
  "command": "C:\\Users\\konta\\.platformio\\packages\\toolchain-xtensa-esp32\\bin\\xtensa-esp32-elf-g++.exe  -DRAK3172_LIB_BUILD=0 -DRAK3172_LIB_MAJOR=4 -DRAK3172_LIB_MINOR=2 -Iconfig -I../../../components/RAK3172/include -I../../../components/RAK3172/include/Modes -I../../../components/RAK3172/include/Definitions -IC:/Users/konta/.platformio/packages/framework-espidf/components/newlib/platform_include -IC:/Users/konta/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/freertos/esp_additions/include/freertos -IC:/Users/konta/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/freertos/esp_additions/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_hw_support/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_hw_support/include/soc -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_hw_support/include/soc/esp32 -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_hw_support/port/esp32/. -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_hw_support/port/esp32/private_include -IC:/Users/konta/.platformio/packages/framework-espidf/components/heap/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/log/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/soc/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/soc/esp32/. -IC:/Users/konta/.platformio/packages/framework-espidf/components/soc/esp32/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/hal/esp32/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/hal/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/hal/platform_port/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_rom/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_rom/include/esp32 -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_rom/esp32 -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_common/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_system/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_system/port/soc -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_system/port/include/private -IC:/Users/konta/.platformio/packages/framework-espidf/components/xtensa/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/xtensa/esp32/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/lwip/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/lwip/include/apps -IC:/Users/konta/.platformio/packages/framework-espidf/components/lwip/include/apps/sntp -IC:/Users/konta/.platformio/packages/framework-espidf/components/lwip/lwip/src/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/lwip/port/esp32/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/lwip/port/esp32/include/arch -IC:/Users/konta/.platformio/packages/framework-espidf/components/driver/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/driver/deprecated -IC:/Users/konta/.platformio/packages/framework-espidf/components/driver/esp32/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_pm/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_ringbuf/include -IC:/Users/konta/.platformio/packages/framework-espidf/components/esp_timer/include  -mlongcalls -Wno-frame-address    -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Og -fmacro-prefix-map=F:/NextCloud/Git/ESP32/Projects/RAK3172_Dev=. -fmacro-prefix-map=C:/Users/konta/.platformio/packages/framework-espidf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu++20 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\\\"5.0.1\\\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -o esp-idf\\RAK3172\\CMakeFiles\\__idf_RAK3172.dir\\src\\rak3172.cpp.obj -c F:\\NextCloud\\Git\\ESP32\\Projects\\RAK3172_Dev\\components\\RAK3172\\src\\rak3172.cpp",
  "file": "F:/NextCloud/Git/ESP32/Projects/RAK3172_Dev/components/RAK3172/src/rak3172.cpp"
},
```

So the flags were added to the build process

igrr, 2023-06-21:
The source code over at that link looks a bit different than what you have posted here (e.g. std::to_string instead of STRINGIFY). Could you please post the version which fails to compile to some branch?

Kampi, 2023-06-21:
Hi @igrr,

yes, that´s true, because I didn´t push the changes from yesterday to the repo. You can use the repo to validate the structure of the component. Please use the code from this issue thread. It´s the code I´m using in the current modification to set the library version. I will update the repo as soon as the code works as expected.

igrr, 2023-06-21:
I have tested this branch, making two changes:
1. Replaced the function in rak3172.h:
   ```cpp
   inline __attribute__((always_inline)) const std::string RAK3172_LibVersion(void)
   {
       #if((defined RAK3172_LIB_MAJOR) && (defined RAK3172_LIB_MINOR) && (defined RAK3172_LIB_BUILD))
           return std::to_string(RAK3172_LIB_MAJOR) + "." + std::to_string(RAK3172_LIB_MINOR) + "." + std::to_string(RAK3172_LIB_BUILD);
       #else
           return "<Not defined>";
       #endif
   }
   ```
2. Changed CMakeLists.txt to add use `idf_component_register` and `target_compile_definitions` as you have shown above.

I have then created an IDF project with the following main.cpp:
```cpp
#include <stdint.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <freertos/event_groups.h>
#include <rak3172.h>

#include "sdkconfig.h"

static RAK3172_t _Device                        = RAK3172_DEFAULT_CONFIG(UART_NUM_0, 1, 2, 115200);
static const char* TAG 							= "main";

extern "C" void app_main(void)
{
    RAK3172_Error_t Error;
    RAK3172_Info_t Info;
    _Device.Info = &Info;
    Error = RAK3172_Init(_Device);
    if(Error != RAK3172_ERR_OK)
    {
        ESP_LOGE(TAG, "Cannot initialize RAK3172! Error: 0x%04X", static_cast<unsigned int>(Error));
    }
    ESP_LOGI(TAG, "Firmware: %s", Info.Firmware.c_str());
    ESP_LOGI(TAG, "Serial number: %s", Info.Serial.c_str());
    ESP_LOGI(TAG, "Current mode: %u", _Device.Mode);
}
```

It got compiled using `idf.py build` without an issue, and "4.2.0" version string is correctly used. So I conclude that RAK3172_LIB_MAJOR and other flags are correctly passed to the compiler.

If you manage to create a branch where the issue you are seeing can be reproduced, please let me know. Otherwise I'll consider this issue resolved.

Kampi, 2023-06-21:
Hi @igrr 

thanks. It works if I´m using

```
inline __attribute__((always_inline)) const std::string RAK3172_LibVersion(void)
{
    #if((defined RAK3172_LIB_MAJOR) && (defined RAK3172_LIB_MINOR) && (defined RAK3172_LIB_BUILD))
        return std::to_string(RAK3172_LIB_MAJOR) + "." + std::to_string(RAK3172_LIB_MINOR) + "." + std::to_string(RAK3172_LIB_BUILD);
    #else
        return "<Not defined>";
    #endif
}
```

but not with

```
inline __attribute__((always_inline)) const std::string RAK3172_LibVersion(void)
{
    return std::to_string(RAK3172_LIB_MAJOR) + "." + std::to_string(RAK3172_LIB_MINOR) + "." + std::to_string(RAK3172_LIB_BUILD);
}
```

Do you have an idea why? It´s optional - the main issue is solved :)
I will close it if you have no answer.

closed 2023-06-25 by Kampi