espressif/esp-idf #11718: How to disable cpu caching

vuthaitdp1, 2023-06-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to disable/enable cpu caching in the interrupt function so that I can calculate the read and write cycles of the peripheral registers correctly

suda-morris, 2023-06-21:
> calculate the read and write cycles of the peripheral registers correctly

You don't have to disable the cache if you just want to measure the time of read/write cycles. There're other approaches. Please check out [this component](https://components.espressif.com/components/espressif/ccomp_timer)

igrr, 2023-06-21:
According to the ESP32 (and ESP32-S3) TRM, CPU access to peripheral registers is not cached.

Only access to code/rodata in Flash is cached. If you place your interrupt handler function into IRAM, then execution should be predictable (up to the variable interrupt latency due to other factors).

vuthaitdp1, 2023-06-21:
> According to the ESP32 (and ESP32-S3) TRM, CPU access to peripheral registers is not cached.
> 
> Only access to code/rodata in Flash is cached. If you place your interrupt handler function into IRAM, then execution should be predictable (up to the variable interrupt latency due to other factors).

I need the error between interrupts to be the lowest, for example: 50us delay +/-0.01us max to trigger DMA transmission on time, if using cache the above error will change incorrectly due to  read and write changes.

closed 2024-05-10 by igrr