espressif/esp-idf #11720: SPI: assert failed: spi_intr spi_master.c:729 (desired_dev)

dibalavs, 2023-06-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-rc1-36-g4bc762621d

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

cutom board

### Power Supply used.

USB

### What is the expected behavior?

No assertions. SPI worked properly.

### What is the actual behavior?

`assert failed: spi_intr spi_master.c:729 (desired_dev)

Backtrace: 0x400825e2:0x3ffb2030 0x400921e5:0x3ffb2050 0x4009ad0e:0x3ffb2070 0x40086bf1:0x3ffb21a0 0x40083b05:0x3ffb21d0 0x4000bfed:0x3ffda650 0x40094d84:0x3ffda660 0x40092944:0x3ffda680 0x4009a319:0x3ffda6c0 0x4009a59b:0x3ffda6f0 0x40094b09:0x3ffda720
0x400825e2: panic_abort at C:/Work/repos/esp_radio/Firmware/esp-idf/components/esp_system/panic.c:452

0x400921e5: esp_system_abort at C:/Work/repos/esp_radio/Firmware/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x4009ad0e: __assert_func at C:/Work/repos/esp_radio/Firmware/esp-idf/components/newlib/assert.c:81

0x40086bf1: spi_intr at C:/Work/repos/esp_radio/Firmware/esp-idf/components/driver/spi/gpspi/spi_master.c:678 (discriminator 1)

0x40083b05: _xt_medint2 at C:/Work/repos/esp_radio/Firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1315

0x40094d84: vPortClearInterruptMaskFromISR at C:/Work/repos/esp_radio/Firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:568
 (inlined by) vPortExitCritical at C:/Work/repos/esp_radio/Firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:532

0x40092944: xQueueReceive at C:/Work/repos/esp_radio/Firmware/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1576

0x4009a319: sys_arch_mbox_fetch at C:/Work/repos/esp_radio/Firmware/esp-idf/components/lwip/port/freertos/sys_arch.c:317

0x4009a59b: tcpip_timeouts_mbox_fetch at C:/Work/repos/esp_radio/Firmware/esp-idf/components/lwip/lwip/src/api/tcpip.c:104
 (inlined by) tcpip_thread at C:/Work/repos/esp_radio/Firmware/esp-idf/components/lwip/lwip/src/api/tcpip.c:142

0x40094b09: vPortTaskWrapper at C:/Work/repos/esp_radio/Firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:1
`

or another crash:

`assert failed: spi_intr spi_master.c:729 (desired_dev)

Backtrace: 0x400825e2:0x3ffb2030 0x400921e5:0x3ffb2050 0x4009ad0e:0x3ffb2070 0x40086bf1:0x3ffb21a0 0x40083b05:0x3ffb21d0 0x40082cc9:0x3ffde3b0 0x401bf52d:0x3ffde3d0 0x401bff30:0x3ffde400 0x401bf2be:0x3ffde430 0x401c119b:0x3ffde480 0x401c1267:0x3ffde4c0 0x401bd70e:0x3ffde500 0x40094b09:0x3ffde530
0x400825e2: panic_abort at C:/Work/repos/esp_radio/Firmware/esp-idf/components/esp_system/panic.c:452

0x400921e5: esp_system_abort at C:/Work/repos/esp_radio/Firmware/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x4009ad0e: __assert_func at C:/Work/repos/esp_radio/Firmware/esp-idf/components/newlib/assert.c:81

0x40086bf1: spi_intr at C:/Work/repos/esp_radio/Firmware/esp-idf/components/driver/spi/gpspi/spi_master.c:678 (discriminator 1)

0x40083b05: _xt_medint2 at C:/Work/repos/esp_radio/Firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1315

0x40082cc9: heap_caps_malloc_default at C:/Work/repos/esp_radio/Firmware/esp-idf/components/heap/heap_caps.c:242

0x401bf52d: esf_buf_alloc_dynamic at ??:?

0x401bff30: esf_buf_alloc at ??:?

0x401bf2be: wDev_IndicateAmpdu at ??:?

0x401c119b: wDev_ProcessRxSucData at ??:?

0x401c1267: wdevProcessRxSucDataAll at ??:?

0x401bd70e: ppTask at ??:?

0x40094b09: vPortTaskWrapper at C:/Work/repos/esp_radio/Firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
`

### Steps to reproduce.

I use custom KaRadio repository with my patches.

1) SPI initialization:
`    esp_err_t ret;
	static const spi_bus_config_t buscfg = {
        .miso_io_num = PIN_SPI_MISO,
        .mosi_io_num = PIN_SPI_MOSI,
        .sclk_io_num = PIN_SPI_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
		.flags = SPICOMMON_BUSFLAG_MASTER
	};

	ret = spi_bus_initialize(SPI_NO, &buscfg, SPI_DMA_CH1);
    ESP_LOGI(TAG, "Init SPI bus. (mosi:%d, miso:%d, clk:%s) = %d", buscfg.mosi_io_num, buscfg.miso_io_num, buscfg.sclk_io_num, esp_err_to_name(ret));
	ESP_ERROR_CHECK(ret);

uint32_t freq =spi_get_actual_clock(APB_CLK_FREQ, 1400000, 128);
	ESP_LOGI(TAG,"VS1053 LFreq: %d",freq);
	spi_device_interface_config_t devcfg={
        .clock_speed_hz=freq,               //Clock out at x Hz
		.command_bits = 8,
		.address_bits = 8,
		.dummy_bits = 0,
		.duty_cycle_pos = 0,
		.cs_ena_pretrans = 0,
		.cs_ena_posttrans = 1,
		.flags = 0,
        .mode = 0,                         //0 = (0-CPOL, 0-CPHA)
        .spics_io_num = PIN_NUM_XCS,               //XCS pin
        .queue_size = 1,                          //We want to be able to queue x transactions at a time
        .pre_cb = NULL,
		.post_cb = NULL
	};

	//slow speed
	ESP_ERROR_CHECK(spi_bus_add_device(SPI_NO, &devcfg, &vsspi));
`

Transaction wrapper:
`#define SAFE_TRANSMIT(bus, transaction, retval) \
({ \
	FREERTOS_ERROR_CHECK(xSemaphoreTake(lock, portMAX_DELAY)); \
	ESP_ERROR_CHECK(spi_device_acquire_bus(bus, portMAX_DELAY)); \
    ret = spi_device_transmit(bus, &transaction); \
	spi_device_release_bus(bus); \
	FREERTOS_ERROR_CHECK(xSemaphoreGive(lock)); \
	ret; \
})
`

Assertion happens when I added spi_device_acquire_bus()/spi_device_release_bus() calls.
I need them, because there are 2 devices connected to the same SPI bus (audio decoder VS1053 and LCD display)

### More Information.

`cat sdkconfig | grep CONFIG_SPI_ | grep -v FLASH
CONFIG_SPI_MASTER_IN_IRAM=y
CONFIG_SPI_MASTER_ISR_IN_IRAM=y
CONFIG_SPI_SLAVE_IN_IRAM=y
CONFIG_SPI_SLAVE_ISR_IN_IRAM=y`

git branch:
* release/v5.1
git log:
commit 4bc762621db30f253742bd3c439f80500fda0fb0 (HEAD -> release/v5.1, origin/release/v5.1)

dibalavs, 2023-06-21:
Hello @wanckl  it looks like automatic acquire/release does not work for interrupt based transmit.

without acquire/release I get assertion inside spi_device_transmit()

https://github.com/espressif/esp-idf/blob/03d4fa28694ee15ccfd5a97447575de2d1655026/components/driver/spi/gpspi/spi_master.c#L969

KaeLL, 2023-06-21:
Are you sure you're abiding by this?
[image]

dibalavs, 2023-06-21:
> if multiple tasks try to access the same SPI Device.

This is not my case.
My case is Multiple SPI devices served by multiple tasks on the same SPI bus.
VS1053 device is wrapped by mutexes to protect access from different tasks.
But issue is that exists another SPI Device on the same bus.

dibalavs, 2023-06-21:
But about isr_cpu_id - is good idea. I can try to assign all tasks, which do SPI communication to the same core and set isr_cpu_id.

For me, it looks strange. ESP-IDF have different API for I2C bus.
It has no such problems. Documentation says:

> The I2C port is protected by mutex, so this function is thread-safe.

Why SPI does not have such protection?

dibalavs, 2023-06-22:
I removed acuire/release and add mutexes.  No assertion and no Gugu Meditation anymore.

wanckl, 2023-06-25:
@dibalavs Hi,   
> My case is Multiple SPI devices served by multiple tasks on the same SPI bus.   

If I understand correctly, this means task not assigned to device one by one, each task is able to access each device. So, as your `VS1053` device is wrapped, it can accessed by only one task at a moment, but another device may accessed by multiple tasks on same time.    
If this, it is right your case

>Why SPI does not have such protection?   

IIC now have no abstract of devices. SPI have another atomic type of `bus_lock` that guarantee the bus is accessed by one device at one time. But can not guarantee one device accessed by multiple tasks, so you should add mutex for your devices.

As SPI is a peripheral faster than IIC a lot, OS mutex will impact SPI's effect, you should add this for your devices if needed.

aboyuanna, 2023-08-11:
The issue [f00b4r0](https://github.com/f00b4r0) encountered is the same as mine.It's worth noting that I didn't encounter this issue in ESP-IDF v5.0, but it occurred in v5.1.

wanckl, 2023-08-17:
Hi @f00b4r0 ,your situation is similar to my friend. there is a [FIX](https://github.com/espressif/esp-idf/commit/b845c66748fe0046203000fbca98875741bb2a0c) on release 5.1 that fix this issue, please update your release version, the newest version on a release branch (not master branch) should has the least bug number, feel free to update the minor version. good luck~

wanckl, 2023-08-17:
@dibalavs I checked that your version also don't have the fix above, maybe you can also update your version to latest 5.1.x to have a test if you don't mind, your feedback will be very worth, thank you!

closed 2025-11-14 by Alvin1Zhang