espressif/esp-idf #11722: 关于ESP32C2的SPI数据线MOSI空闲电平的配置问题

gm7370, 2023-06-21:
### Is your feature request related to a problem?

请问ESP32C2这个芯片，如何配置SPI的MOSI空闲状态电平。我们看了数据手册，说是需要通过修改寄存器SPI_CTRL_REG (0x0008)的SPI_D_POL位，但是spi_device_interface_config_t的结构体中似乎没有提供这个接口参数。如果真的可以配置SPI_D_POL，具体需要如何操作？

wanckl, 2023-06-21:
这，，，

wanckl, 2023-06-21:
@gm7370 
现在esp官方貌似还没有开出来这个功能，你如果急需用的话可以这样（正常情况不建议直接碰寄存器）：
```c
#include "soc/spi_struct.h"
GPSPI2.ctrl.d_pol = 0;
```
如果编译不过的话，你用的哪个芯片相应改一下名字。拿走不谢！

gm7370, 2023-06-21:
非常感谢你的帮助！我和我的朋友今天也讨论过，使用多种方法都能实现。其中就提到过你这种方式，但是这个只能修改SPI2。另外两种方法我也和大家分享一下吧！
1、通过调用SPI_LL库获得SPI实例基址，将它传给spi_dev_t的一个实例指针，然后就可以操作D_POL了，代码如下：
    spi_dev_t *hw;
    hw=SPI_LL_GET_HW(HOST_ID);
    hw->ctrl.d_pol=0;
2、直接通过指针，指向0x60024008，可直接配置SPI_CRL_REG。但是这个方法也是不太建议这么做！所以也没有验证过。
    uint32_t *p = (uint32_t *)0x60024008;
    *p &= ~(1<<19);

wanckl, 2025-06-03:
Hi @gm7370  

This option already configurable from driver side by using `spi_bus_config_t::data_io_default_level` since version `5.4.1`, thanks.

closed 2025-06-03 by Alvin1Zhang