espressif/esp-idf #11723: Strange behaviour of sprintf()

stefancn, 2023-06-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-rc1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

When I does this statement:

  char Json_str[400];  
  float temperature = 26.6;
  double seconds = 1687183452;

  sprintf(Json_str," {\"time\":%lld,\"value\":%4.2f,\"nodename\":\"ESP32_TEST\",\"nodelocation\":\"home\"} ",seconds,temperature);

I should get:

{"time":1687183452,"value":26.60,"nodename":"ESP32_TEST","nodelocation":"home"} 

### What is the actual behavior?

Instead I get:

" {"time":1687183452,"value":26.80,"nodename":"ESP32_TEST","nodelocation":"home"} \000\377\377;\000\000\000\n\000\000\000372216\001@\377\377\377\377;\000\000\000\n\000\000\000276216\001@\377\377\377\377;\000\000\000376\377\377\377^222\001@245245245245245245245245\000\000\000\000306252\200@"

### Steps to reproduce.

Just run the code above ...
...

### Debug Logs.

```plain
***ERROR*** A stack overflow in task DHT_reader_task has been detected.
Core  0 register dump:
MEPC    : 0x4080062e  RA      : 0x408063de  SP      : 0x40814560  GP      : 0x408111f0
0x4080062e: panic_abort at D:/LANGUAGE/CRASMB/ESP32/esp-idf/components/esp_system/panic.c:452
```

### More Information.

Any ideas ....

Regards 

Stefan

txgk, 2023-06-22:
printf-family of functions usually require a lot of memory and error message `***ERROR*** A stack overflow in task DHT_reader_task has been detected.` implies that the task is out of memory, hence the strange behavior in your case. Just allocate more memory for your `DHT_reader_task` and it should be fine (stack depth of `3072` worked for me with this code). Also you should use `%lf` to print `double` variables instead of `%lld`.

igrr, 2023-06-23:
@stefancn Could you please also show the code which you use to obtain the "I should get:" and "Instead I get:" lines? That is, how are you inspecting the value of `Json_str` variable?

(What @txgk has noted regarding the stack size and %lf instead of %lld is also correct!)

filzek, 2023-07-02:
why you not menset the Json_str with zeroes first? I think there is not a end char on it.

closed 2023-10-11 by ESP-Marius