espressif/esp-idf #11727: EDP-IDF for Version 5.1-rc2 installation generates a Folder Name including  ","  in python_env instead of a "."

TestUserForIOT, 2023-06-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Tag v5.1-rc2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

When installing the ESP-IDF compiler for the Version 5.1-rc2 the installation script generates a folder named "idf5,0_py3.9_env" under the Path C:/Users/UserName/.espressif/python_env.

There is a "," in between the idf Version Number instead of a ".". Previous versions all had a dot.

Now when building the example "Hello World" project  the compiler crashes with the Error Message "The command "C:\Users\UserName\.espressif\python_env\idf5" is ether spelled incorrect or could not be found".

This is due to that windows uses the "," as a separation. Therefore, the compiler does not use the correct folder name to find the needed includes.

If the installation code is changed back such that the folder is automatically names with a dot, this error wont occur.

Is there a fast method to fix this for Windows? Only changing the folder name does not solve the problem, because internal paths, which are automatically set during the installation process, are searching for the name with ","

### What is the actual behavior?

Look ahead please.

### Steps to reproduce.

Install the ESP IDF Version 5.1-rc2 on a Windows 11 PC and try to build the example "Hello-World" project.

### Build or installation Logs.

```plain
C:\Espressif\frameworks\esp-idf-v5.1-rc2\examples\get-started>cd hello_world

C:\Espressif\frameworks\esp-idf-v5.1-rc2\examples\get-started\hello_world>idf.py build
C:\Espressif\frameworks\esp-idf-v5.1-rc2\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running cmake in directory C:\Espressif\frameworks\esp-idf-v5.1-rc2\examples\get-started\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Users\UserName\.espressif\python_env\idf5,0_py3.9_env\Scripts\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 C:\Espressif\frameworks\esp-idf-v5.1-rc2\examples\get-started\hello_world"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Users/UserName/.espressif/tools/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/UserName/.espressif/tools/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/UserName/.espressif/tools/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/UserName/.espressif/tools/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
[... 15 lines cut ...]
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "hello_world" version: v5.1-rc2
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/examples/get-started/hello_world/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/app_trace C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/app_update C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/bootloader C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/bt C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/cmock C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/console C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/cxx C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/driver C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/efuse C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp-tls C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_adc C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_coex C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_common C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_eth C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_event C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_hid C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_http_client C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_http_server C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_https_server C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_lcd C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_local_ctrl C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_mm C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_netif C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_netif_stack C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_partition C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_phy C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_pm C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_psram C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_ringbuf C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_system C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_timer C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esp_wifi C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/espcoredump C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/fatfs C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/freertos C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/hal C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/heap C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/http_parser C:/Espressif/frameworks/esp-idf-v5.1-rc2/components/idf_test C:/Espressif/frameworks/esp-idf-v5.1-rc2/component [... cut ...]
Der Befehl "C:\Users\UserName\.espressif\python_env\idf5" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
[3/884] Generating memory.ld linker script...ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-idf-v5.1-rc2\examples\get-started\hello_world\build\log\idf_py_stderr_output_21256 and C:\Espressif\frameworks\esp-idf-v5.1-rc2\examples\get-started\hello_world\build\log\idf_py_stdout_output_21256
```

dobairoland, 2023-06-22:
`idf5,0_py3.9_env` - this doesn't seem right for another reason beside having a coma in it. If you install 5.1 then it should be `idf5.1_py3.9_env`. Also I would suggest to use the installer from https://dl.espressif.com/dl/esp-idf/ and use the Python offered by it. It will solve several potential issues for you.

Have you set an `IDF_PYTHON_ENV_PATH` environment variable? I think you've overwritten the path by this.

If you set any (ESP-IDF) environment variables then I suggest to remove them and re-run the installer.

TestUserForIOT, 2023-06-22:
Thank you very much for the fast respond.

I used the exact install tool from the website/link you mentioned.

I checked the environmental variables. The path of `IDF_PYTHON_ENV_PATH` included the ",".

So I changed the folder Name and the Environmental variable path, then it worked. 
But maybe you should change this mistake such that not every windows user must manually change the folder name and path.

dobairoland, 2023-06-22:
@georgik Do you think it is possible that somehow a comma ended up in `IDF_PYTHON_ENV_PATH` instead of a dot producing `idf5,0_py3.9_env` (caused by the installer)?

georgik, 2023-06-22:
@dobairoland That is strange. 

The installer uses two methods: 
- read the version from the header file - https://github.com/espressif/idf-installer/blob/main/src/InnoSetup/Configuration.iss#L136
- parse version from downloaded artifact - https://github.com/espressif/idf-installer/blob/main/src/InnoSetup/Configuration.iss#L172

None of these methods use commas.

Also, the version number is strange. It should be 5.1 for rc2, not 5,0.

georgik, 2023-06-22:
The installer also did not deploy the environment because the installer is provided only with Python 3.8 or Python 3.11 in newer versions. Python mentioned in the log is 3.9.
No idf-installer was produced with the version.

dobairoland, 2023-06-22:
Thank you @georgik. My assumption was also that probably the user set the environment variable manually some time ago and forgotten about it.

closed 2023-06-22 by dobairoland

georgik, 2023-06-22:
Just for the complete record: We did test with idf-installer 2.21 and 5.1-rc2
Deployment was without issue. Python environment was created in `C:\Espressif\python_env\idf5.1_py3.11_env`.
No comma in the path to Python virtual environment.