espressif/esp-idf #11728: esp-tls: [sock=54] select() timeout

Noushadalik, 2023-06-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

External 3.3V

### What is the expected behavior?

Publish a message by connecting MQTT clients seamlessly.

### What is the actual behavior?

  we have encountered while testing the Cavli C16QS GSM board with various network providers. Specifically,we have successfully tested the board with Jio and VI networks. However, When we are connecting MQTT clients experiencing intermittent TLS connectivity problems when using Airtel SIM cards, despite being able to establish a connection to the SNTP server without any issues
ESP IDF error log mentioned below

E (29032) esp-tls: [sock=54] select() timeout
E (29032) esp-tls: Failed to open new connection
E (29032) TRANSPORT_BASE: Failed to open a new connection
E (29032) MQTT_CLIENT: Error transport connect
E (49262) MQTT: MQTT_EVENT_DISCONNECTED

### Steps to reproduce.

Connect Cavli c16QS gsm board to esp32 and run ESP ppos example code.

### Debug Logs.

```plain
verbose logs :

I (10) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (10) boot: compile time 13:46:35
I (10) boot: chip revision: 1
I (12) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (19) boot.esp32: SPI Speed      : 80MHz
I (24) boot.esp32: SPI Mode       : DIO
I (28) boot.esp32: SPI Flash Size : 4MB
I (33) boot: Enabling RNG early entropy source...
I (38) boot: Partition Table:
I (42) boot: ## Label            Usage          Type ST Offset   Length
I (49) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (56) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (64) boot:  2 app0             OTA app          00 10 00010000 00100000
I (71) boot:  3 app1             OTA app          00 11 00110000 00100000
I (79) boot:  4 spiffs           Unknown data     01 82 00210000 001f0000
I (86) boot: End of partition table
[... 488 lines cut ...]
E (25022) TRANSPORT_BASE: Failed to open a new connection
E (25022) MQTT_CLIENT: Error transport connect
I (25032) MQTT: MQTT_EVENT_ERROR
D (25032) MQTT_CLIENT: Reconnect after 10000 ms
E (25032) MQTT: MQTT_EVENT_DISCONNECTED
D (25042) MQTT_CLIENT: mqtt_enqueue id: 55933, type=3 successful
D (25042) OUTBOX: ENQUEUE msgid=55933, msg_type=3, len=145, size=145
D (25052) MQTT_CLIENT: Publish: client is not connected
D (35042) MQTT_CLIENT: Reconnecting...
I (35042) MQTT: MQTT other event id: 7
D (35042) esp-tls: host:onesensetestnewhub.azure-devices.net: strlen 36
V (35232) esp-modem: uart_get_buffered_data_len()=1
V (35232) esp-modem: uart_get_buffered_data_len()=64
V (35242) esp-modem: uart_get_buffered_data_len()=64
V (35242) esp-modem: uart_get_buffered_data_len()=44
D (35242) esp-tls: [sock=54] Resolved IPv4 address: 20.37.199.0
D (35242) esp-tls: [sock=54] Connecting to server. HOST: onesensetestnewhub.azure-devices.net, Port: 8883
E (45252) esp-tls: [sock=54] select() timeout
E (45252) esp-tls: Failed to open new connection
E (45252) TRANSPORT_BASE: Failed to open a new connection
E (45252) MQTT_CLIENT: Error transport connect
I (45262) MQTT: MQTT_EVENT_ERROR
D (45262) MQTT_CLIENT: Reconnect after 10000 ms
E (45262) MQTT: MQTT_EVENT_DISCONNECTED
```

ESP-YJM, 2023-06-29:
@Noushadalik It seems the TCP connect fail.  It seems the TCP SYN packet not successfully sent to esp_modem. It may not find a suitable router or netif to send the SYN packet.

I think you can enable lwip log. Such as LWIP_TCP_DEBUG, LWIP_IP_DEBUG, LWIP_NETIF_DEBUG and share the log.

TAMILOLI, 2023-07-17:
@ESP-YJM  Same Issues will return to me; I can shared here my LWIP log information;
IDF version : 4.4

**This code run it 2 weeks before** just now this issues happened? please suggest solution.

I (0) cpu_start: App cpu up.
I (357) cpu_start: Pro cpu start user code
I (357) cpu_start: cpu freq: 240000000
I (357) cpu_start: Application information:
I (362) cpu_start: Project name:     app_project
I (367) cpu_start: App version:      1
I (372) cpu_start: Compile time:     Jul 17 2023 19:15:29
I (378) cpu_start: ELF file SHA256:  dfe2a91043b8e219...
I (384) cpu_start: ESP-IDF:          v4.4.4-dirty
I (389) heap_init: Initializing. RAM available for dynamic allocation:
I (396) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (402) heap_init: At 3FFB4A38 len 0002B5C8 (173 KiB): DRAM
I (409) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (415) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (421) heap_init: At 4008E01C len 00011FE4 (71 KiB): IRAM
I (429) spi_flash: detected chip: generic
I (432) spi_flash: flash io: dio
I (437) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (476) GSM: Module GSM un-powered
I (977) GSM: Module GSM powered
I (1477) GSM: Module GSM PWKEY LOW
I (2577) GSM: Module GSM PWKEY HIGH
netif: netmask of interface  set to 255.0.0.0
netif: GW address of interface  set to 127.0.0.1
netif_set_ipaddr: netif address being changed
netif: added interface lo IP addr 127.0.0.1 netmask 255.0.0.0 gw 127.0.0.1
I (12587) GSM: STATE GSM_STATE_FIRSTINIT
I (12691) GSM: AT COMMAND: [AT..]
I (12894) GSM: AT RESPONSE: [..OK..]
I (12994) GSM: AT COMMAND: [ATZ..]
I (13197) GSM: AT RESPONSE: [..OK..]
I (13297) GSM: AT COMMAND: [ATE0..]
I (13500) GSM: AT RESPONSE: [..OK..]
I (13600) GSM: AT COMMAND: [AT+CFUN=4..]
I (16261) GSM: AT RESPONSE: [..OK..]
I (17361) GSM: AT COMMAND: [ATH..]
I (17564) GSM: AT RESPONSE: [..OK..]
I (17664) GSM: AT COMMAND: [AT+CFUN=1..]
I (17871) GSM: AT RESPONSE: [..OK..]
I (18971) GSM: AT COMMAND: [AT+CPIN?..]
I (19365) GSM: AT RESPONSE: [..+CPIN: READY....OK....SMS Ready..]
I (19465) GSM: AT COMMAND: [AT+CREG?..]
E (34569) GSM: AT: TIMEOUT
W (34569) GSM: Wrong response, restarting...
I (37569) GSM: Skip command: [AT..]
I (37569) GSM: Skip command: [ATZ..]
I (37569) GSM: Skip command: [ATE0..]
I (37570) GSM: Skip command: [AT+CFUN=4..]
I (37575) GSM: Skip command: [ATH..]
I (37579) GSM: Skip command: [AT+CFUN=1..]
I (37584) GSM: Skip command: [AT+CPIN?..]
I (37689) GSM: AT COMMAND: [AT+CREG?..]
I (37894) GSM: AT RESPONSE: [..+CREG: 0,1....OK..]
I (39994) GSM: AT COMMAND: [AT+CGDCONT=1,"IP","airtelgprs.com"..]
I (40203) GSM: AT RESPONSE: [..OK..]
I (40303) GSM: AT COMMAND: [AT+CGDATA="PPP",1..]
I (40508) GSM: AT RESPONSE: [..CONNECT..]
E (41508) GSM: inicializando PPPoS...
netif: netmask of interface  set to 255.255.255.255
netif: added interface pp IP addr 0.0.0.0 netmask 255.255.255.255 gw 0.0.0.0
netif: setting default interface pp
netif: GW address of interface pp set to 192.168.254.254
netif_set_ipaddr: netif address being changed
I (44700) GSM: status_cb: Connected
I (44702) GSM:    ipaddr    = 100.109.13.253
I (44707) GSM:    gateway   = 192.168.254.254
I (44712) GSM:    netmask   = 255.255.255.255
I (44717) GSM:    ip6addr   = ::
I (44722) system_api: Base MAC address is not set
I (44727) system_api: read default base MAC address from EFUSE
I (44735) main: : MQTT_EVENT_BEFORE_CONNECT
[image]
[image]
[image]

Alvin1Zhang, 2023-08-04:
Thanks for reporting and sharing the updates, feel free to reopen.

closed 2023-08-04 by Alvin1Zhang

wenjiaxing974959, 2023-12-22:
怎么解决的，我现在也是遇到的同样的问题，esp-tls: [sock=54] select() timeout；TRANSPORT_BASE: Failed to open a new connection: 32774；log和上述是一致的

ESP-YJM, 2023-12-22:
这个错误的原因是 TCP SYN 包没有人回，所以你需要确认下服务器是否有问题，设备发的 TCP SYN 包是否发送成功，是否发送正确。

wenjiaxing974959, 2023-12-22:
怎么查看这个设备发的 TCP SYN 包呢

ESP-YJM, 2023-12-22:
抓包或者打开 lwip 层 TCP 相关的日志。

vitara-campbell, 2024-07-07:
> @TAMILOLI It seems the TCP SYN is no response. Could you confirm that the PPPoS module is access to Internet. Besides, you can ping your PPP IP to check the PPP connection is OK.

How is the PPPoS module accessed in the menuconfig?