espressif/esp-idf #11730: esp-va-sdk

prabhu012, 2023-06-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4, v4.2, v4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

NA

### What is the actual behavior?

Hi, I am working with ESP-VA-SDK, and I am fairly new to ESP-IDF. I am using the ESP32-S3 Eye v2.2. When I build the code, I am getting this error
and our esp-idf version is 4.4.4 and I also tried esp-idf version V_4.2.4 and V_4.1.4

Could anyone provide help on how to solve this error

### Steps to reproduce.

1. Step
2. Step
3. Step
...
I have put the esp-va-sdk in component directry of esp-idf  and trying to build the amazon alexa example code

### Build or installation Logs.

```plain
C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>idf.py build
Executing action: all (aliases: build)
Running cmake in directory C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant"...
-- AUDIO_BOARD_PATH not defined
-- Configuring done

CMake Error
The detected version of Ninja () is less than the version of Ninja required
by CMake (1.3).

CMake Warning:
Manually-specified variables were not used by the project:

CCACHE_ENABLE
ESP_PLATFORM
PYTHON_DEPS_CHECKED
CMake Generate step failed. Build files cannot be regenerated correctly.
cmake failed with exit code 1

C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>cmake --version
cmake version 3.23.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>ninja --version
1.10.2

C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>
```

prabhu012, 2023-06-24:
Could anyone provide help on how to solve this error its argent

sudeep-mohanty, 2023-06-28:
Hi @prabhu012,
Apologies for the delayed response.
It looks like the real issue here is that you haven't set the  `AUDIO_BOARD_PATH` in your environment variables. Could you try to export it and rebuild? 

```
$ export AUDIO_BOARD_PATH=/path/to/esp-va-sdk/components/audio_hal/audio_board/audio_board_vaquita_dspg
```

9305128621, 2023-06-30:
Hi @sudeep-mohanty , Thank you for your response.

I added the path according to your suggested path, and it was added while I build this project it shows below two errors.

1. include could not find load file.
2. Failed to resolve component 'esp_codec'.

```
XYZ:~/esp_voice_asistence/esp-va-sdk/examples/amazon_aia$ export AUDIO_BOARD_PATH=esp_voice_asistence/esp-va-sdk/components/audio_hal/audio_board/audio_board_esp_eye

XYZ:~/esp_voice_asistence/esp-va-sdk/examples/amazon_aia$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/whitelion/esp_voice_asistence/esp-va-sdk/examples/amazon_aia/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/whitelion/esp_voice_asistence/esp-va-sdk/examples/amazon_aia"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /home/whitelion/esp_4.3/esp/esp-idf/components/esp-va-sdk does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
CMake Error at CMakeLists.txt:23 (include):
  include could not find load file:

    esp_voice_asistence/esp-va-sdk/components/audio_hal/audio_board/audio_board_esp_eye/audio_board.cmake

-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /home/whitelion/esp_4.3/esp/esp-idf/tools/cmake/build.cmake:193 (message):
  Failed to resolve component 'esp_codec'.
Call Stack (most recent call first):
  /home/whitelion/esp_4.3/esp/esp-idf/tools/cmake/build.cmake:224 (__build_resolve_and_add_req)
  /home/whitelion/esp_4.3/esp/esp-idf/tools/cmake/build.cmake:472 (__build_expand_requirements)
  /home/whitelion/esp_4.3/esp/esp-idf/tools/cmake/project.cmake:371 (idf_build_process)
  CMakeLists.txt:37 (project)

-- Configuring incomplete, errors occurred!
See also "/home/whitelion/esp_voice_asistence/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

prabhu012, 2023-06-30:
Hi @sudeep-mohanty , Thank you for your response.

I added the path according to your suggested path, and it was added while I build this project it shows below two errors.

1. include could not find load file.
2. Failed to resolve component 'esp_codec'.

```
XYZ:~/esp_voice_asistence/esp-va-sdk/examples/amazon_aia$ export AUDIO_BOARD_PATH=esp_voice_asistence/esp-va-sdk/components/audio_hal/audio_board/audio_board_esp_eye

XYZ:~/esp_voice_asistence/esp-va-sdk/examples/amazon_aia$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/whitelion/esp_voice_asistence/esp-va-sdk/examples/amazon_aia/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/whitelion/esp_voice_asistence/esp-va-sdk/examples/amazon_aia"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /home/whitelion/esp_4.3/esp/esp-idf/components/esp-va-sdk does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
CMake Error at CMakeLists.txt:23 (include):
  include could not find load file:

    esp_voice_asistence/esp-va-sdk/components/audio_hal/audio_board/audio_board_esp_eye/audio_board.cmake

-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/whitelion/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /home/whitelion/esp_4.3/esp/esp-idf/tools/cmake/build.cmake:193 (message):
  Failed to resolve component 'esp_codec'.
Call Stack (most recent call first):
  /home/whitelion/esp_4.3/esp/esp-idf/tools/cmake/build.cmake:224 (__build_resolve_and_add_req)
  /home/whitelion/esp_4.3/esp/esp-idf/tools/cmake/build.cmake:472 (__build_expand_requirements)
  /home/whitelion/esp_4.3/esp/esp-idf/tools/cmake/project.cmake:371 (idf_build_process)
  CMakeLists.txt:37 (project)

-- Configuring incomplete, errors occurred!
See also "/home/whitelion/esp_voice_asistence/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

sudeep-mohanty, 2023-06-30:
@prabhu012 You have not set the `AUDIO_BOARD_PATH` correctly. Your system environment variable must be set to the complete file path. As I can see, `esp_voice_asistence/esp-va-sdk/components/audio_hal/audio_board/audio_board_esp_eye` is not being resolved by the build-system since this path does not exist. Please make sure you set the environment variable correctly.

sudeep-mohanty, 2023-07-04:
Hi @prabhu012, Since the issue is not seen locally, I'm closing this ticket. Please feel free to open a new ticket if you see more problems.

closed 2023-07-04 by sudeep-mohanty