espressif/esp-idf #11731: system/console/basic example input over USB does not work on ESP32-S3

morgana2313, 2023-06-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1128-g03d4fa2869

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3

### Power Supply used.

USB

### What is the expected behavior?

The `examples/system/console/basic` input via usb-otg seems broken; any input over USB is not seen by the esp32-s3.

Using the uart RX/TX console works just fine. Input given on the RX uart input even shows the expected output over the USB console.

Expected behaviour when I press enter:

```
esp32s3> ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037bc2e
0x4037bc2e: esp_cpu_wait_for_intr at /Users/jasper/esp/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1240
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa5c
load:0x403cc700,len:0x2d18
entry 0x403c9890
I (206) cpu_start: Multicore app
I (206) cpu_start: Pro cpu up.
I (207) cpu_start: Starting app cpu, entry point is 0x4037560c
0x4037560c: call_start_cpu1 at /Users/jasper/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (225) cpu_start: Pro cpu start user code
I (225) cpu_start: cpu freq: 160000000 Hz
I (225) cpu_start: Application information:
I (228) cpu_start: Project name:     console
I (233) cpu_start: App version:      de1ee52-dirty
I (238) cpu_start: Compile time:     Jun 23 2023 17:01:11
I (244) cpu_start: ELF file SHA256:  be07aaa1f...
I (250) cpu_start: ESP-IDF:          v5.2-dev-1128-g03d4fa2869
I (256) cpu_start: Min chip rev:     v0.0
I (261) cpu_start: Max chip rev:     v0.99 
I (266) cpu_start: Chip rev:         v0.1
I (270) heap_init: Initializing. RAM available for dynamic allocation:
I (278) heap_init: At 3FCA19B0 len 00047D60 (287 KiB): DRAM
I (284) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (290) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (297) heap_init: At 600FE070 len 00001F78 (7 KiB): RTCRAM
I (304) spi_flash: detected chip: gd
I (307) spi_flash: flash io: dio
W (311) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (325) sleep: Configure to isolate all GPIO pins in sleep state
I (331) sleep: Enable automatic switching of GPIO sleep configuration
I (338) app_start: Starting scheduler on CPU0
I (343) app_start: Starting scheduler on CPU1
I (343) main_task: Started on CPU0
I (353) main_task: Calling app_main()
I (363) example: Command history enabled

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.

Line editing and history features are disabled.

On Windows, try using Putty instead.
esp32s3> I (873) main_task: Returned from app_main()  <======= ENTER gives a new prompt.
esp32> 
``` 

### What is the actual behavior?

```
esp32s3> ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037bc2e
0x4037bc2e: esp_cpu_wait_for_intr at /Users/jasper/esp/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1240
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa5c
load:0x403cc700,len:0x2d18
entry 0x403c9890
I (206) cpu_start: Multicore app
I (206) cpu_start: Pro cpu up.
I (207) cpu_start: Starting app cpu, entry point is 0x4037560c
0x4037560c: call_start_cpu1 at /Users/jasper/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (225) cpu_start: Pro cpu start user code
I (225) cpu_start: cpu freq: 160000000 Hz
I (225) cpu_start: Application information:
I (228) cpu_start: Project name:     console
I (233) cpu_start: App version:      de1ee52-dirty
I (238) cpu_start: Compile time:     Jun 23 2023 17:01:11
I (244) cpu_start: ELF file SHA256:  be07aaa1f...
I (250) cpu_start: ESP-IDF:          v5.2-dev-1128-g03d4fa2869
I (256) cpu_start: Min chip rev:     v0.0
I (261) cpu_start: Max chip rev:     v0.99 
I (266) cpu_start: Chip rev:         v0.1
I (270) heap_init: Initializing. RAM available for dynamic allocation:
I (278) heap_init: At 3FCA19B0 len 00047D60 (287 KiB): DRAM
I (284) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (290) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (297) heap_init: At 600FE070 len 00001F78 (7 KiB): RTCRAM
I (304) spi_flash: detected chip: gd
I (307) spi_flash: flash io: dio
W (311) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (325) sleep: Configure to isolate all GPIO pins in sleep state
I (331) sleep: Enable automatic switching of GPIO sleep configuration
I (338) app_start: Starting scheduler on CPU0
I (343) app_start: Starting scheduler on CPU1
I (343) main_task: Started on CPU0
I (353) main_task: Calling app_main()
I (363) example: Command history enabled

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.

Line editing and history features are disabled.

On Windows, try using Putty instead.
esp32s3>                                 <======= ENTER is not seen.
```

### Steps to reproduce.

1. Build & flash  `examples/system/console/basic` 
2. Monitor esp32-s3 output
3. Press enter 

### More Information.

I've tested this with multiple esp32s3 boards:

* LilyGo T SIM7080G S3
* Unexpected Maker TinyS3/FeatherS3/S3Pro

Using Vscode monitor, but also with `idf.py monitor`. I use 'UART' as flash method.

```
.espressif/python_env/idf5.2_py3.11_env/bin/python ~/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem2101 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32sbash-3.2$ .espressif/python_env/idf5.2_py3.11_env/bin/python ~/esp/esp-idf/too200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /Users/jasper/mp-v5-flexy/projects/basic_console/build/console.elf
```

morgana2313, 2023-06-23:
This may be related to #9233, however the command prompt appears now.

igrr, 2023-06-23:
Could you please attach your sdkconfig file?

morgana2313, 2023-06-23:
Solved by adding `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y` to sdkconfig.

May I suggest adding this to the `examples/system/console/README.md` ?

igrr, 2023-06-23:
We do mention this in the readme of the console example: https://github.com/espressif/esp-idf/tree/master/examples/system/console/basic#using-with-usb_serial_jtag

morgana2313, 2023-06-24:
Ah, my bad. I didn't realise my USB console had anything to do with JTAG. May I suggest:

```diff
- On chips with USB_SERIAL_JTAG peripheral, console example can be used over the USB serial port.
- On chips with USB_SERIAL_JTAG peripheral  (esp32-s2, esp32-s3), console example can be used over the USB serial port.

* First, connect the USB cable to the USB_SERIAL_JTAG interface.
-* Second, run idf.py menuconfig and enable CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG option.
+* Second, run idf.py menuconfig and enable CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG option, otherwise console input via USB won't work.
```

igrr, 2023-06-25:
Console itself doesn't have anything to do with JTAG, USB_SERIAL_JTAG is the name of the peripheral in ESP32-C3, ESP32-S3, ESP32-C6, ESP32-H2 which provides a fixed-function USB composite device with two features: CDC (serial port) and a vendor-specific JTAG adapter.

Your proposed edit suggests that you might be confusing it with the USB_OTG peripheral, which indeed is available on ESP32-S2 and ESP32-S3. It is a programmable USB peripheral which can be used to implement either a device or a host.

Thanks for the suggestions, we'll update the troubleshooting section of the readme file with this specific situation.

closed 2023-07-14 by espressif-bot