espressif/esp-idf #11732: python.exe doesn't support asyncio. The issue can be worked around by re-running idf.py with the "--no-hints" argument.

Yolnit, 2023-06-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1-rc2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I should be able to flash a code without any error (the helloworld from examples) and see in the terminal the message every 10 secondes.

### What is the actual behavior?

For the build it seems there is no problem but for running i got this message in the console : 
C:\esp-idf-v5.1-rc2\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
import pkg_resources
Executing action: flash
Running ninja in directory C:**\eclipse-workspace_esp32\test4\build
Executing "ninja flash"...
ERROR: C:\Espressif\python_env\idf5.1_py3.7_env\Scripts\python.exe doesn't support asyncio. The issue can be worked around by re-running idf.py with the "--no-hints" argument.

### Steps to reproduce.

1. Step click on build (Eclipse 2022-06 version)
2. Step click on run and make sure the correct target is selected
3. Seeing the result in the terminal 

### Build or installation Logs.

```plain
none
```

### More Information.

none

peterdragun, 2023-06-26:
I believe the log is clear here. To use hints you need python that supports asyncio. (Hints are enabled by default) If your python environment does not support it you can run idf.py without hints by adding `--no-hints` option. The other option is to update to newer python (at least 3.8) with support for asyncio on Windows.

> C:\esp-idf-v5.1-rc2\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. 

Just to be clear, this deprecation warning is not related to this issue and is handled separately. This is just a warning and it is not causing any issues.

dobairoland, 2023-06-27:
The recommended way is to use the Windows Tool Installer: https://dl.espressif.com/dl/esp-idf/ It can deploy a compatible Python environment. 

> idf5.1_py3.7_env

Python 3.7 was never used by our installer. That is why I think you haven't used it.

Yolnit, 2023-06-29:
Thank you for your help.

It was the python version the one and only problem in my case.

Have all a good day and wish you the best :)

closed 2023-07-14 by espressif-bot