espressif/esp-idf #11733: ESP32C6 Unable to join existing Zigbee PRO network with HA_on_off_light example

lordldx, 2023-06-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1128-g03d4fa2869

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-C6-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

I expected that the device would join the network.

### What is the actual behavior?

The device fails during network steering

### Steps to reproduce.

1. in VSCode: run ESP-IDF: Show Examples Projects
2. Create a new example project of HA_on_off_light
3. Set correct COM port
4. Set target device to esp32c6
5. Build
6. Flash
7. Monitor

### Debug Logs.

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40805aac
0x40805aac: rv_utils_wait_for_intr at C:/Users/dieryckxl/esp/esp-idf/components/riscv/include/riscv/rv_utils.h:33
 (inlined by) esp_cpu_wait_for_intr at C:/Users/dieryckxl/esp/esp-idf/components/esp_hw_support/cpu.c:119        

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x181c
load:0x4086c410,len:0xd44
load:0x4086e610,len:0x2df8
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2-dev-1128-g03d4fa2869-dirty 2nd stage bootloader
I (23) boot: compile time Jun 23 2023 23:40:19
I (24) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 2MB
[... 180 lines cut ...]
I (1053) ESP_ZB_ON_OFF_LIGHT: ZDO signal: 23, status: -1
I (1053) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (1053) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (3713) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
```

### More Information.

I sniffed the network traffic (TI CC2531 + ZBOSS sniffer + wireshark) and the join seems to start properly:
-> First beacon request and a beacon from the ZC
-> Next: Association request from the ZED (my esp32c6), and an association response from the ZC. The response indicates success and assigns a network address to the ZED.
-> Next, the ZC transmits the transport key. ZED emits an ACK, but that's it. No more communication from the ZED from then on.

From then on, the whole process repeats every two seconds.

I observed the same behaviour on v5.1-rc2.

When looking at the network steering spec (from the Zigbee BDB spec 2.1, it seems that something goes wrong in Step 9. Looks to me as if the esp32c6 answers "no" to the "Was the network key received succesfully" question. Even though I see it being transmitted and ACKed in the traffic.
[image]

I'm honestly a bit disappointed in the amount of logging the zigbee api emits.
I've configured Default log verbosity to "Verbose", but the only zigbee related output I'm seeing is the Information logging emited by the HA_on_off_light application code itself. Some lower-level zb or zboss logging would be really helpful here!

I can't seem to attach the pcap file directly in this issue, but here's a screenshot of it. Let me know if you need to see more details:
[image]

xieqinan, 2023-06-28:
Hi @lordldx ,
The log message of the ESP Zigbee SDK will be provided in more detail in the upcoming release version. Joining network failure is a tricky issue, and it is difficult to determine the exact reasons solely based on the aforementioned message. Conducting additional tests may prove helpful in resolving this problem.

The entire Wireshark message associated with the issue serves as a vital resource for us to address the bug effectively.

Could you please attempt to use the `Ha_on_off_switch` example as the Zigbee coordinator to establish the network? Additionally, kindly provide us with the Wireshark message captured during the network formation process.

xieqinan, 2023-06-28:
Hi @lordldx ,
Can you try use `esp_zb_secur_TC_standard_preconfigure_key_set()` to set the `HA_on_off_light` device and keep the same Trust Center standard key in your zigbee network? 

The corresponding description for the transport key can be found in Chapter 4.4.2 of the Zigbee specification.
[image]

lordldx, 2023-06-28:
Hi @xieqinan 

Creating the network seems to work. See file "create_network-v5.2-dev-1128-g03d4fa2869.pcapng" in the attached zip.
Not sure if you'll be able to derive much from it, since the data is encrypted and I'm not sure where I can find the security key for the network the esp32c6 creates. (If you can point me to this, I can also provide you with an unencrypted log)

I also added the wireshark logging of my initial report. See file "join_network_fail-v5.2-dev-1128-g03d4fa2869.pcapng" in the attached zip.

Please let me know how I can assist any further. I'd love to get this fixed!

[wireshark_dumps.zip](https://github.com/espressif/esp-idf/files/11892875/wireshark_dumps.zip)

xieqinan, 2023-06-28:
Hi @lordldx ,

In a general Zigbee network, the security key can typically be obtained from the transport key section.

I am confused about the process of joining a network on your Zigbee network. Firstly, I noticed that the Zigbee coordinator/router sends a transport key packet with a source address of ff:ff:ff:ff:ff:ff:ff:ff to the Zigbee end device. If your Zigbee network is centralized, the specification requires the source address to be the local device address.
[image]

f your Zigbee network is distributed, please verify your network using the following guidelines.

 [image]

So what is your network topology?

Based on the Wireshark log you provided, it seems that your network topology is centralized since only the Zigbee coordinator is distributing the short address (0x0000). However, I am unsure how to explain the presence of the long address (ff:ff:ff:ff:ff:ff:ff:ff). Therefore, we would need more information about your Zigbee network to clarify this.
 
Besides, it is helpful to tell us whether network security feature is supported in your network Pro?

lordldx, 2023-06-28:
Hi @xieqinan 

Thank you for your fast responses and thourough investigation.
To answer your questions:

1. Can you try use `esp_zb_secur_TC_standard_preconfigure_key_set()`?
I tried, but I think I'm doing something wrong, because I get even less traffic.
Now the esp32c6 only emits a beacon request and that's it.
I may have misused the function above though, I'm still very uncomfortable with the esp_zb api...
I used it like this:
```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack with Zigbee end-device config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    uint8_t key[16] = {0xb6, 0x00, 0x08, 0x56, 0xbe, 0x74, 0x7a, 0xb9, 0xf1, 0x50, 0xed, 0x5f, 0x82, 0x8b, 0x2c, 0xc0};
    esp_zb_secur_TC_standard_preconfigure_key_set(&key);

    /* set the on-off light device config */
    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    esp_zb_device_register(esp_zb_on_off_light_ep);
    esp_zb_device_add_set_attr_value_cb(attr_cb);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

2. What is your network topology?
It's a centralized network. The simplest possible: I have just a ZC which is also an actuator, and then a ZED which is a battery operated button that turns the light on and off. So now I'm just trying to add the esp32c6 as a second ZED.

4. Is network security feature supported in your network Pro?
I actually don't know. The vendor of the ZC I'm using is very scarce with information.
I'm using a [bticino 3571](https://ecataleg.be/nl/produit/3571), which is really just a relay with a Zigbee interface.
I bought it in 2019 and it has been discontinued by the manufacturer, so I guess the technology is at least five years old.
I tried getting some additional information on how to interface with it from the manufacturer, but I don't get any response.
Is there any way I can figure this out from the traffic? From the beacon, or perhaps during initial network creation?
I have no problem with completely resetting and reconfiguring the device if this would be any help.

Oh, and I haven't mentioned this before, but it might be important.
I'm referencing esp-zigbee-lib v0.6.2, which is higher than the v0.5.0 with which the HA_on_off_light examples is preconfigiured:
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zigbee-lib: "~0.6.2"
  espressif/esp-zboss-lib: "~0.4.0"
  espressif/led_strip: "~2.0.0"
  ## Required IDF version
  idf:
    version: ">=5.0.0"
```

xieqinan, 2023-07-04:
Hi @lordldx ,
This problem is triggered when the end device receives an unsecured frame on the application support sublayer. According to the specifications 4.4.2.3 section, `If the device is operating in the joined and authorized state it may accept a NWK broadcast transport key command with Key type field set to 0x01 (that is, network key) where the message has no APS encryption.`

However, the Wireshark log from the [Bticino 3571](https://ecataleg.be/nl/produit/3571) indicates that the frame is not encrypted, unsafe, and not using the broadcast address.
[image]

Therefore, if the coordinator can enable the transport key security feature through programming, the issue can be resolved, and the device will be able to function properly.

xieqinan, 2023-07-19:
Hi @lordldx ,
From Figure 4-21, we can deduce that the Zigbee end device will remain in the `Joined but unauthorized` state upon receiving the `association response.` Only when the ZED receives the Network Key, it will transition into the "Join and Authorized" state. 

Additionally, the Zigbee R22 certification requires the joining ZED to have the APS security key, and it allows the ZED to join the network without APS security when it is in the `Joined and Authorized` state. In other words, if the device has already joined the network, there is no need to perform an exchange of network keys during the next joining process.
 
Based on this information, I suspect that the "bticino 3571" may not be compliant with the R22 certification, or this `bticino 3571` has been pre-configured at the factory to be already eligible for joining its network.

chshu, 2023-11-10:
@lordldx please re-open if any follow up comments.