espressif/esp-idf #11734: Why the idf.py encrypted-app-flash update the app will let the system can not running normally

WangShuoran, 2023-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am use ESP-IDF v4.4.4 with ESP32C3, my question can not let `idf.py encrypted-app-flash` corrently, I do not know why, so I let this question is generate question. I am use flash_encrypt example.

I use the Flash encrypted with developed mode, but when I use `idf.py encrypted-app-flash` update the app for ESP32C3. But if I flash the system, I find the ESP32C3 will show error:
```

```

Does this command can not flash the updated app? 

And the example can not write encrypt flash partition (In example is storage partition), because this example write the storage partition and restart will show error:
```

```

WangShuoran, 2023-06-25:
I find the NVS encryption partition causing the problem, when I use Flash encryption with Devlop mode, and I follow the ESP-FAQ nvs send:
[image]

And, I use `idf.py erase-flash` to erase all Flash content, and use `idf.py encrypted-flash monitor` to write app and other partition to  
ESP32C3.
The first running:
```
PS D:\mine\nvs\flash_encryption> idf.py erase-flash
Executing action: erase-flash
Serial port COM11
Connecting....
Detecting chip type... ESP32-C3
Running esptool.py in directory d:\mine\nvs\flash_encryption\build
Executing "C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py -p COM11 -b 460800 --before default_reset --after no_reset --chip esp32c3 erase_flash"...
esptool.py v3.3.2
Serial port COM11
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:77:4e:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 19.6s
Staying in bootloader.
Done
PS D:\mine\nvs\flash_encryption> idf.py encrypted-flash monitor
Executing action: encrypted-flash
Serial port COM11
Connecting....   
Detecting chip type... ESP32-C3
Running ninja in directory d:\mine\nvs\flash_encryption\build
Executing "ninja encrypted-flash"...
[1/5] cmd.exe /C "cd /D D:\mine\nvs\flash_encryption\build\esp-idf\esptool_py && C:\Espressif\python...n/build/partition_table/partition-table.bin D:/mine/nvs/flash_encryption/build/flash_encryption.bin" 
flash_encryption.bin binary size 0x2a9c0 bytes. Smallest app partition is 0x100000 bytes. 0xd5640 bytes (83%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D D:\mine\nvs\flash_encryption\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.3/components/partition_table/check_sizes.py --offset 0xf000 bootloader 0x0 D:/mine/nvs/flash_encryption/build/bootloader/bootloader.bin"
Bootloader binary size 0x7930 bytes. 0x76d0 bytes (50%) free.
[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.3\components\esptool_py && C:\Espressif...yption/build" -P C:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/run_serial_tool.cmake" 
esptool.py esp32c3 -p COM11 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB --encrypt 0x0 bootloader/bootloader.bin 0x100000 flash_encryption.bin 0xf000 partition_table/partition-table.bin
esptool.py v3.3.2
Serial port COM11
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:77:4e:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00007fff...
Flash will be erased from 0x00100000 to 0x0012afff...
Flash will be erased from 0x0000f000 to 0x0000ffff...

WARNING: - compress and encrypt options are mutually exclusive
Will flash bootloader/bootloader.bin uncompressed
Writing at 0x00000000... (50 %)
Writing at 0x00004000... (100 %)
Wrote 32768 bytes at 0x00000000 in 0.9 seconds (283.5 kbit/s)...

WARNING: - compress and encrypt options are mutually exclusive
Will flash flash_encryption.bin uncompressed
Writing at 0x00100000... (9 %)
[... 93 lines cut ...]
And the next running:
```
�ѽɕ�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5978,len:0x29f0
load:0x403cc710,len:0x930 
load:0x403ce710,len:0x45bc
Checksum failure. Calculated 0xd8 stored 0xd3
ets_main.c 333
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5978,len:0x29f0
load:0x403cc710,len:0x930
load:0x403ce710,len:0x45bc
Checksum failure. Calculated 0xd8 stored 0xd3
ets_main.c 333
```
Why the ESP32 generate NVS encryption partition will let the Flash encryption failure, Does the release mode have same question? Does this must need our NVS encryption partition write to the ESP32C3 first?

mahavirj, 2023-06-26:
@WangShuoran 

This is quite strange that device is unable to boot in subsequent reset. Can you please share detailed steps to recreate this issue? Also please share EFuse summary of this device.

mahavirj, 2023-07-03:
@WangShuoran Any further update on this?

closed 2023-07-07 by Alvin1Zhang