espressif/esp-idf #11735: How to prevent component from getting registered twice in diamond dependency

higaski, 2023-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

vESP-IDF v5.2-dev-823-g903af13e84

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I have a library which defines an ESP-IDF component as additional component like this

```
if(ESP_PLATFORM)
  idf_component_register(...)
endif()
```

This component is just an addition to this library and it makes sense for other projects to use this library without using the component. The problem is, that once I build another library on top of the one which defines the component and then include both libraries in some top-level project (therefor creating a diamond dependency), CMake gives me an error message that there already exists a component with this name.

It looks like it tries to register the component twice. Apparently that's an issue and for some reason it can't figure out that the components are actually the same? How can I fix this issue?

The CMakeList.txt files of the libraries I'm talking about can be found here:
https://github.com/ZIMO-Elektronik/DCC/blob/master/CMakeLists.txt
https://github.com/ZIMO-Elektronik/DCC_EIN/blob/master/CMakeLists.txt

As you can see the second library has a dependency to the first one.

### What is the actual behavior?

Described above. Plain CMake can handle such cases.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

higaski, 2023-06-26:
I've solved the issue by conditionally including the library based on whether the target already exists.

```cmake
if(NOT TARGET DCC::DCC)
  ... 
endif()
```

higaski, 2023-06-26:
Ok, turns out that this wasn't enough. You still need to make sure that the component isn't registered twice...
I've now also added a check for `CMAKE_BUILD_EARLY_EXPANSION` in the original library which registers the component.

```cmake
if(ESP_PLATFORM AND CMAKE_BUILD_EARLY_EXPANSION)
  ...
endif()
```

This is another great example how the whole component shenanigans works against CMake and not with it. CMake is already terrible enough... it makes you jump through so many hoops to get things working. The component system on the other hand... it sets the hoop on fire and adds a crocodile pit under neath. :wink:

ESP-Marius, 2023-06-29:
> This is another great example how the whole component shenanigans works against CMake and not with it. CMake is already terrible enough... it makes you jump through so many hoops to get things [...]

Yeah, we are aware that the component layer we've built on top of CMake do cause some confusion and frustration for users that are trying to integrate it together with a "normal" CMake project.

Finding ways to mitigate this/refactoring the build system to have less "surprises" is on our todo list, so hopefully in the future we will have less issues related to this.

 >  Another huge problem is that you can't name components. 

Same here, it is in our backlog to add a way to name components.

Sorry for not being able to give you any useful answers right now, i'll try to see if we can find someone that can assist you!

higaski, 2023-07-09:
For some reason the component manager does not seem to have an issue with multiple `__idf_src` components even if they are in the same ("global") namespace. I've just included a second selfmade component to my manifest file with the same folder layout but it still works. I would not have expected that to be honest. This makes naming components less of an issue.