espressif/esp-idf #11736: rgb_panel example no work with lcd panel BM043SE2-K28 and esp32s3 board

r126401, 2023-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I import proyect rgb panel about BM043SE2-K28 with board esp32s3 from luatos and psram 8M but lcd no work. Backlight is off and I have intent all posibilities. The info about board is https://wiki.luatos.com/chips/esp32s3/index.html.

¿Is necessary additional commands in order work fine the lcd?. The driver id is nv3047 and the pins configuration is :

#define EXAMPLE_LCD_PIXEL_CLOCK_HZ     (9 * 1000 * 1000)
#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL  1
#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL
#define EXAMPLE_PIN_NUM_BK_LIGHT       18
#define EXAMPLE_PIN_NUM_HSYNC          9
#define EXAMPLE_PIN_NUM_VSYNC          13
#define EXAMPLE_PIN_NUM_DE             8
#define EXAMPLE_PIN_NUM_PCLK           0
#define EXAMPLE_PIN_NUM_DATA0          12 // B0
#define EXAMPLE_PIN_NUM_DATA1          6 // B1
#define EXAMPLE_PIN_NUM_DATA2          11 // B2
#define EXAMPLE_PIN_NUM_DATA3          7 // B3
#define EXAMPLE_PIN_NUM_DATA4          10 // B4
#define EXAMPLE_PIN_NUM_DATA5          15 // G0
#define EXAMPLE_PIN_NUM_DATA6          3 // G1
#define EXAMPLE_PIN_NUM_DATA7          14 // G2
#define EXAMPLE_PIN_NUM_DATA8          4 // G3
#define EXAMPLE_PIN_NUM_DATA9          5 // G4
#define EXAMPLE_PIN_NUM_DATA10         48 // G5
#define EXAMPLE_PIN_NUM_DATA11         47  // R0
#define EXAMPLE_PIN_NUM_DATA12         1  // R1
#define EXAMPLE_PIN_NUM_DATA13         17 // R2
#define EXAMPLE_PIN_NUM_DATA14         2 // R3
#define EXAMPLE_PIN_NUM_DATA15         16 // R4
#define EXAMPLE_PIN_NUM_DISP_EN        -1

¿Can you help me, please?

suda-morris, 2023-06-25:
@r126401 I think the NV3047's  RGB interface can't work our-of the box before you do proper initialization by SPI

closed 2023-06-26 by espressif-bot

r126401, 2024-04-03:
Thanks, finally the problem was the circuit DC-Boost. I solved finally ant the lcd works fine.