espressif/esp-idf #11738: mbedtls_ssl_handshake server stucks in failing with MBEDTLS_ERR_RSA_VERIFY_FAILED error after some point

HamzaHajeir, 2023-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v4.4

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32DevKit1

### Power Supply used.

USB

### What is the expected behavior?

I host TLS webserver using [LwIP ALTCP](https://www.nongnu.org/lwip/2_1_x/group__altcp__api.html) TLS - MBEDTLS port, I expect every client (browser) to be able to connect anytime. 

Essentially, It does behave okay to some extent, making handshakes okay and communicating with no issues.
The problem starts after some amount of time/process, denying any new connections, even from browsers were already connected (By the refresh button).

I can assure that the heap memory is large enough for such a communication.

The error which is returned by the `mbedtls_ssl_handshake` when the error occur is fixed at -17280 (-0x4380) (MBEDTLS_ERR_RSA_VERIFY_FAILED).

Before this point, I noticed some times it's preceeded by error code -17168 (-0x4310), which I've tried to dissolve the code meaning but failed to, nearest suggestion is (MBEDTLS_ERR_RSA_PRIVATE_FAILED | MBEDTLS_ERR_MPI_NEGATIVE_VALUE).

mbedTLS version is 2.28.3

Is there any hint for what's going wrong?
With thanks,

### What is the actual behavior?

Log shows:
> IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3542: mbedtls_pk_sign() returned -17280 (-0x4380)

Or in the other described case:
> IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3542: mbedtls_pk_sign() returned -17168 (-0x4310)

### Steps to reproduce.

Here's the running ALTCP MBEDTLS Code, pointing to the handshake API call:

https://github.com/HamzaHajeir/esp-lwip/blob/2.1.2-esp-patch/src/apps/altcp_tls/altcp_tls_mbedtls.c#L278

### Debug Logs.

```plain
Example of a complete verbose log:
(You can expect other data being interfered with UART port at printing, those which are printed from the other core).

00:31:38.711 > IDF/components/mbedtls/mbedtls/library/ssl_tls.c:5804: => handshake
00:31:38.713 > H4AT:1: H=58248 M=42996 B=42996 S=5292 NEW CONNECTION 0x3ffdeb8c --> IDF/components/mbedtls/mbedtls/library/ssl_srv.c:4511: server state: 0
[... 275 lines cut ...]
00:31:39.936 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3377:  d8 70 7a 3f a6 53 ec 31 e9 cf 5e 6e 3e 81 83 a9
00:31:39.937 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3377: value of 'ECDH: Q(Y)' (383 bits) is:
00:31:39.938 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3377:  47 46 41 b3 5a c0 3e 16 b8 cb 31 94 fc 6e e0 79
00:31:39.939 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3377:  6d 41 f8 e8 9a 53 fe 4a de 64 ae 38 63 f7 a1 b4
00:31:39.941 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3377:  bd e6 f3 4e a0 6f 17 36 7f 39 68 9e 24 3c 3c 06
00:31:39.942 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3438: pick hash algorithm 6 for signing
00:31:39.943 > IDF/components/mbedtls/mbedtls/library/ssl_tls.c:7575: Perform mbedtls-based computation of digest of ServerKeyExchange
00:31:39.944 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3474: dumping 'parameters hash' (32 bytes)
00:31:39.945 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3474: 0000:  55 6e 08 d5 49 ea 86 99 3e 1a 89 da cd c8 16 77  Un..I...>......w
00:31:39.946 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3474: 0010:  ee 74 15 b9 28 53 09 f5 2c de 7c 3c b3 26 96 15  .t..(S..,.|<.&..
00:31:40.031 > SLOG: tick 0x80000000 206
00:31:40.033 > SLOG: upTime 0x00004000   0 00:03:26
00:31:40.072 > H=47440 M=42996 S=5292 
00:31:40.073 > SLOG: heap 0x00004000 47492
00:31:40.342 > IDF/components/mbedtls/mbedtls/library/ssl_srv.c:3542: mbedtls_pk_sign() returned -17280 (-0x4380)
00:31:40.344 > IDF/components/mbedtls/mbedtls/library/ssl_tls.c:5815: <= handshake
00:31:40.345 > tcp_recved: received 517 bytes, wnd 5744 (0).
00:31:40.345 > mbedtls_ssl_handshake failed: -17280
```

### More Information.

Building was done using espressif/esp32-arduino-lib-builder tools, with patched LwIP code (linked above).

Meanwhile this issue, the MCU is able to reconnect to a MQTT server by TLS with successful handshakes.

HamzaHajeir, 2023-07-02:
Here's an issue raised in parallel to mbedTLS, I'm not sure where the issue but suspecting either mbedTLS or LwIP ALTCP TLS port, LwIP comes with with less probability because I've did trace the memory alloc/free there.

https://github.com/Mbed-TLS/mbedtls/issues/7833

AdityaHPatwardhan, 2023-07-03:
Hi @HamzaHajeir 

Looking at the the context, it looks related to memory leaking/corruption.
Can you share your sdkconfig file?
What are the H, M, S values? are those the Free heap and Minimum heap values?

Also can you share the free heap and minimum free heap sizes at the time of the issue?

Also, you mentioned that the previously connected devices continue the connection without any problem, Have you tried disconnecting those clients and trying to establish the connection once again after you have faced the above issue?

HamzaHajeir, 2023-07-03:
Hi @AdityaHPatwardhan,

> Looking at the the context, it looks related to memory leaking/corruption. Can you share your sdkconfig file? What are the H, M, S values? are those the Free heap and Minimum heap values?
H: Free Heap, M: Max heap block, S: Stack watermark level.

Here's the sdkconfig, I'm now experimenting the same issue under ESP32S3 (This sdkconfig belongs to ESP32S3 environment):
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/11936038/sdkconfig.txt)

> Also can you share the free heap and minimum free heap sizes at the time of the issue?
I'll Insert that for the upcoming logs.

> Also, you mentioned that the previously connected devices continue the connection without any problem, Have you tried disconnecting those clients and trying to establish the connection once again after you have faced the above issue?
When that happened, even connected clients cannot reconnect after a disconnect (As Shift+F5 Refresh).

HamzaHajeir, 2023-07-03:
When the memory leak of 164 bytes happens, I've notices unchanged minimum heap block size of `2137063` (PSRAM is activated), while the heap decreases by 164 bytes per new connection/handshake.

I'm not sure why the minimum heap block is unchanged.

** Just got an instance where the leak is 176 bytes, the modules debug is off for current builds, please ask for it if needed.

AdityaHPatwardhan, 2023-07-03:
Please get the readings for `MALLOC_CAP_INTERNAL`,
In your `sdkconfig` I can see that the option `CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC` has been set to 1 ( Which is the default behaviour for mbedTLS)
Minimum heap block notifies the lowest value for free_heap_size through its lifetime, so it can remain constant but would be lower than free_heap_size.

HamzaHajeir, 2023-07-03:
Here's a log with the changes.
I've manually marked the reference heap point, which is the value prior to a new curl HTTPS request.

You can look for marks `===== REFRENCE POINT =====` and `==== AFTER SCAVENGE ==== (XXXXXX)`

Wherein the scavenge occurs after the connection goes idle and got cleaned by the system internals.

[device-monitor-230703-155511 copy.log](https://github.com/espressif/esp-idf/files/11937093/device-monitor-230703-155511.copy.log)

Note that the debug is turned off, and further I've manually removed unrelated logs (publishes and "H4P" logs).

AdityaHPatwardhan, 2023-07-21:
Hi @HamzaHajeir I dont think the error stems from a code that is a part of esp-idf. We have not seen such error with any of the esp-idf examples.
It is highly likely that the issue is caused by the patch that you have applied.
I would recommend playing close attention to the part of code which is triggered when 
1) New session is created
2) Old session is destroyed

I am closing the issue since it doesn't seem to be related to esp-idf.
Please feel free to re-open in you still think it is an esp-idf issue and would be fixed by changing some part of code from esp-idf.

Thanks,
Aditya

HamzaHajeir, 2023-07-21:
Hi Aditya,

I've read the code again, there's no session initiation in altcp_tls_mbedtls.c, nor in application code.

It's proven by logs (by inserted prints) that `altcp_mbedtls_dealloc()` [being called per connection close](      mbedtls_ssl_free(&state->ssl_context);), which do call `mbedtls_ssl_free()`.

As servers share the configurations, corresponding keys and certificated wont be freed unless the LISTEN connection gets closed.

> It is highly likely that the issue is caused by the patch that you have applied.

Does esp-idf use LWIP_ALTCP, or have someone tested with it and confirmed the issue absence? If so, please inform with the exact version of LWIP_ALTCP used in tests.

Alternatively, can you confirm a successful memory leak test is done for TLS servers? Please inform with such projects.

Regards,

AdityaHPatwardhan, 2023-07-24:
Hi @HamzaHajeir,

esp-idf example are tested against possible leaks in multiple areas. Though I won't say every leak always get caught.
Just to check here, I have attached a log where I have attempted more than 200 connections to a TLS server created with https_server example. I dont observe any leak other than some memory changes due to socket buffers (which are released after some time hence might be confused with memory leak).
[leak_test.log](https://github.com/espressif/esp-idf/files/12141930/leak_test.log)
I am re-opening the issue just in case, but I think if we are making changes to lwip and then some leak is occurring, wouldn't it be better to raise issue at https://github.com/lwip-tcpip/lwip (through appropriate means)
In order to be sure about that, is it possible for you to provide some kind of reproducible code which does not use any external patch but reproduces memory leak?

reinismu, 2023-08-01:
I switched to WolfSSL https://github.com/esp-rs/esp-idf-sys/issues/226 and I still get the same issue

HamzaHajeir, 2023-08-01:
> I have attached a log where I have attempted more than 200 connections to a TLS server created with https_server example. I dont observe any leak other than some memory changes due to socket buffers [...]

I've checked it, it depends on latest ESP-IDF (v5.2) which is accompanied with major updated mbedTLS version (v3 instead of v2.28.3), so it won't be helpful to ensure the issue. (It could be any of ESP-IDF v4.4, mbedTLS 2.28.3 or lwip and furthers).

reinismu, 2023-08-01:
> Per error, [check this line](https://github.com/Mbed-TLS/mbedtls/blob/cbc495eb0f29c9ccfa5e9da2d4c5b3a2691c49c3/include/mbedtls/ssl.h#L87C1-L89C1).
> 
> You might activate mbedTLS debug to see what does really happen after that point.

Based that WolfSSL had the same issue as mbedtls I assume issue is in my code or esp-idf. 
Could be my issue is not related with your. You could try sending multiple failed SSL handshakes and see if that breaks stuff

AdityaHPatwardhan, 2024-01-08:
Hi @HamzaHajeir,
I have tested the functioning of esp_https_server at my end.
I was not able to reproduce this issue.
@reinismu were you able to reproduce this once again?

reinismu, 2024-01-08:
I did solve my issue. In the end it was my code. It did deadlock in some cases. ESP websockets were not threadsafe, so I had to change how I used them.

HamzaHajeir, 2024-01-13:
Hi @AdityaHPatwardhan, I'm a way of HTTPS webserver really, but I've launched just a normal test (trying to stress manually, debug off), I'll update with further notes.

HamzaHajeir, 2024-01-14:
@AdityaHPatwardhan 
The testbed which run for ~30 hours started perfectly but ended up with serious issues:
- Memory leak.
- WiFi doesn't reconnect except by explicit `WiFi.begin(xxx,yyy)`. (Might be an issue related to Arduino core).
- The server fails at the TCP handshake.

The performance was in degradation over time, At first it started perfectly with 6 websockets secure (wss) clients simultaneously, this number has degraded over time due to memory leaks in the internal memory (For the certificates to take place), until there was only one client can connect at a time.

It's worth noting: 
- I'm building with absent debug for mbedtls, which is a probable reason for this error.
- I've made too many reloads (Even with full reload, forcing full handshakes) over time.
- Accompanying HTTP client requests does work without issues.
- Generally TLS clients have zero issues running over days and weeks.
- Unsecured webserver (HTTP) works the same.
- Even the full house (MQTTS/HTTPS client/Webserver) does work without issues for undefinite period.

If you believe it's not an issue in esp-idf, mbedtls port, or mbedtls itself, please feel free to close.

AdityaHPatwardhan, 2024-02-04:
Hi @HamzaHajeir,
I tried the https server again with using esp-idf as baseline.
I did not find any visual leaks that might explain the issue.
I am closing the issue since I am unable to reproduce the issue at my end and my own tests did not give any indication of memory leak.
Thanks for the issue.
Please feel free to re-open if you find any additional material with which I can reproduce the issue at my end.