espressif/esp-idf #11740: Struggeling with esp_sleep_enable_ext0_wakeup on ESP32-S3

Kampi, 2023-06-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I try to use  `esp_sleep_enable_ext0_wakeup` on an ESP32-S3 to wake up the processor. I have added the interrupt pin from my port expander (OC output) to `GPIO 5` and used the following line to wake up the processor as soon as a low-level assertion from the port expander happens:

```
esp_sleep_enable_ext0_wakeup(GPIO_NUM_5, false);
```

But the processor wakes up directly without a signal and the pin is set to low, but not from the port expander. Now I´m wondering if the processor did it. The docu in the source code says that only GPIO `0,2,4,12-15,25-27,32-39` can be used as input because they have RTC capability, but the [docu](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html) in the web list different pins.

Which one is correct and why does the pin is pulled low from the processor?

songruo, 2023-06-26:
Hi @Kampi,

Besides `esp_sleep_enable_ext0_wakeup`, you need to call `rtc_gpio_pullup_en` and `rtc_gpio_pulldown_dis` on the pin to tie wakeup pins to inactive level before entering sleep.

GPIO 5 is a RTC-capable IO, it can be used to trigger EXT0 wakeup on ESP32S3. We have fixed the incorrect doxygen comment some while ago, it should match with the programming guide on the master branch. Please see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/sleep_modes.html#_CPPv428esp_sleep_enable_ext0_wakeup10gpio_num_ti).

Kampi, 2023-06-26:
Hi @songruo 

thanks for the tip. I thought the sleep management is setting the pull-up / pull-down resistors automatically.

closed 2023-06-26 by Kampi