espressif/esp-idf #11742: stuck in "bus_lock: dev 0 acquired." when restart modbus-tcp-slave using w5500

fivekernels, 2023-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board with ESP32-S3-WROOM

### Power Supply used.

External 5V

### What is the expected behavior?

modbus tcp restart successfully.

### What is the actual behavior?

The program stucked after calling "mbc_slave_start()". no assert, error or reset after stuck, only change log level to "verbose" can see the relevant log: "V bus_lock: dev 0 acquired.". After that, my user program is no longer executed and no other logs are output.

### Steps to reproduce.

1. initialize w5500

```c
void EthernetW5500_Init(void)
{
    eth_mac_config_t mac_config_spi = ETH_MAC_DEFAULT_CONFIG();
    mac_config_spi.rx_task_stack_size = 3072; 
    eth_phy_config_t phy_config_spi = ETH_PHY_DEFAULT_CONFIG();
    phy_config_spi.phy_addr = 1; 
    phy_config_spi.reset_gpio_num = -1;

    gpio_install_isr_service(0);

    spi_bus_config_t buscfg = {
        .sclk_io_num = W5500_SCLK,
        .miso_io_num = W5500_MISO,
        .mosi_io_num = W5500_MOSI,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };
    ESP_ERROR_CHECK(spi_bus_initialize(W5500_SPIHOST, &buscfg, SPI_DMA_CH_AUTO));

    spi_device_interface_config_t spi_devcfg = {
        .mode = 0,
        .clock_speed_hz = W5500_SPI_FREQ * 1000 * 1000,
        .spics_io_num = W5500_CS,
        .queue_size = 20
    };

    eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(W5500_SPIHOST, &spi_devcfg);
    w5500_config.int_gpio_num = W5500_INT;
    esp_eth_mac_t *mac_spi = esp_eth_mac_new_w5500(&w5500_config, &mac_config_spi);
    esp_eth_phy_t *phy_spi = esp_eth_phy_new_w5500(&phy_config_spi);

    esp_eth_config_t eth_config_spi = ETH_DEFAULT_CONFIG(mac_spi, phy_spi); 
    esp_eth_handle_t eth_handle_spi = NULL; 
    ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config_spi, &eth_handle_spi)); 

    /* The SPI Ethernet module might not have a burned factory MAC address, we cat to set it manually.
    02:00:00 is a Locally Administered OUI range so should not be used except when testing on a LAN under your control.
    */
    ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle_spi, ETH_CMD_S_MAC_ADDR, (uint8_t[]) {
        0x02, 0x00, 0x00, 0x32, 0x28, 0x04
    }));
    
    esp_event_loop_create_default(); 
    esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL); 

    ESP_ERROR_CHECK(esp_netif_init());
    // Create new default instance of esp-netif for Ethernet
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&cfg); 

    ESP_ERROR_CHECK(esp_netif_dhcpc_stop(eth_netif)); // disable dhcp
    esp_netif_ip_info_t info_t = {0};
    info_t.ip.addr = esp_ip4addr_aton((const char *)"172.17.1.2");
    info_t.netmask.addr = esp_ip4addr_aton((const char *)"255.255.255.0");
    info_t.gw.addr = esp_ip4addr_aton((const char *)"172.17.1.1");
    ESP_ERROR_CHECK( esp_netif_set_ip_info(eth_netif, &info_t));

    esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle_spi));
    esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL); 

    ESP_ERROR_CHECK(esp_eth_start(eth_handle_spi)); 
    
    ETH_Handle_W5500 = eth_handle_spi;
    ETH_Netif_W5500 = eth_netif;
}
```

2. Initialize Modbus TCP Slave

```c
esp_err_t ModbusTCPSlaveInitialize(void)
{
    /* 1.Modbus Port Initialization */ // https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/port_initialization.html#modbus-api-port-initialization
    // The ESP_Modbus supports Modbus SERIAL and TCP ports and a port must be initialized before calling any other Modbus API. 
    // The functions below are used to create and then initialize Modbus controller interface (either master or slave) 
    // over a particular transmission medium (either Serial or TCP/IP):
    void* slave_handler = NULL; // Pointer to allocate interface structure
    esp_err_t err = mbc_slave_init_tcp(&slave_handler); // Initialization of Modbus slave for TCP
    if (slave_handler == NULL || err != ESP_OK)
    {
        // Error handling is performed here
        free(slave_handler);
        ESP_RETURN_ON_ERROR(err, TAG, "mbc_slave_init() returns %s(0x%X)", esp_err_to_name(err), err);
    }

    /* 2.Configuring Slave Data Access */
    err = ModbusSlaveDataAccessConfiguration();
    if (err != ESP_OK)
        return err;
    /* 3.Slave Communication Options */
    mb_communication_info_t comm_info = {
        .ip_port = CONFIG_FMB_TCP_PORT_DEFAULT, // Modbus TCP port number (default = 502)
        .ip_addr_type = MB_IPV4,                // version of IP protocol
        .ip_mode = MB_MODE_TCP,                 // Port communication mode
        .ip_addr = NULL,                        // This field keeps the client IP address to bind, NULL - bind to any client
        // .ip_netif_ptr = esp_netif_ptr      // esp_netif_ptr - pointer to the corresponding network interface
        .ip_netif_ptr = ETH_Netif_W5500              // esp_netif_ptr - pointer to the corresponding network interface
    };
    err = mbc_slave_setup((void*)&comm_info); // Setup communication parameters and start stack
    if (err != ESP_OK)
    {
        ESP_RETURN_ON_ERROR(err, TAG, "mbc_slave_setup() fail, returns %s(0x%X)", esp_err_to_name(err), err);
    }

    /* 4.Slave Communication */
    err = mbc_slave_start();
    if (err != ESP_OK)
    {
        ESP_RETURN_ON_ERROR(err, TAG, "mbc_slave_start() fail, returns %s(0x%X)", esp_err_to_name(err), err);
    }

    return ESP_OK;
}
```

3. In another task, destroy modbus and restart it later

```c
while (1)
{
    vTaskDelay(3000 / portTICK_PERIOD_MS);
    ESP_ERROR_CHECK_WITHOUT_ABORT(mbc_slave_destroy());
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    ModbusTCPSlaveInitialize();
}
```

### Debug Logs.

```plain
I (661940) MB_TCP_SLAVE_PORT: Socket (#54), lisV bus_lock: dev 0 acquired.
```

### More Information.

I've updated the version of esp-modbus to 1.0.11.

This issue does not occur the first time modbus-tcp is started, and does not occur every time modbus-tcp is restarted; This problem usually occurs after destroy and then re-config Modbus TCP for about ten minutes, but sometimes it occurs quickly or not for a long time.

fivekernels, 2023-06-27:
Hi @alisitsyn ,

Thank you for your attention, the ESP-IDF commit I use is 5181de8ac5. My W5500 is using SPI2 and not using another SPI device; Here are the macro definitions that are relevant to the code provided at the beginning:

```c
#define W5500_SPIHOST 2
#define W5500_SCLK 12
#define W5500_MISO 13
#define W5500_MOSI 11
#define W5500_CS 10
#define W5500_INT 14
#define W5500_SPI_FREQ 36 // MHz
```

The sdkconfig file and complete log are as follows:
[skdconfig-and-log.zip](https://github.com/espressif/esp-idf/files/11876994/skdconfig-and-log.zip)

By the way, I found that if I reduce the number of logs printed during initializing modbus, the problem doesn't seem to occur or doesn't occur so quickly (my function ModbusSlaveDataAccessConfiguration() has some log output from the debugging process, which is also reflected in the attachment, and seems to improve the situation if commented out, but I haven't tested it for a long time at the moment).

Kind Regards.

alisitsyn, 2023-06-27:
Hi @fivekernels,

Thank you for update. It looks like it is not easy to reproduce this issue. I will continue with this issue once get appropriate hardware. I will update you as soon as possible.

alisitsyn, 2023-06-28:
Hi @fivekernels,

Please let me know additional information. I am missing the information :
- what are priorities of your tasks
- do you read/write the data using modbus or just start stack then destroy and do it periodically?
- do you destroy periodically the W5500 or just init it once on start?

Could you send your code to reproduce this issue? Thanks.

fivekernels, 2023-06-28:
Hi @alisitsyn ,

Here are some more detailed information you need:

- There are two tasks in my application, one that continuously executes mbc_slave_check_event() to handle modbus events, and the other cycle executes destroy modbus and then reconfigures modbus; Both tasks have a priority of 6.
- During the whole process, there are no modbus read and write operations; Even though I don't plug the network cable into the w5500 module, the problem persists.
- The W5500 module is initialized only once when the system is powered on. I have tried to reconfigure the w5500 but have not found a proper way to destroy.

I'll provide the code later. Thank you.

Kind Regards.

fivekernels, 2023-06-29:
Hi again @alisitsyn ,

Below is my complete project, I kept the build folder (.zip) and the ESP Monitor output. I also found that for the same bin file, the moment the program gets stuck after burning is the same (for the bin file I uploaded, the program must stop at moment 51770 after each reset).

[ESP32S3_Test_Issue11742.zip](https://github.com/espressif/esp-idf/files/11901575/ESP32S3_Test_Issue11742.zip)

Thank you.

Kind Regards.

alisitsyn, 2023-07-19:
Hi @fivekernels ,

Sorry for long delay with the answer.

I was trying to reproduce the issue and was able to do this with the different hardware for now.
Your use case is very specific when you try reinitialize stack in one task and process the data in other task.

Please consider the changes below.

1. The ModbusProcess_Task task shall be synchronized with the TeardownModbus_Task and do not access the Modbus API while stack is not initialized.
This can be done using the semaphore object instead of ModbusConnectionMode.
The `ModbusSlaveEventProcess()` then asynchronously check the events and process them with the timeout.

APP_Modbus.c:
```
SemaphoreHandle_t xMbSemaphore;

SemaphoreHandle_t xSemaCreate()
{
    SemaphoreHandle_t xSema = xSemaphoreCreateBinary();
    xSemaphoreGive(xSema);
    return xSema;
}

void vSemaDelete()
{
    vSemaphoreDelete(xMbSemaphore);
}

bool xSemaTake(uint32_t timeout)
{
    BaseType_t status = pdTRUE;
    status = xSemaphoreTake(xMbSemaphore, timeout);
    ESP_RETURN_ON_FALSE((status == pdTRUE), false , TAG, "could not take sema during timeout.");
    return true;
}

void xSemaRelease()
{
    BaseType_t status = pdFALSE;
    status = xSemaphoreGive(xMbSemaphore);
    ESP_RETURN_ON_FALSE((status == pdTRUE), ; , TAG, "could not release sema (already released?).");
}

void ModbusProcess_Task(void *arg)
{
    esp_err_t ret = 0;
    mb_param_info_t reg_info;
    //EthernetW5500_Init();
    xMbSemaphore = xSemaCreate();
    EthernetIP101_WIFI_Init();
    xSemaRelease();
    while (1)
    {
        xSemaTake(portMAX_DELAY);
        ESP_LOGI(TAG, "Check parameters.");
        ret = ModbusSlaveEventProcess(2000); // Wait 2 sec for events
        if ((ret != ESP_OK) && (ret != ESP_ERR_INVALID_STATE)) {
            ESP_LOGE(TAG, "ModbusSlaveEventProcess() returns %s(0x%X)", esp_err_to_name(ret), ret);
        }
        ESP_LOGI(TAG, "End check parameters, release.");
        xSemaRelease();
        vTaskDelay(20 / portTICK_PERIOD_MS);
    }
    ESP_LOGI(TAG, "ModbusSlaveEventProcess() is Completed");
    vTaskDelete(NULL);
```
2. The `mbc_slave_check_event()` function in the stack waits for incoming events forever and exits when the stack is destroyed from TearDown task.
This prevents to use the synchronization approach. Instead of this function the `mbc_slave_get_param_info()` function can be used to wait for events with the timeout;

```
esp_err_t ModbusSlaveEventProcess(uint32_t timeout_ms)
{
    esp_err_t err = ESP_OK;
    mb_param_info_t reg_info; // keeps the Modbus registers access information
    mb_event_group_t event;

    // Get information about data accessed from master
    err = mbc_slave_get_param_info(&reg_info, timeout_ms);
    if ((err != ESP_OK)) {
        return err; // timeout happens waiting for events
    }

    event = reg_info.type;
    const char* rw_str = (event & MB_READ_MASK) ? "READ" : "WRITE";

    // Filter events and process them accordingly
    if (event & (MB_EVENT_COILS_RD | MB_EVENT_COILS_WR)) // Coil
    {
        ESP_LOGI(TAG, "[%s]Coil (%lu us), ADDR:%lu, TYPE:%lu, INST_ADDR:0x%.4lx, SIZE:%lu", rw_str,
            (uint32_t)reg_info.time_stamp, (uint32_t)reg_info.mb_offset, (uint32_t)reg_info.type,
            (uint32_t)reg_info.address, (uint32_t)reg_info.size
        );
    //..............other code

```

3. The `TeardownModbus_Task` can be changed as below:

```
void TeardownModbus_Task(void *arg)
{
    vTaskDelay(3000 / portTICK_PERIOD_MS);
    while (1)
    {
        ESP_LOGI(TAG, "modbus will be destroyed after 3s");
        vTaskDelay(3000 / portTICK_PERIOD_MS);
        xSemaTake(portMAX_DELAY);
        ESP_LOGI(TAG, "modbus TCP destroy.");
        ModbusDestroy();
        ESP_LOGI(TAG, "modbus have been destroyed, and will be created after 1s");
        //vTaskDelay(1000 / portTICK_PERIOD_MS);
        ESP_LOGI(TAG, "modbus TCP create.");
        ModbusTCPSlaveInitialize();
        xSemaRelease();
        ESP_LOGI(TAG, "modbus TCP have been created");
    }
    // ....
```
The changes above do not solve the issue completely but can make code predictable.
As per results of investigation the issue is in stack enable - disable sequence and using of vTaskSuspend(), vTaskResume() in the port code.
I made also updates in the port code but it is still under testing. Please apply the changes in the patch below or update the `port_tcp_slave.c` file in `espressif__esp-modbus` which should be moved into `components` folder of your project. In this case the updated component will be linked instead of existing one. 

[tcp_slave_fix_start_destroy_sequence.patch.log](https://github.com/espressif/esp-idf/files/12092724/tcp_slave_fix_start_destroy_sequence.patch.log)

[port_tcp_slave.zip.log](https://github.com/espressif/esp-idf/files/12092736/port_tcp_slave.zip.log)

fivekernels, 2023-07-20:
Hi @alisitsyn ,

Thanks for the fix,

I'll try it out and give you feedback later.

Thanks.

fivekernels, 2023-08-14:
Hi again @alisitsyn ,

I've tested the fix above and I think it works well for me.

Thank you.

Kind Regards.

alisitsyn, 2023-08-21:
Hi fivekernels,

Thank you for update. The fix will go through the formal process and will be merged in scheduled  update.
Let me know if you have some further issues related to it.

alisitsyn, 2023-10-10:
Hi @fivekernels ,

the issue with stack enable - disable sequence and using of vTaskSuspend(), vTaskResume() has been fixed in esp-modbus v1.0.12. Please update your project and verify if it solves your issue then please close the issue. Thanks.

alisitsyn, 2023-11-21:
The issue is closed.

closed 2023-11-21 by alisitsyn