espressif/esp-idf #11746: Hello_world project build fails on M1 Pro mac

cwahn, 2023-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hey, after I followed all the steps listed at the link; 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html 

`idf.py build` fails with the following output. 

```
[1/8] Performing build step for 'bootloader'
[1/1] cd /Users/myname/esp/hello_world/build/bootloader/esp-idf/esptool_py && /Users/myname/.espressif/python_env/idf5.2_py3.11_env/bin/python /Users/myname/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/myname/esp/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x68a0 bytes. 0x760 bytes (7%) free.
[2/8] No install step for 'bootloader'
[3/8] Completed 'bootloader'
[4/8] Generating ld/sections.ld
FAILED: esp-idf/esp_system/ld/sections.ld /Users/myname/esp/hello_world/build/esp-idf/esp_system/ld/sections.ld 
cd /Users/myname/esp/hello_world/build/esp-idf/esp_system && /Users/myname/.espressif/python_env/idf5.2_py3.11_env/bin/python /Users/myname/esp/esp-idf/tools/ldgen/ldgen.py --config /Users/myname/esp/hello_world/sdkconfig --fragments-list "/Users/myname/esp/esp-idf/components/xtensa/linker.lf;/Users/myname/esp/esp-idf/components/esp_ringbuf/linker.lf;/Users/myname/esp/esp-idf/components/driver/linker.lf;/Users/myname/esp/esp-idf/components/esp_pm/linker.lf;/Users/myname/esp/esp-idf/components/esp_mm/linker.lf;/Users/myname/esp/esp-idf/components/spi_flash/linker.lf;/Users/myname/esp/esp-idf/components/esp_system/linker.lf;/Users/myname/esp/esp-idf/components/esp_system/app.lf;/Users/myname/esp/esp-idf/components/esp_rom/linker.lf;/Users/myname/esp/esp-idf/components/hal/linker.lf;/Users/myname/esp/esp-idf/components/log/linker.lf;/Users/myname/esp/esp-idf/components/heap/linker.lf;/Users/myname/esp/esp-idf/components/soc/linker.lf;/Users/myname/esp/esp-idf/components/esp_hw_support/linker.lf;/Users/myname/esp/esp-idf/components/freertos/linker_common.lf;/Users/myname/esp/esp-idf/components/freertos/linker.lf;/Users/myname/esp/esp-idf/components/newlib/newlib.lf;/Users/myname/esp/esp-idf/components/newlib/system_libs.lf;/Users/myname/esp/esp-idf/components/esp_common/common.lf;/Users/myname/esp/esp-idf/components/esp_common/soc.lf;/Users/myname/esp/esp-idf/components/app_trace/linker.lf;/Users/myname/esp/esp-idf/components/esp_event/linker.lf;/Users/myname/esp/esp-idf/components/esp_phy/linker.lf;/Users/myname/esp/esp-idf/components/lwip/linker.lf;/Users/myname/esp/esp-idf/components/esp_netif/linker.lf;/Users/myname/esp/esp-idf/components/esp_wifi/linker.lf;/Users/myname/esp/esp-idf/components/bt/linker.lf;/Users/myname/esp/esp-idf/components/esp_adc/linker.lf;/Users/myname/esp/esp-idf/components/esp_gdbstub/linker.lf;/Users/myname/esp/esp-idf/components/esp_psram/linker.lf;/Users/myname/esp/esp-idf/components/esp_lcd/linker.lf;/Users/myname/esp/esp-idf/components/espcoredump/linker.lf;/Users/myname/esp/esp-idf/components/ieee802154/linker.lf;/Users/myname/esp/esp-idf/components/openthread/linker.lf" --input /Users/myname/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in --output /Users/myname/esp/hello_world/build/esp-idf/esp_system/ld/sections.ld --kconfig /Users/myname/esp/esp-idf/Kconfig --env-file /Users/myname/esp/hello_world/build/config.env --libraries-file /Users/myname/esp/hello_world/build/ldgen_libraries --objdump /Users/myname/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump
Traceback (most recent call last):
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 152, in _get_infos_from_file
    results = parser.parseString(info.content, parseAll=True)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/myname/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/pyparsing/util.py", line 256, in _inner
    return fn(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/myname/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/pyparsing/core.py", line 1190, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected end of text, found 'default'  (at char 1), (line:2, col:1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen.py", line 175, in <module>
    main()
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen.py", line 151, in main
    mapping_rules = generation_model.generate(sections_infos)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/generation.py", line 514, in generate
    root_node.insert(entity, sections, target, flags, entities)
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/generation.py", line 241, in insert
    self.child_placement(entity, sections, target, flags, sections_db)
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/generation.py", line 233, in child_placement
    child.insert(entity, sections, target, flags, sections_db)
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/generation.py", line 241, in insert
    self.child_placement(entity, sections, target, flags, sections_db)
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/generation.py", line 233, in child_placement
    child.insert(entity, sections, target, flags, sections_db)
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/generation.py", line 241, in insert
    self.child_placement(entity, sections, target, flags, sections_db)
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/generation.py", line 296, in child_placement
    found_sections = sections_db.get_sections(self.parent.name, self.name)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 197, in get_sections
    obj = self._match_obj(archive, obj)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 183, in _match_obj
    objs = self.get_objects(archive)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 176, in get_objects
    self._process_archive(archive)
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 164, in _process_archive
    parsed = self._get_infos_from_file(stored)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/myname/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 154, in _get_infos_from_file
    raise ParseException('Unable to parse section info file ' + info.filename + '. ' + p.msg)
pyparsing.exceptions.ParseException: Unable to parse section info file /Users/myname/esp/hello_world/build/esp-idf/esp_event/libesp_event.a. Expected end of text  (at char 0), (line:1, col:1)
ninja: build stopped: subcommand failed.

```

What could be the issue?

I am using the default esp_idf (master) at M1 Pro MacBook Pro with macOS Ventura 13.4.

fhrbata, 2023-06-26:
Hello @cwahn , this issue should be already fixed, but it looks like we are facing some hiccup with the github sync. Hopefully this will be resolved soon. In the meantime, as a temporary workaround, you can downgrade the pyparsing package to 3.0.9 version e.g. with `$ pip install 'pyparsing<3.1.0'`  in the active esp-idf python venv(after including the .export.sh).
Thank you for the report!

fhrbata, 2023-06-26:
Hello @cwahn, I believe this was fixed with https://github.com/espressif/esp-idf/commit/4020b0deb036ffa3fd01d2dac2297670f65fb7c3. I'm closing this, but please feel free to re-open if the problem persists. Thank you!

closed 2023-06-26 by fhrbata