espressif/esp-idf #11748: pyparsing-3.1.0 breaks idf.py build

morgana2313, 2023-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1183-gd2fb7240a7

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

`pyparsing-3.1.0` (released last week) breaks `idf.py build`.  `pyparsing-3.0.9` works without problems.

I expect `~/esp-idf/examples/get-started/hello_world$ idf.py clean &&  idf.py build` to build without errors.

### What is the actual behavior?

Instead it gives a build error:

```
Traceback (most recent call last):
  File "/home/s/esp-idf/tools/ldgen/ldgen/entity.py", line 152, in _get_infos_from_file
    results = parser.parseString(info.content, parseAll=True)
  File "/home/s/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/pyparsing/util.py", line 256, in _inner
    return fn(self, *args, **kwargs)
  File "/home/s/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/pyparsing/core.py", line 1190, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected end of text, found 'default'  (at char 1), (line:2, col:1)
```

### Steps to reproduce.

1. Either:
   * Install esp-idf: `git clone --recursive https://github.com/espressif/esp-idf.git && cd esp-idf && ./install.sh` 
   * `~/.espressif/python_env/idf5.2_py3.10_env/bin/pip install --force-reinstall pyparsing==3.1.0`
2. `idf.py build` gives an error
3. `~/.espressif/python_env/idf5.2_py3.10_env/bin/pip install --force-reinstall pyparsing==3.0.9`
4. `idf.py build` builds without error.

### Build or installation Logs.

```plain
<details><summary>details</summary>

s@dev-s5:~/esp-idf/examples/get-started/hello_world$ idf.py clean &&  idf.py build
/home/s/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: clean
Running ninja in directory /home/s/esp-idf/examples/get-started/hello_world/build
Executing "ninja clean"...
[2/2] Cleaning all built files...
Cleaning... 891 files.
Done
/home/s/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running ninja in directory /home/s/esp-idf/examples/get-started/hello_world/build
[... 78 lines cut ...]
    child.insert(entity, sections, target, flags, sections_db)
  File "/home/s/esp-idf/tools/ldgen/ldgen/generation.py", line 241, in insert
    self.child_placement(entity, sections, target, flags, sections_db)
  File "/home/s/esp-idf/tools/ldgen/ldgen/generation.py", line 233, in child_placement
    child.insert(entity, sections, target, flags, sections_db)
  File "/home/s/esp-idf/tools/ldgen/ldgen/generation.py", line 241, in insert
    self.child_placement(entity, sections, target, flags, sections_db)
  File "/home/s/esp-idf/tools/ldgen/ldgen/generation.py", line 296, in child_placement
    found_sections = sections_db.get_sections(self.parent.name, self.name)
  File "/home/s/esp-idf/tools/ldgen/ldgen/entity.py", line 197, in get_sections
    obj = self._match_obj(archive, obj)
  File "/home/s/esp-idf/tools/ldgen/ldgen/entity.py", line 183, in _match_obj
    objs = self.get_objects(archive)
  File "/home/s/esp-idf/tools/ldgen/ldgen/entity.py", line 176, in get_objects
    self._process_archive(archive)
  File "/home/s/esp-idf/tools/ldgen/ldgen/entity.py", line 164, in _process_archive
    parsed = self._get_infos_from_file(stored)
  File "/home/s/esp-idf/tools/ldgen/ldgen/entity.py", line 154, in _get_infos_from_file
    raise ParseException('Unable to parse section info file ' + info.filename + '. ' + p.msg)
pyparsing.exceptions.ParseException: Unable to parse section info file /home/s/esp-idf/examples/get-started/hello_world/build/esp-idf/esp_event/libesp_event.a. Expected end of text  (at char 0), (line:1, col:1)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/s/esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_23717 and /home/s/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_23717
```
</details>
```

igrr, 2023-06-26:
https://github.com/espressif/esp-idf/issues/11746

dobairoland, 2023-06-27:
Anyone who stumbles upon this, there are two related fixes:

1. To re-run the install script again. It should install `pyparsing<3.1` for now.
2. https://github.com/espressif/esp-idf/commit/4020b0deb036ffa3fd01d2dac2297670f65fb7c3 should be the proper fix for `pyparsing>=3.1`.  

The restriction of `pyparsing<3.1` will be lifter later giving enough time to everyone to update to the linked fix.

We are sorry for the inconvenience!

closed 2023-06-27 by dobairoland