espressif/esp-idf #11749: Provisioning Manager reporting WIFI_PROV_STA_AUTH_ERROR with valid credentitals

homedad, 2023-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Does the code below map too many error cases into "Auth Error"?

We have seen this ourselves during development, and now we have several users reporting that they have input their credentials correctly, and the provisioning library reports WIFI_PROV_STA_AUTH_ERROR, with which we tell the user, "Bad Creds". This doesn't help the user solve the problem.

Excerpt from components/wifi_provisioning/src/manager.c (IDF Version 4.4.3):

```        
switch (disconnected->reason) {
        case WIFI_REASON_AUTH_EXPIRE:
        case WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT:
        case WIFI_REASON_AUTH_FAIL:
        case WIFI_REASON_ASSOC_EXPIRE:
        case WIFI_REASON_HANDSHAKE_TIMEOUT:
        case WIFI_REASON_MIC_FAILURE:
            ESP_LOGE(TAG, "STA Auth Error");
            prov_ctx->wifi_disconnect_reason = WIFI_PROV_STA_AUTH_ERROR;
```

mahavirj, 2023-07-07:
@homedad I think you had reported similar issue in https://github.com/espressif/esp-idf/issues/9399, and it was fixed in IDF 4.4.4 release. Can you please confirm if you have that fix available in your code-base? Alternatively, is it possible for you to share the WiFi packet capture for the part when you observe this issue?

homedad, 2023-07-07:
Correct. This is a related issue. I will collect data. It’s is rare however
in my home environment so this may take some time.

closed 2023-07-27 by espressif-bot