espressif/esp-idf #11750: 0x40375310: call_start_cpu1 at D:/ESP32/ESP32_5.0/Espressif/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

szwlxj, 2023-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.2-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-WROOM 1

### Power Supply used.

USB

### What is the expected behavior?

新建的空项目，没有添加代码

### What is the actual behavior?

可以编译，运行，只是有这个提示（字符是黄色的）
0x40375310: call_start_cpu1 at D:/ESP32/ESP32_5.0/Espressif/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

### Steps to reproduce.

#include <stdio.h>
#include "freertos/freertos.h"
#include "freertos/task.h"

void app_main(void)
{
}

### Debug Logs.

```plain
G:\My_workspace\SCK210\ESP32_S3\lt_0.0.1>idf.py -p com6 monitor
Executing action: monitor
Running idf_monitor in directory G:\My_workspace\SCK210\ESP32_S3\lt_0.0.1
Executing "D:\ESP32\ESP32_5.0\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe D:\ESP32\ESP32_5.0\Espressif\frameworks\esp-idf-v5.0.2\tools/idf_monitor.py -p com6 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 G:\My_workspace\SCK210\ESP32_S3\lt_0.0.1\build\lt_0.0.1.elf --force-color -m 'D:\ESP32\ESP32_5.0\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe' 'D:\ESP32\ESP32_5.0\Espressif\frameworks\esp-idf-v5.0.2\tools\idf.py' '-p' 'com6'"...
--- idf_monitor on com6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x165c
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2d9c
entry 0x403c9900
I (25) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (25) boot: compile time 13:53:25
I (25) boot: chip revision: v0.1
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 16MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=098e4h ( 39140) map
I (100) esp_image: segment 1: paddr=0001990c vaddr=3fc92100 size=02a84h ( 10884) load
I (103) esp_image: segment 2: paddr=0001c398 vaddr=40374000 size=03c80h ( 15488) load
I (113) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19f14h (106260) map
I (137) esp_image: segment 4: paddr=00039f3c vaddr=40377c80 size=0a3d8h ( 41944) load
I (152) boot: Loaded app from partition at offset 0x10000
I (153) boot: Disabling RNG early entropy source...
I (164) octal_psram: vendor id    : 0x0d (AP)
I (164) octal_psram: dev id       : 0x02 (generation 3)
I (165) octal_psram: density      : 0x03 (64 Mbit)
I (169) octal_psram: good-die     : 0x01 (Pass)
I (175) octal_psram: Latency      : 0x01 (Fixed)
I (180) octal_psram: VCC          : 0x01 (3V)
I (185) octal_psram: SRF          : 0x01 (Fast Refresh)
I (191) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (197) octal_psram: BurstLen     : 0x01 (32 Byte)
I (202) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (208) octal_psram: DriveStrength: 0x00 (1/1)
I (213) esp_psram: Found 8MB PSRAM device
I (218) esp_psram: Speed: 40MHz
I (222) cpu_start: Pro cpu up.
I (226) cpu_start: Starting app cpu, entry point is 0x40375310
0x40375310: call_start_cpu1 at D:/ESP32/ESP32_5.0/Espressif/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (963) esp_psram: SPI SRAM memory test OK
I (973) cpu_start: Pro cpu start user code
I (973) cpu_start: cpu freq: 240000000 Hz
I (973) cpu_start: Application information:
I (976) cpu_start: Project name:     lt_0.0.1
I (981) cpu_start: App version:      1
I (985) cpu_start: Compile time:     Jun 27 2023 13:53:12
I (991) cpu_start: ELF file SHA256:  78a135be4cd21205...
I (997) cpu_start: ESP-IDF:          v5.0.2-dirty
I (1003) cpu_start: Min chip rev:     v0.0
I (1007) cpu_start: Max chip rev:     v0.99
I (1012) cpu_start: Chip rev:         v0.1
I (1017) heap_init: Initializing. RAM available for dynamic allocation:
I (1024) heap_init: At 3FC95608 len 00054108 (336 KiB): DRAM
I (1031) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1037) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1044) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (1050) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (1058) spi_flash: detected chip: gd
I (1062) spi_flash: flash io: dio
I (1067) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1087) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
```

### More Information.

这个提示在示例代码上出有出现，不知道是不是我哪里设置出问题了，使用的开发板上网店买的，模块为ESP32-S3-WROOM 1  N16R8

ESP-Marius, 2023-06-27:
This is normal, the app logs:

`I (226) cpu_start: Starting app cpu, entry point is 0x40375310`

And `esp-idf-monitor` will try to automatically decode anything that looks like an address to executable code to show you the source file/line number for it.

See [自动解码地址](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/tools/idf-monitor.html#id3) for more info.

szwlxj, 2023-06-27:
哦，谢谢，明白了

closed 2023-06-27 by ESP-Marius