espressif/esp-idf #11754: Guru Meditation Error: Core 1 panic'ed (IllegalInstruction)

choralou, 2023-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi

an esp32 and a bme280, send temperature pressure and humidity to a php page.
The php page receives the data and writes it to the sql database.
But I get errors, which make the esp32 reboot before the end of the delay.
Serial monitor copy

`Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (219) psram: PSRAM ID read error: 0xffffffff

Connecting to: Freebox-XXXXXX
......Internet connected
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT

HTTP_EVENT_ON_HEADER, key=Server, value=nginx
HTTP_EVENT_ON_HEADER, key=Date, value=Thu, 22 Jun 2023 17:42:17 GMT
HTTP_EVENT_ON_HEADER, key=Content-Type, value=text/html; charset=UTF-8
HTTP_EVENT_ON_HEADER, key=Transfer-Encoding, value=chunked
HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
HTTP_EVENT_ON_HEADER, key=X-SSO-WAT, value=You've just been SSOed
HTTP_EVENT_ON_HEADER, key=Content-Security-Policy, value=upgrade-insecure-requests
HTTP_EVENT_ON_HEADER, key=X-Content-Type-Options, value=nosniff
HTTP_EVENT_ON_HEADER, key=X-XSS-Protection, value=1; mode=block
HTTP_EVENT_ON_HEADER, key=X-Download-Options, value=noopen
HTTP_EVENT_ON_HEADER, key=X-Permitted-Cross-Domain-Policies, value=none
HTTP_EVENT_ON_HEADER, key=X-Frame-Options, value=SAMEORIGIN
HTTP_EVENT_ON_HEADER, key=Permissions-Policy, value=interest-cohort=()
HTTP_EVENT_ON_HEADER, key=Strict-Transport-Security, value=max-age=63072000; includeSubDomains; preload
HTTP_EVENT_ON_DATA, len=27
HTTP_EVENT_ON_DATA, len=5
HTTP_EVENT_ON_FINISH
esp_http_client_get_status_code: 200
HTTP_EVENT_DISCONNECTED
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d2fa0: 6502ad02 00002d52 b1004136
Core  1 register dump:
PC      : 0x400d2fa6  PS      : 0x00060d30  A0      : 0x800d6384  A1      : 0x3ffb21e0  
A2      : 0x3ffcccdc  A3      : 0x3ffc428c  A4      : 0x00000014  A5      : 0x00000004  
A6      : 0x3ffb8188  A7      : 0x80000001  A8      : 0x800d2fa6  A9      : 0x3ffb21c0  
A10     : 0x00000000  A11     : 0x00000003  A12     : 0x0000000a  A13     : 0x00000001  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000020  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4008ab4e  LEND    : 0x4008ab59  LCOUNT  : 0x00000000  

Backtrace: 0x400d2fa3:0x3ffb21e0 0x400d6381:0x3ffb2290

ELF file SHA256: 40c402ae728632f7

Rebooting...`

Copy of program sent to esp32 Wroom 32

`#include "esp_http_client.h"
#include <WiFi.h>
#include "Arduino.h"
#include <forcedClimate.h>

const char* ssid = "Freebox-XXXXXX";
const char* password = "XXXXXXXXXXXXXXXXXXXX";

const char *post_url = "https://monsite/site/client_temp.php"; // Location to send POSTed data

char post_data[140];

// const String *post_data;

const char *rootCACertificate =  "-----BEGIN CERTIFICATE-----\n" \
    "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx\n" \
"-----END CERTIFICATE-----\n";

bool internet_connected = false;

// Create an instance of a BME280. The parameters I2C bus and I2C address are optional. For example:
// - ForcedClimate climateSensor = ForcedClimate(Wire, 0x76);
// - ForcedClimate climateSensor = ForcedClimate(TinyWireM, 0x77);
ForcedClimate climateSensor = ForcedClimate();

void setup()
{
  Serial.begin(115200);

   //initialisation BME280
   climateSensor.begin(); 
   //variable du capteur
float temp=climateSensor.getTemperatureCelcius();
 
  float pression=climateSensor.getPressure();
 
  float humid=climateSensor.getRelativeHumidity();

  

  if (init_wifi()) { // Connected to WiFi
    internet_connected = true;
    Serial.println("Internet connected");
  }
  

snprintf(post_data, sizeof(post_data),
                "temperature= %f &pression= %f &humidite= %f",
                  temp, pression, humid); 

  

}

bool init_wifi()
{
  int connAttempts = 0;
  Serial.println("\r\nConnecting to: " + String(ssid));
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED ) {
    delay(500);
    Serial.print(".");
    if (connAttempts > 10) return false;
    connAttempts++;
  }
  return true;
}

esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
  switch (evt->event_id) {
    case HTTP_EVENT_ERROR:
      Serial.println("HTTP_EVENT_ERROR");
      break;
    case HTTP_EVENT_ON_CONNECTED:
      Serial.println("HTTP_EVENT_ON_CONNECTED");
      break;
    case HTTP_EVENT_HEADER_SENT:
      Serial.println("HTTP_EVENT_HEADER_SENT");
      break;
    case HTTP_EVENT_ON_HEADER:
      Serial.println();
      Serial.printf("HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
      break;
    case HTTP_EVENT_ON_DATA:
      Serial.println();
      Serial.printf("HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
      if (!esp_http_client_is_chunked_response(evt->client)) {
        // Write out data
        // printf("%.*s", evt->data_len, (char*)evt->data);
      }
      break;
    case HTTP_EVENT_ON_FINISH:
      Serial.println("");
      Serial.println("HTTP_EVENT_ON_FINISH");
      break;
    case HTTP_EVENT_DISCONNECTED:
      Serial.println("HTTP_EVENT_DISCONNECTED");
      break;
  }
  return ESP_OK;
}

static esp_err_t post_something()
{
  esp_err_t res = ESP_OK;

  esp_http_client_handle_t http_client;
  
  esp_http_client_config_t config_client = {0};
  config_client.url = post_url;
  config_client.cert_pem = rootCACertificate ;
  config_client.event_handler = _http_event_handler;
  config_client.method = HTTP_METHOD_POST;

  http_client = esp_http_client_init(&config_client);
  
  esp_http_client_set_post_field(http_client, post_data, strlen(post_data));

  // esp_http_client_set_header(http_client, "Content-Type", "image/jpg"); // sending a jpg file

  esp_err_t err = esp_http_client_perform(http_client);
  if (err == ESP_OK) {
    Serial.print("esp_http_client_get_status_code: ");
    Serial.println(esp_http_client_get_status_code(http_client));
  }

  esp_http_client_cleanup(http_client);
  displaySensorDetails();
}

void loop()
{
  
    post_something();
     delay(600000); 

}`
en installant esp exception decoder
j'obtiens le décodage suivant

`PC: 0x400d2fa6
EXCVADDR: 0x00000000
Decoding stack results
0x400d2fa3:
loop() at /chemin_vers_mon_fichier/mon_fichier.ino line 162`
this line contains the call :
'esp_http_client_cleanup(http_client);'
and also
'0x400d6381: loopTask(void*) at /mon_chemin/.arduino15/packages/esp32/hardware/esp32/2.0.9/cores/esp32/main.cpp line 50'
on line 50
'loop()'
Thanks for your help
Daniel

igrr, 2023-06-27:
The issue is probably that the function `esp_err_t post_something()` has no `return` statement in it.

It is an undefined behavior in C++ to end a non-`void` function without actually returning a value. The compiler [optimizes the program under the assumption that undefined behavior never occurs](https://en.cppreference.com/w/cpp/language/ub#UB_and_optimization), which in this case means that the end of the function is assumed to be unreachable. (If it did, undefined behavior would occur!) Since the function epilogue is assumed to be unreachable, it is removed, as that results in slightly smaller output, and that is what the compiler is optimizing for.

See [compiler explorer](https://godbolt.org/z/ffa9fof9c) for a simplified example.

Previous discussions of this topic:
* https://github.com/espressif/esp-idf/issues/8244#issuecomment-1014657386
* https://github.com/esp8266/Arduino/discussions/8160#discussioncomment-906792

choralou, 2023-06-27:
Thanks igrr
you are right the function returns nothing and crashes.
I got around the problem by inventing a variable that serves no purpose. I initialize it to 0, in the function I bring it back to 0 if it's strictly positive, and at din I return it increased by 1.

I've removed the error!
not an elegant solution, but it works.
thanks to the specialists for finding a satisfactory solution
Danie

0xjakob, 2023-06-29:
@choralou Alternatively, you could change the function to `static void post_something()` if you can handle all errors inside the function.

We're closing this issue since it has been fixed. If this is indeed not the case, please feel free to re-open it.

closed 2023-06-29 by 0xjakob