espressif/esp-idf #11755: Detection of external 32 kHz XTAL in ESP32-C3

ysba, 2023-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi!

I have a hardware application that uses an ESP32-C3-MINI-1 module. This application has a logging feature which relies on the RTC system time. The RTC module is driven by an external 32 kHz crystal for better accuracy.

I'm facing an issue at some units in production. In the very first lines of the ESP log, the following line is shown indicating that there is some problem with the external 32 kHz crytal:
```
W (276) clk: 32 kHz XTAL not found, switching to internal 150 kHz oscillator
```

First thing I did was to check if it is a soldering issue. I checked both XTAL_32K_P and XTAL_32K_N signals with an oscilscope. There was indeed a 32 kHz signal in both pins, indicating that probably was not a soldering issue.

So I went further to see why my ESP module was indicating "crystal not found". The message ``W (276) clk: 32 kHz XTAL not found, switching to internal 150 kHz oscillator`` is written only by the ``select_rtc_slow_clk()`` function, located at ``components/esp_system/port/soc/esp32c3/clk.c`` source file in the ESP-IDF.

The ``select_rtc_slow_clk()`` decide if the external 32 kHz crytal is preset based on the return of ``rtc_clk_cal()`` function, which is located in ``components/esp_hw_support/port/esp32c3/rtc_time.c``. This function calls another one in the same file, called ``rtc_clk_cal_32k_valid()``, which, in turns, verifies if the amount of ticks from the main oscillator (40 MHz in my case) detected at the RCT oscillator match the theoretical expected value (1250000, in my case).

The amount of ticks found in the RTC oscillator is tested to see if it fits in a +/- 0.05 % range (``expected_xtal_cycles`` and ``delta`` variables). If it doesn't fit, this function returns ``false``, the above warning message is shown in log and *the RTC clock is changed to the internal 150 kHz clock source*.

Back to my units in production. I filled up the afore mentioned functions in ESP-IDF to get debug information. The ``expected_xtal_cycles`` value is set to 1250000, and ``delta`` value is set to 625. I got the following information for two units. In both cases, the ``switching to internal 150 kHz oscillator`` was shown.
* One unit has 1249264 ticks, out of range, but only by -0.059%.
* Another one has 1249367ticks, out of range, but only by -0.051%.

One more thing to consider: I was not able to find information about the internal 150 kHz accuracy at the datasheets or technical reference manual of ESP32-C3. However, some theads at the ESP32 forums say that the accuracy is 5% ([link](https://esp32.com/viewtopic.php?f=12&t=5391&sid=2f756237d8c1f86727699f3ab884549f
)).

Taking all this into account, my questions are:
* In this case ``W (276) clk: 32 kHz XTAL not found, switching to internal 150 kHz oscillator`` is misleading. The crystal is in fact preset, and is able to be find by the ESP module. It is slightly out of range (very slightly!). The warning message could just say that the crystal is out of range, and possibly say how much out.
* What is the point of changing from a crystal with 0,059% measured error for another one that is expected to have 5% of accuracy? My opinion is that this swiching only should occur if no external crystal is detected at all.

Thanks in advance!
Cheers,
Yuri

ysba, 2023-06-27:
I have used ESP-IDF versions ``4.4.5`` and ``5.0.2``, both with same results.

Pavelovec, 2023-08-09:
I'm experiencing a similar issue with the external 32 kHz crystal on some of my devices with ESP32-C3. I am using ESP-IDF version 
`4.4.4`. It would be great to have more clarity on this behavior and whether switching to a less accurate internal oscillator is necessary in such cases.

ysba, 2023-08-09:
In order to solve my issue in production, I have created a custom esp-idf based on v5.1, on which I changed the xtal tolerance from +/- 0.05% to 5%.

[xtal_fix.patch](https://github.com/espressif/esp-idf/files/12303612/xtal_fix.patch)

Pavelovec, 2023-08-10:
Big thanks for sharing the patch. I'll give it a shot as well.

cyberhuman, 2023-09-10:
@ysba do you mind sharing what 32kHz crystal you used and with what load?
I thought I had the same problem but in fact, my crystal doesn't even start. The debug output in your patch was very helpful to figure that out. I tried many things but still can't make it work.

ysba, 2023-09-11:
We are currently using this 32 kHz crystal, with no load capacitor: [datasheet link](https://www.chinafronter.com/product/smd3215)

We have been using this same partnumber for years with a dedicated RTC IC with no issues. Now that we are using ESP32-C3, issue I have reported here seems to happen in only some of the units.

cyberhuman, 2023-09-11:
@ysba thank you for the information!

ncleesattel, 2024-01-09:
@ysba many thanks for the patch and mentioning the xtal tolerance in the rtc_time.c
I've been trying for many days now to activate the external time signal and this tolerance band only makes sense if actual ESP32 hardware is within that tolerance band. In may case the ESP32 is off by 0.14% or about 2minutes per day.

ioviedodev, 2024-06-28:
Hi, thank you for your support. Iam using esp32, where I can change the tolerance? In Sdkconfig??

ysba, 2024-06-28:
> Hi, thank you for your support. Iam using esp32, where I can change the tolerance? In Sdkconfig??

I am not sure whether this issue has been fixed or not... When I was working around this issue, xtal tolerance was hard coded. You can find a patch in the comments above.

ioviedodev, 2024-06-28:
Thank you @ysba  for your answer.